esphome/esphome #9523: Preset None gives error

Barry40, 2025-07-15:
### The problem

Hi everyone, i use domoticz and the mqtt tautodiscovery.
I don't know for sure if this error is related to domoticz or esphome, but i never seen it before so i want to be sure where to find the problem.
My log is flooding with errors: `2025-07-15 20:30:32.083 Error: MQTT-HomeAssistant: Climate device invalid/unknown preset_mode received! (ESPclimateaircokantoor: none)
2025-07-15 20:31:39.236 Error: MQTT-HomeAssistant: Climate device invalid/unknown preset_mode received! (ESPclimateaircoslaapkamer: none)
2025-07-15 20:32:32.180 Error: MQTT-HomeAssistant: Climate device invalid/unknown preset_mode received! (ESPclimateaircowoonkamer: none)
2025-07-15 20:33:08.861 Error: MQTT-HomeAssistant: Climate device invalid/unknown preset_mode received! (ESPclimateaircoslaapkamer: none)
2025-07-15 20:37:57.360 Error: MQTT-HomeAssistant: Climate device invalid/unknown preset_mode received! (ESPclimateaircologeerkamer: none)`
I think (but i'm not sure) that is has to do with this preset? :
`[20:47:19][C][midea:620]:   [x] Supported presets:
[20:47:19][C][midea:622]:       - NONE
[20:47:19][C][midea:622]:       - BOOST
[20:47:19][C][midea:622]:       - ECO
[20:47:19][C][midea:622]:       - SLEEP`
Since i didnt update esphome for a long time and all midea's are using older version of 2024, i think it has to be in domoticz, but want to do the question first here before debugging to domoticz.
Anyone know what can cause this issue/error?

Thanks a lot!
Barry

### Which version of ESPHome has the issue?

2025.6.3

### What type of installation are you using?

Docker

### What platform are you using?

ESP8266

### Component causing the issue

Midea

### YAML Config

```yaml

```

### Anything in the logs that might be useful for us?

```txt

```

jesserockz, 2025-07-16:
`to_climate_preset` is called when the data comes in from the device/library to ESPHome. 

This is where ESPHome gets the current preset from, so looks like the underlying library will send none if none of the presets are actually set.
https://github.com/dudanov/MideaUART/blob/6cbde3802daf6cf9164478e0584c8e094c0c1396/src/Appliance/AirConditioner/StatusData.cpp#L62-L72

Does your device actually support having no preset? If so, maybe add `none` to the list you have in YAML?

```yaml
    supported_presets:          # All capabilities in this section detected by autoconf.
      - NONE
      - ECO
      - BOOST
      - SLEEP  
```

jesserockz, 2025-07-18:
Please try the linked PR as an external component. There is a comment in the PR on how to do that.

Barry40, 2025-07-20:
Thanks Jesse,

updated all (6) esphome's to latest dev with the external components, cleaned all mqtt (retained) messages and installed all new plugins and stuff, but still recieve the errors unfortunately.. :(
- NONE is added to presets as far as i can see, but still not fixed for me.

`2025-07-20 12:44:18.970 Error: MQTT-HomeAssistant: Climate device invalid/unknown preset_mode received! (ESPclimateaircokantoor: none)
2025-07-20 12:45:06.796 Error: MQTT-HomeAssistant: Climate device invalid/unknown preset_mode received! (ESPclimateaircoslaapkamer: none)
2025-07-20 12:45:09.667 Error: MQTT-HomeAssistant: Climate device invalid/unknown preset_mode received! (ESPclimateaircokantoor: none)
2025-07-20 12:45:11.314 Error: MQTT-HomeAssistant: Climate device invalid/unknown preset_mode received! (ESPclimateaircowoonkamer: none)`

Barry40, 2025-07-21:
I'm going deeper into this error message, and it seems that the preset none is not added in mqtt?
I can see all presets, but there is no NONE into the presets?
`{
  "curr_temp_t": "Inventor/aircoslaapkamer/climate/aircoslaapkamer/current_temperature/state",
  "mode_cmd_t": "Inventor/aircoslaapkamer/climate/aircoslaapkamer/mode/command",
  "mode_stat_t": "Inventor/aircoslaapkamer/climate/aircoslaapkamer/mode/state",
  "modes": [
    "off",
    "cool",
    "heat",
    "fan_only",
    "dry",
    "heat_cool"
  ],
  "temp_cmd_t": "Inventor/aircoslaapkamer/climate/aircoslaapkamer/target_temperature/command",
  "temp_stat_t": "Inventor/aircoslaapkamer/climate/aircoslaapkamer/target_temperature/state",
  "min_temp": 16,
  "max_temp": 30,
  "temp_step": 1,
  "precision": 1,
  "temp_unit": "C",
  "min_hum": 30,
  "max_hum": 99,
  "pr_mode_cmd_t": "Inventor/aircoslaapkamer/climate/aircoslaapkamer/preset/command",
  "pr_mode_stat_t": "Inventor/aircoslaapkamer/climate/aircoslaapkamer/preset/state",
  "preset_modes": [
    "boost",
    "eco",
    "sleep",
    "freeze protection"
  ],
  "fan_mode_cmd_t": "Inventor/aircoslaapkamer/climate/aircoslaapkamer/fan_mode/command",
  "fan_mode_stat_t": "Inventor/aircoslaapkamer/climate/aircoslaapkamer/fan_mode/state",
  "fan_modes": [
    "auto",
    "low",
    "medium",
    "high",
    "silent",
    "turbo"
  ],
  "swing_mode_cmd_t": "Inventor/aircoslaapkamer/climate/aircoslaapkamer/swing_mode/command",
  "swing_mode_stat_t": "Inventor/aircoslaapkamer/climate/aircoslaapkamer/swing_mode/state",
  "swing_modes": [
    "off",
    "both",
    "vertical",
    "horizontal"
  ],
  "name": "aircoslaapkamer",
  "avty_t": "Inventor/aircoslaapkamer/status",
  "uniq_id": "ESPclimateaircoslaapkamer",
  "dev": {
    "ids": "bcff4d8dd4f3",
    "name": "aircoslaapkamer",
    "sw": "2025.8.0-dev (Jul 20 2025, 20:49:37)",
    "mdl": "esp12e",
    "mf": "Espressif",
    "cns": [
      [
        "mac",
        "macxxx"
      ]
    ]
  }
}`

Can this be the problem of these error messages?
It drives me crazy...

jesserockz, 2025-07-22:
Did you read my comment on the PR? https://github.com/esphome/esphome/pull/9696#issuecomment-3094814117

Barry40, 2025-07-23:
Great job done, thanks Jesse.
Is it to me to open a bug request in Domoticz, or could you let them about this?

Thanks!