esphome/esphome #9549: Cannot disable web_server OTA

Jonybat, 2025-07-16:
### The problem

Updated some devices to 2025.7.0, and the OTA upload is now always visible in the webserver, even if not specified under `ota` and with `ota: false` under `web_server`

According to docs and PR (https://github.com/esphome/esphome/pull/9274), this is not expected

Tested with esp8266/d1_mini and esp32/esp32-c3-devkitm-1. Also tried with `version: 2` and `version: 3` of the `web_server`

[image]

### Which version of ESPHome has the issue?

2025.7.0

### What type of installation are you using?

pip

### What platform are you using?

ESP8266

### Component causing the issue

web_server

### YAML Config

```yaml
substitutions:
  device_name: garage01
  device_password: !secret garage01_password
  device_api_key: !secret garage01_api_key

esphome:
  name: ${device_name}
  name_add_mac_suffix: false

esp8266:
  board: d1_mini

# Enable logging
logger:
  level: DEBUG
  baud_rate: 115200

# Enable Home Assistant API
api:
  encryption:
    key: "${device_api_key}"

ota:
  - platform: esphome
    password: "${device_password}"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  fast_connect: true
  reboot_timeout: 10min

  manual_ip:
    static_ip: !secret garage01_static_ip
    gateway: !secret wifi_gateway
    subnet: !secret wifi_subnet
    dns1: !secret wifi_dns1
    dns2: !secret wifi_dns2

  ap:
    ssid: "esp-${device_name}"
    password: "${device_password}"

captive_portal:

web_server:
  port: 80
  version: 3
  local: true
  ota: false

### MAIN
binary_sensor:
  - platform: gpio
    pin:
      number: D4
      mode: INPUT_PULLUP
      inverted: False
    name: Front gate
    device_class: door
```

### Anything in the logs that might be useful for us?

```txt

```

bdraco, 2025-07-16:
captive_portal component auto-loads ota.web_server

bdraco, 2025-07-16:
tracking as https://github.com/esphome/esphome/issues/9565

closed 2025-07-16 by bdraco