esphome/esphome #9554: Compile time error for Captive_Portal in LibreTiny Device

jweston2112, 2025-07-16:
### The problem

during the compile an error is reported that emptyString is undefined

### Which version of ESPHome has the issue?

2025.7.0

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

BK72XX

### Component causing the issue

captive_portal

### YAML Config

```yaml
esphome:
  name: kmc-1
  friendly_name: KMC-1

bk72xx:
  board: generic-bk7231n-qfn32-tuya 
  framework:
    version: latest

logger:

wifi:
  ssid: !secret iot_wifi_ssid
  password: !secret iot_wifi_password
  ap:
    ssid: "KMC-1 Fallback Hotspot"
    password: !secret wifi_ap_password

web_server:
  port: 80
 
captive_portal:

mdns:

api:

ota:
  - platform: esphome
  - platform: web_server
    
button:
- platform: restart
  name: Restart

debug:
  update_interval: 30s

text_sensor:
- platform: debug
  reset_reason:
    name: Reset Reason
- platform: libretiny
  version:
    name: LibreTiny Version

sensor:
- platform: uptime
  name: Uptime

switch:
- platform: gpio
  id: switch_1
  name: Relay 1
  pin: P26
- platform: gpio
  id: switch_2
  name: Relay 2
  pin: P24
- platform: gpio
  id: switch_3
  name: Relay 3
  pin: P6

binary_sensor:
- platform: gpio
  id: binary_switch_all
  pin:
    number: P10
    inverted: true
    mode: INPUT_PULLUP
  on_press:
    then:
    - switch.toggle: switch_1
    - switch.toggle: switch_2
    - switch.toggle: switch_3

status_led:
  pin: P11
```

### Anything in the logs that might be useful for us?

```txt
Compiling .pioenvs/kmc-1/src/esphome/components/captive_portal/captive_portal.cpp.o
In file included from src/esphome/components/web_server_base/web_server_base.h:11,
                 from src/esphome/components/captive_portal/captive_portal.h:11,
                 from src/esphome/components/captive_portal/captive_portal.cpp:1:
.piolibdeps/kmc-1/ESPAsyncWebServer/src/ESPAsyncWebServer.h:463:75: error: 'emptyString' was not declared in this scope
  463 |     beginResponse(FS &fs, const String &path, const String &contentType = emptyString, bool download = false, AwsTemplateProcessor callback = nullptr) {
      |                                                                           ^~~~~~~~~~~
.piolibdeps/kmc-1/ESPAsyncWebServer/src/ESPAsyncWebServer.h:470:81: error: 'emptyString' was not declared in this scope
  470 |     beginResponse(File content, const String &path, const String &contentType = emptyString, bool download = false, AwsTemplateProcessor callback = nullptr) {
      |                                                                                 ^~~~~~~~~~~
.piolibdeps/kmc-1/ESPAsyncWebServer/src/ESPAsyncWebServer.h:684:77: error: 'emptyString' was not declared in this scope
  684 |   const String &getAttribute(const char *name, const String &defaultValue = emptyString) const;
      |                                                                             ^~~~~~~~~~~
.piolibdeps/kmc-1/ESPAsyncWebServer/src/ESPAsyncWebServer.h: In member function 'const arduino::String& AsyncWebServerRequest::arg(int) const':
.piolibdeps/kmc-1/ESPAsyncWebServer/src/ESPAsyncWebServer.h:575:20: error: 'emptyString' was not declared in this scope
  575 |     return i < 0 ? emptyString : arg((size_t)i);
      |                    ^~~~~~~~~~~
.piolibdeps/kmc-1/ESPAsyncWebServer/src/ESPAsyncWebServer.h: In member function 'const arduino::String& AsyncWebServerRequest::argName(int) const':
.piolibdeps/kmc-1/ESPAsyncWebServer/src/ESPAsyncWebServer.h:579:20: error: 'emptyString' was not declared in this scope
  579 |     return i < 0 ? emptyString : argName((size_t)i);
      |                    ^~~~~~~~~~~
.piolibdeps/kmc-1/ESPAsyncWebServer/src/ESPAsyncWebServer.h: In member function 'const arduino::String& AsyncWebServerRequest::pathArg(int) const':
.piolibdeps/kmc-1/ESPAsyncWebServer/src/ESPAsyncWebServer.h:591:20: error: 'emptyString' was not declared in this scope
  591 |     return i < 0 ? emptyString : pathArg((size_t)i);
      |                    ^~~~~~~~~~~
.piolibdeps/kmc-1/ESPAsyncWebServer/src/ESPAsyncWebServer.h: In member function 'const arduino::String& AsyncWebServerRequest::header(int) const':
.piolibdeps/kmc-1/ESPAsyncWebServer/src/ESPAsyncWebServer.h:606:20: error: 'emptyString' was not declared in this scope
  606 |     return i < 0 ? emptyString : header((size_t)i);
      |                    ^~~~~~~~~~~
.piolibdeps/kmc-1/ESPAsyncWebServer/src/ESPAsyncWebServer.h: In member function 'const arduino::String& AsyncWebServerRequest::headerName(int) const':
.piolibdeps/kmc-1/ESPAsyncWebServer/src/ESPAsyncWebServer.h:610:20: error: 'emptyString' was not declared in this scope
  610 |     return i < 0 ? emptyString : headerName((size_t)i);
      |                    ^~~~~~~~~~~
.piolibdeps/kmc-1/ESPAsyncWebServer/src/ESPAsyncWebServer.h: In member function 'void AsyncWebServerRequest::setAttribute(const char*, bool)':
.piolibdeps/kmc-1/ESPAsyncWebServer/src/ESPAsyncWebServer.h:668:39: error: 'emptyString' was not declared in this scope
  668 |     _attributes[name] = value ? "1" : emptyString;
      |                                       ^~~~~~~~~~~
In file included from .piolibdeps/kmc-1/ESPAsyncWebServer/src/ESPAsyncWebServer.h:1232,
                 from src/esphome/components/web_server_base/web_server_base.h:11,
                 from src/esphome/components/captive_portal/captive_portal.h:11,
                 from src/esphome/components/captive_portal/captive_portal.cpp:1:
.piolibdeps/kmc-1/ESPAsyncWebServer/src/WebResponseImpl.h: At global scope:
.piolibdeps/kmc-1/ESPAsyncWebServer/src/WebResponseImpl.h:25:83: error: 'emptyString' was not declared in this scope
   25 |   AsyncBasicResponse(int code, const String &contentType, const String &content = emptyString)
      |                                                                                   ^~~~~~~~~~~
Compiling .pioenvs/kmc-1/src/esphome/components/debug/debug_component.cpp.o
Compiling .pioenvs/kmc-1/src/esphome/components/debug/debug_esp32.cpp.o
*** [.pioenvs/kmc-1/src/esphome/components/captive_portal/captive_portal.cpp.o] Error 1
========================= [FAILED] Took 15.19 seconds =========================
```

### Additional information

[logs_kmc-1_run.txt](https://github.com/user-attachments/files/21259526/logs_kmc-1_run.txt)

ssieb, 2025-07-17:
I can't reproduce this.  Try a clean build.

ssieb, 2025-07-17:
Then try re-installing the add-on.

jweston2112, 2025-07-17:
Well I copied the yaml to my beta HA instance where it had never been compiled and did a clean build files just for completeness sake (it also validates the yaml) and it compiled so I guess you were right .... but I usually do a clean build files if I get a compiler error. and try again ... but hey I am 60 years old... i dont remember if I did or not....

DJBenson, 2025-07-17:
> Oh, wait. You're overriding the framework version. That's not supported. Sorry, that's the original poster. You didn't show your config.

Turned out to be the same issue - it was overriding the framework to "latest" - has built fine for a long time but broke with the latest release. Not overriding the framework version allowed it to build. Thanks for the inadvertent pointer.