esphome/esphome #9555: Can't OTA any devices (2025.7) The program size (1970436 bytes) is greater than maximum allowed (1835008 bytes)

broyuken, 2025-07-16:
### The problem

It seems like esphome put on a bunch of weight with this release, I can't ota update any of my nodes anymore because the file size is too big. This is anything from a device just running the ble proxy to light strips, to controlling my standing desk, etc. They're all too large. 

```
RAM:   [==        ]  18.9% (used 62064 bytes from 327680 bytes)
Flash: [==========]  107.4% (used 1970436 bytes from 1835008 bytes)
Error: The program size (1970436 bytes) is greater than maximum allowed (1835008 bytes)
*** [checkprogsize] Explicit exit, status 1
========================== [FAILED] Took 7.59 seconds ==========================
```

### Which version of ESPHome has the issue?

2025.7

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

### YAML Config

```yaml
substitutions:
  name: justins-room-mirror-lightstrip
  friendly_name: Justins Room Mirror lightstrip

esphome:
  name: ${name}
  friendly_name: ${friendly_name}
  min_version: 2024.6.0
  name_add_mac_suffix: false
  project:
    name: esphome.web
    version: '1.0'

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "xxxxxxxxxxxxxxxxxxxxxxxxxxxx"

ota:
  - platform: esphome

# Enable Bluetooth Proxy
bluetooth_proxy:
  active: true

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Esphome-Web-1Fd1C4"
    password: "xxxxxxxxxxxxxxxxxxxxxxxxxxxx"

captive_portal:

# LED Strip
light:
  - platform: rgb
    name: "Justins Light Strip"
    red: gpio_red
    green: gpio_green
    blue: gpio_blue

# LED Strip Output Pins
output:
  - platform: ledc
    pin: GPIO19
    id: gpio_red
    inverted: true

  - platform: ledc
    pin: GPIO18
    id: gpio_green
    inverted: true

  - platform: ledc
    pin: GPIO21
    id: gpio_blue
    inverted: true

switch:
  - platform: restart
    name: Restart ${friendly_name}
```

### Anything in the logs that might be useful for us?

```txt
INFO ESPHome 2025.7.0
INFO Reading configuration /config/esphome/justins-room-mirror-lightstrip.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing justins-room-mirror-lightstrip (board: esp32dev; framework: arduino; platform: https://github.com/pioarduino/platform-espressif32/releases/download/53.03.13/platform-espressif32.zip)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - framework-arduinoespressif32 @ 3.1.3 
 - framework-arduinoespressif32-libs @ 5.3.0+sha.489d7a2b3a 
 - tool-esptoolpy @ 4.8.6 
 - tool-mklittlefs @ 3.2.0 
 - tool-riscv32-esp-elf-gdb @ 14.2.0+20240403 
 - tool-xtensa-esp-elf-gdb @ 14.2.0+20240403 
 - toolchain-xtensa-esp-elf @ 13.2.0+20240530
Dependency Graph
|-- Networking @ 3.1.3
|-- AsyncTCP @ 3.4.5
|-- WiFi @ 3.1.3
|-- FS @ 3.1.3
|-- Update @ 3.1.3
|-- ESPAsyncWebServer @ 3.7.10
|-- ESP32 Async UDP @ 3.1.3
|-- DNSServer @ 3.1.3
|-- ESPmDNS @ 3.1.3
|-- noise-c @ 0.1.10
Error: The program size (1970436 bytes) is greater than maximum allowed (1835008 bytes)
RAM:   [==        ]  18.9% (used 62064 bytes from 327680 bytes)
Flash: [==========]  107.4% (used 1970436 bytes from 1835008 bytes)
*** [checkprogsize] Explicit exit, status 1
========================== [FAILED] Took 7.51 seconds ==========================
```

### Additional information

Installs just fine on 2025.6.3

Idanit, 2025-07-16:
> type: esp-idf

It helped... TYVM

broyuken, 2025-07-16:
seems to have worked for me as well. 3 fixed, 2 to go.

andreas-bulling, 2025-07-16:
Can this easily be changed and updated OTA or does the device have to be flashed directly (i.e. with a cable)?

closed 2025-07-17 by jesserockz