esphome/esphome #9556: Not able to install on esp32

lahayeg, 2025-07-16:
### The problem

Installing new version off ESPHome on esp32 stop after unpacking builder. See logs

### Which version of ESPHome has the issue?

2025.7.0

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

### YAML Config

```yaml
substitutions:
  devicename: iot12-buanderie
  
esphome:
  name: $devicename
esp32:
  board: mhetesp32devkit
#  project:
#    name: esphome.bluetooth-proxy
 #   version: "1.0"

wifi:
  ssid: "x"
  password: "x"
  
  manual_ip:
    static_ip: 192.168.x.x
    gateway: 192.168.x.x
    subnet: 255.255.255.0

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: ${devicename} Fallback Hotspot
    password: "x"

captive_portal:

# Enable logging
logger:
  level: DEBUG

# Enable Home Assistant API
api:

web_server:
  port: 80

ota:
  - platform: esphome
  - platform: web_server

esp32_ble_tracker:
#  scan_parameters:
 #   interval: 1100ms
  #  window: 1100ms
   # active: true

bluetooth_proxy:

sensor:
  - platform: wifi_signal
    name: ${devicename} WiFi Signal Sensor
    update_interval: 60s
    
button:
- platform: restart
  name: ${devicename} Restart
- platform: safe_mode
  name: ${devicename} Safe Mode Boot
  entity_category: diagnostic
```

### Anything in the logs that might be useful for us?

```txt
INFO ESPHome 2025.7.0
INFO Reading configuration /config/esphome/iot12-buanderie.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing iot12-buanderie (board: mhetesp32devkit; framework: arduino; platform: https://github.com/pioarduino/platform-espressif32/releases/download/53.03.13/platform-espressif32.zip)
--------------------------------------------------------------------------------
Tool Manager: Installing https://github.com/espressif/esp32-arduino-lib-builder/releases/download/idf-release_v5.3/esp32-arduino-libs-idf-release_v5.3-489d7a2b-v1.zip
INFO Installing https://github.com/espressif/esp32-arduino-lib-builder/releases/download/idf-release_v5.3/esp32-arduino-libs-idf-release_v5.3-489d7a2b-v1.zip
Unpacking  [####################################]  100%
```

adgud, 2025-07-17:
Same issue started to come up today for me. ESPHome Device Builder 2025.7.0, installation to device gets stuck at either "unpacking" stage as in OP's case or at this point:

```
INFO ESPHome 2025.7.0
INFO Reading configuration /config/esphome/esp-nawadnianie.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing esp-nawadnianie (board: esp32-c3-devkitm-1; framework: arduino; platform: https://github.com/pioarduino/platform-espressif32/releases/download/53.03.13/platform-espressif32.zip)
--------------------------------------------------------------------------------
```

Builds get stuck even on configurations that used to compile and install without problems.

lahayeg, 2025-07-17:
@jesserockz  @adgud tried twice to uninstall + reinstall latest version 2025.7.1 but still stucked after unpacking 100%

[logs_iot12-buanderie_run (1).txt](https://github.com/user-attachments/files/21296630/logs_iot12-buanderie_run.1.txt)

jesserockz, 2025-07-18:
Is your storage full in HA?

I have no idea why platformio gets stuck at unpacking, and there is not a lot that ESPHome itself can do in these cases at this stage.

ssieb, 2025-07-18:
How are you accessing HA?  Is it going through the proxy?

lahayeg, 2025-07-18:
> Is your storage full in HA?
> 
> I have no idea why platformio gets stuck at unpacking, and there is not a lot that ESPHome itself can do in these cases at this stage.

300 Gb free... I understand your comment.

lahayeg, 2025-07-18:
> How are you accessing HA?  Is it going through the proxy?

At work yes but just try this morning from home. No proxy and same behaviour

bdraco, 2025-07-18:
Can you get an strace of the process in the container?

Either via [port 22222](https://developers.home-assistant.io/docs/operating-system/debugging/) or the community ssh addon with protection mode disabled 

`docker exec -it <container> /bin/bash` (`docker container ls` to find the ESPHome container you want to enter)

`ps uxawww | grep esphome`

`apt update`

`apt install strace`

`strace -p <pid of esphome> -f -s 4096`

lahayeg, 2025-07-18:
@bdraco I didn't tried your solution because I managed to solve it :

I launch the update through HA settings -> update. After several minutes, HA became unavailable. I add to power restart it. Then, go on ESPHome add-on and install again, the dependencies were already installed and was able to start compilation (still on going)

I will closed it because seems edge case

nxor, 2025-07-23:
I still have the issue... what you mentionned doesnt look like a fix to me... 
I uninstalled the addon and reinstalled same issue (uninstalling cleared 10GB of junk)