esphome/esphome #9566: Compilation issue with LibreTiny on docker image

lamauny, 2025-07-16:
### The problem

captive_portal do not compile on LibreTiny platforms with 2025.7.0 and 2025.8.0-dev docker images

### Which version of ESPHome has the issue?

2025.7.0

### What type of installation are you using?

Docker

### What platform are you using?

BK72XX

### Component causing the issue

captive_portal

### YAML Config

```yaml
esphome:
  name: bk

bk72xx:
  board: wb3s

wifi:
  ssid: MySSID
  password: password1

captive_portal:
```

### Anything in the logs that might be useful for us?

```txt
INFO ESPHome 2025.7.0
INFO Reading configuration test2.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing bk (board: wb3s; framework: arduino; platform: libretiny @ 1.9.1)
----------------------------------------------------------------------------------------------------------------------
LibreTiny: Creating a virtual environment for Python dependencies
LibreTiny: Installing Python dependencies
Collecting ltchiptool<5.0,>=4.12.0
  Downloading ltchiptool-4.12.0-py3-none-any.whl.metadata (4.2 kB)
Collecting bitstruct<9.0.0,>=8.1.1 (from ltchiptool<5.0,>=4.12.0)
  Downloading bitstruct-8.21.0-cp312-cp312-manylinux_2_17_x86_64.manylinux2014_x86_64.whl.metadata (7.3 kB)
Collecting bk7231tools<3.0.0,>=2.0.0 (from ltchiptool<5.0,>=4.12.0)
  Downloading bk7231tools-2.1.0-py3-none-any.whl.metadata (841 bytes)
Collecting click<9.0.0,>=8.1.3 (from ltchiptool<5.0,>=4.12.0)
  Downloading click-8.2.1-py3-none-any.whl.metadata (2.5 kB)
Collecting colorama<0.5.0,>=0.4.5 (from ltchiptool<5.0,>=4.12.0)
  Downloading colorama-0.4.6-py2.py3-none-any.whl.metadata (17 kB)
Collecting hexdump<4.0,>=3.3 (from ltchiptool<5.0,>=4.12.0)
  Downloading hexdump-3.3.zip (12 kB)
  Installing build dependencies: started
  Installing build dependencies: finished with status 'done'
  Getting requirements to build wheel: started
  Getting requirements to build wheel: finished with status 'done'
  Preparing metadata (pyproject.toml): started
  Preparing metadata (pyproject.toml): finished with status 'done'
Collecting importlib-metadata (from ltchiptool<5.0,>=4.12.0)
  Downloading importlib_metadata-8.7.0-py3-none-any.whl.metadata (4.8 kB)
Collecting prettytable<4.0.0,>=3.3.0 (from ltchiptool<5.0,>=4.12.0)
  Downloading prettytable-3.16.0-py3-none-any.whl.metadata (33 kB)
Collecting py-datastruct<2.0.0,>=1.0.0 (from ltchiptool<5.0,>=4.12.0)
[... 138 lines cut ...]
/root/.platformio/lib/RPAsyncTCP/src/async_config.h:9:25: error: 'MBED_CONF_LWIP_TCP_MSS' was not declared in this scope
    9 |   #define TCP_MSS       MBED_CONF_LWIP_TCP_MSS        //(1460)
      |                         ^~~~~~~~~~~~~~~~~~~~~~
/root/.platformio/lib/RPAsyncTCP/src/SyncClient.h:41:55: note: in expansion of macro 'TCP_MSS'
   41 |     SyncClient(AsyncClient *client, size_t txBufLen = TCP_MSS);
      |                                                       ^~~~~~~
Compiling .pioenvs/bk/core_common_arduino_libraries/api/WiFi/WiFiEvents.cpp.o
Compiling .pioenvs/bk/core_common_arduino_libraries/api/WiFi/WiFiGeneric.cpp.o
Compiling .pioenvs/bk/core_common_arduino_libraries/api/WiFi/WiFiSTA.cpp.o
Compiling .pioenvs/bk/core_common_arduino_libraries/api/WiFi/WiFiScan.cpp.o
Compiling .pioenvs/bk/core_common_arduino_libraries/common/FS/FS.cpp.o
*** [.pioenvs/bk/lib46e/RPAsyncTCP/SyncClient.cpp.o] Error 1
============================================ [FAILED] Took 13.38 seconds ============================================
```

### Additional information

Tested for these HW platforms :
- bk72xx : fails
- rtl87xx : fails
- esp32 : success

Same behaviour with docker image on linux PC and Home Assistant add-on on HASSOS (amd64)

Not tested on a Python native install but seems to work : @bdraco [#7230 [captive_portal] Broken on libretiny](https://github.com/esphome/issues/issues/7230)

bdraco, 2025-07-16:
tried the HA addon and it compiles there fine as well
```
INFO ESPHome 2025.7.0
INFO Reading configuration /config/esphome/issue9566.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing bk (board: wb3s; framework: arduino; platform: libretiny @ 1.9.1)
--------------------------------------------------------------------------------
Platform Manager: Installing libretiny @ 1.9.1
INFO Installing libretiny @ 1.9.1
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Platform Manager: libretiny@1.9.1 has been installed!
INFO libretiny@1.9.1 has been installed!
Tool Manager: Installing git+https://github.com/libretiny-eu/framework-beken-bdk#v2021.06.07
INFO Installing git+https://github.com/libretiny-eu/framework-beken-bdk#v2021.06.07
git version 2.39.5
Cloning into '/data/cache/platformio/cache/tmp/pkg-installing-e5w6g78_'...
Note: switching to '6491b8c001b737c790cba0bee16e548859d5e6fe'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by switching back to a branch.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -c with the switch command. Example:

  git switch -c <new-branch-name>

Or undo this operation with:

  git switch -

Turn off this advice by setting config variable advice.detachedHead to false

Tool Manager: framework-beken-bdk@0.0.0+v2021.06.07.sha.6491b8c has been installed!
INFO framework-beken-bdk@0.0.0+v2021.06.07.sha.6491b8c has been installed!
Tool Manager: Installing git+https://github.com/libretiny-eu/ArduinoCore-API#2022.08.24
INFO Installing git+https://github.com/libretiny-eu/ArduinoCore-API#2022.08.24
git version 2.39.5
Cloning into '/data/cache/platformio/cache/tmp/pkg-installing-9f34nmya'...
Note: switching to '237b10a4f6da14bf32e5e0963b7f28865216a60c'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by switching back to a branch.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -c with the switch command. Example:

  git switch -c <new-branch-name>

Or undo this operation with:

  git switch -

Turn off this advice by setting config variable advice.detachedHead to false

Tool Manager: framework-arduino-api@2022.8.24+sha.237b10a has been installed!
INFO framework-arduino-api@2022.8.24+sha.237b10a has been installed!
Tool Manager: Installing git+https://github.com/libretiny-eu/lwip#2.2.0-bdk
INFO Installing git+https://github.com/libretiny-eu/lwip#2.2.0-bdk
git version 2.39.5
Cloning into '/data/cache/platformio/cache/tmp/pkg-installing-4jcy0hls'...
Tool Manager: library-lwip@2.2.0-bdk+sha.48da4a2 has been installed!
INFO library-lwip@2.2.0-bdk+sha.48da4a2 has been installed!
Tool Manager: Installing git+https://github.com/libretiny-eu/library-freertos#9.0.0
INFO Installing git+https://github.com/libretiny-eu/library-freertos#9.0.0
git version 2.39.5
Cloning into '/data/cache/platformio/cache/tmp/pkg-installing-31e143in'...
[... 733 lines cut ...]
Compiling .pioenvs/bk/bdk_core/func/wlan_ui/wlan_ui.c.o
Compiling .pioenvs/bk/bdk_boot/boot_handlers_bk7231u.S.o
Archiving .pioenvs/bk/libbdk_boot.a
Archiving .pioenvs/bk/libbdk_driver.a
Archiving .pioenvs/bk/libbdk_app.a
Archiving .pioenvs/bk/libbdk_core.a
Linking .pioenvs/bk/raw_firmware.elf
|-- Image 1: raw_firmware.elf
|   |-- image_bk7231t_app.0x011000.rbl
|   |   |-- raw_firmware.bin
|   |-- image_bk7231t_app.0x011000.crc
|   |-- image_bk7231t_app.0x129F0A.rblh
|   |-- image_bk7231t_app.ota.rbl
|   |-- image_bk7231t_app.ota.ug.bin
RAM:   [====      ]  44.3% (used 116044 bytes from 262144 bytes)
Flash: [=====     ]  50.7% (used 549601 bytes from 1083136 bytes)
Building UF2 OTA image
|-- esphome_2025.7.0_wb3s_bk7231t_lt1.9.1.uf2
|-- firmware.uf2
|-- firmware.bin
========================= [SUCCESS] Took 69.49 seconds =========================
Warning! Non-Git installations are NOT SUPPORTED.
INFO Successfully compiled program.
ERROR Cannot upload Over the Air as the ota configuration is not present or does not include platform: esphome
```

szupi-ipuzs, 2025-07-16:
@lamauny 
Usually, in cases of weird compiler errors I remove the whole esphome cache (/config/.esphome). Very often it helps. Of course it needs to re-download all the packages afterwards.

skooj, 2025-07-16:
> [@lamauny](https://github.com/lamauny) Usually, in cases of weird compiler errors I remove the whole esphome cache (/config/.esphome). Very often it helps. Of course it needs to re-download all the packages afterwards.

I gave this a shot and it did not solve the problem for me.

FPA-ChrisLaird, 2025-07-17:
A quick work around that worked for me.

  platformio_options:
    lib_ignore:
      - RPAsyncTCP

ssieb, 2025-07-17:
@DJBenson that's a different issue: https://github.com/esphome/esphome/issues/9554

ssieb, 2025-07-17:
I have a local git install and clearing the platformio cache fixed it for me.

DJBenson, 2025-07-17:
> @DJBenson that's a different issue: https://github.com/esphome/esphome/issues/9554

Copied it over to the other issue. Apologies.

leodrivera, 2025-08-07:
I've tried with 2025.7.5, but I am still getting the same error.

Tested with 2 different devices running generic-bk7231n-qfn32-tuya

FPA-ChrisLaird, 2025-08-07:
> I've tried with 2025.7.5, but I am still getting the same error.
> 
> Tested with 2 different devices running generic-bk7231n-qfn32-tuya
> 
> ```
> Compiling .pioenvs/luz-cozinha/src/main.cpp.o

> Installed versions: platform: libretiny @ 1.9.1 ESP32Async/AsyncTCP @ 3.4.5 ESP32Async/ESPAsyncWebServer @ 3.7.10

Yes, the fix hasn't been rolled in yet.  I recall reading that it would be included in the Aug 11th release.

radsonpatrick, 2025-08-31:
> > A quick work around that worked for me.
> > 
> > platformio_options: lib_ignore: - RPAsyncTCP
> 
> Will this still allow the use of the webserver?

I work for me !

tyzen9, 2025-09-03:
I can verify adding this to my YAML config "stopped" the build issue, and still permitted the web server.  
However, I do not know what the side effect will be of leaving this ignore in place, and recompiling 20+ TreatLife Tuya based switches.  I think I will wait for a fix to be released.

```
esphome:
  name: $device_name
  friendly_name: $device_friendly_name ($device_location_descriptor)
  comment: $device_make $device_model $device_type

  # -- This is a temp fix to avoid this issue: https://github.com/esphome/esphome/issues/9566
  platformio_options:
    lib_ignore: RPAsyncTCP
```

rosstroha, 2025-09-03:
> A quick work around that worked for me.
> 
> platformio_options: lib_ignore: - RPAsyncTCP

This might be obvious to some, but for me it wasn't. Cleaning the build files is seemlingly required for this to work.

immortalserg, 2025-09-05:
bk7231n/esp8286 config:

esphome:
  platformio_options:
    lib_ignore:
      - RPAsyncTCP

rosstroha, 2025-09-05:
> Even if I just enter this, I still get the error I posted two messages
> above. The build files probably need to be deleted, but I don't know how.
> Any suggestions?

In esphome builder the option is at the main page under each devices
context menu