esphome/esphome #9568: Error - assert failed: udp_new_ip_type /IDF/components/lwip/lwip/src/core/udp.c:1277 (Required to lock TCPIP core functionality!)

gtjadsonsantos, 2025-07-16:
### The problem

[image]

### Which version of ESPHome has the issue?

2025.7.0

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

### Component causing the issue

https://github.com/espressif/esp-lwip

### YAML Config

```yaml

```

### Anything in the logs that might be useful for us?

```txt

```

gtjadsonsantos, 2025-07-16:
```
[16:59:51][W][component:279]: mqtt set Warning flag: unspecified

[16:59:51]assert failed: udp_new_ip_type /IDF/components/lwip/lwip/src/core/udp.c:1277 (Required to lock TCPIP core functionality!)

[16:59:51]Backtrace: 0x400836ed:0x3ffb1c60 0x4008affd:0x3ffb1c80 0x400912fe:0x3ffb1ca0 0x4011afde:0x3ffb1dd0 0x401138fc:0x3ffb1df0 0x400e5baf:0x3ffb1e30 0x400e5c45:0x3ffb1e80 0x400e67be:0x3ffb1ea0 0x40179af9:0x3ffb1ef0 0x40179bc1:0x3ffb1f10 0x400eb7e2:0x3ffb1f30 0x400f2ee9:0x3ffb1f70 0x400fdc13:0x3ffb2270 0x4008ba36:0x3ffb2290
WARNING Found stack trace! Trying to decode it
WARNING Decoded 0x400836ed: panic_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/panic.c:463
WARNING Decoded 0x4008affd: esp_system_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/port/esp_system_chip.c:92
WARNING Decoded 0x400912fe: __assert_func at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/assert.c:80
WARNING Decoded 0x4011afde: udp_new_ip_type at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/udp.c:1277 (discriminator 1)
WARNING Decoded 0x401138fc: dns_alloc_random_port at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/dns.c:959
 (inlined by) dns_alloc_pcb at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/dns.c:999
 (inlined by) dns_enqueue at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/dns.c:1613
 (inlined by) dns_gethostbyname_addrtype at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/dns.c:1791
WARNING Decoded 0x400e5baf: esphome::mqtt::MQTTClientComponent::start_dnslookup_() at /Users/gtjadsonsantos/Repositorys/github/firmwares/becocastelo/.esphome/build/flexip-training/src/esphome/components/mqtt/mqtt_client.cpp:200 (discriminator 1)
WARNING Decoded 0x400e5c45: esphome::mqtt::MQTTClientComponent::enable() at /Users/gtjadsonsantos/Repositorys/github/firmwares/becocastelo/.esphome/build/flexip-training/src/esphome/components/mqtt/mqtt_client.cpp:532
 (inlined by) esphome::mqtt::MQTTClientComponent::enable() at /Users/gtjadsonsantos/Repositorys/github/firmwares/becocastelo/.esphome/build/flexip-training/src/esphome/components/mqtt/mqtt_client.cpp:526
WARNING Decoded 0x400e67be: esphome::mqtt::MQTTClientComponent::setup() at /Users/gtjadsonsantos/Repositorys/github/firmwares/becocastelo/.esphome/build/flexip-training/src/esphome/components/mqtt/mqtt_client.cpp:84
WARNING Decoded 0x40179af9: esphome::Component::call_setup() at /Users/gtjadsonsantos/Repositorys/github/firmwares/becocastelo/.esphome/build/flexip-training/src/esphome/core/component.cpp:127
WARNING Decoded 0x40179bc1: esphome::Component::call() at /Users/gtjadsonsantos/Repositorys/github/firmwares/becocastelo/.esphome/build/flexip-training/src/esphome/core/component.cpp:163
WARNING Decoded 0x400eb7e2: esphome::Application::setup() at /Users/gtjadsonsantos/Repositorys/github/firmwares/becocastelo/.esphome/build/flexip-training/src/esphome/core/application.cpp:60
WARNING Decoded 0x400f2ee9: setup() at /Users/gtjadsonsantos/Repositorys/github/firmwares/becocastelo/.esphome/build/flexip-training/src/main.cpp:3425
WARNING Decoded 0x400fdc13: loopTask(void*) at /Users/gtjadsonsantos/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:59
WARNING Decoded 0x4008ba36: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:139
```

bdraco, 2025-07-16:
Looks like a thread safety problem in ethernet

bdraco, 2025-07-16:
I think this needs the lock https://github.com/esphome/esphome/blob/c93b892ccc71aa6fb2794fbf2f08556339ed7f3b/esphome/components/mqtt/mqtt_client.cpp#L197

closed 2025-07-17 by jesserockz