esphome/esphome #9572: compilation error with LN-02

wiltet, 2025-07-16:
### The problem

With the new ESPHome Builder version 7.0 on HA came the LN882 libretiny support and I tried to compile a basic downloadable image to flash a LN-02 board with itchiptool.

The process ends with a error. |-- Generated file not found: .pioenvs/new1/raw_firmware.elf
Compiling with other board types with other MCUs works.

Don't know what's wrong.

### Which version of ESPHome has the issue?

2025.7.0

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

LN882X

### YAML Config

```yaml
esphome:
  name: new1
  friendly_name: new1

ln882x:
  board: ln-02

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "xxx"

ota:
  - platform: esphome
    password: !secret ota_password

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "New1 Fallback Hotspot"
    password: !secret wifi_ap_password

captive_portal:
```

### Anything in the logs that might be useful for us?

```txt
INFO ESPHome 2025.7.0
INFO Reading configuration /config/esphome/new1.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing new1 (board: ln-02; framework: arduino; platform: libretiny @ 1.9.1)
--------------------------------------------------------------------------------
Library Manager: Installing ESP32Async/AsyncTCP @ 3.4.5
INFO Installing ESP32Async/AsyncTCP @ 3.4.5
Unpacking  [####################################]  100%
Library Manager: AsyncTCP@3.4.5 has been installed!
INFO AsyncTCP@3.4.5 has been installed!
Library Manager: Installing ESP32Async/ESPAsyncWebServer @ 3.7.10
[... 617 lines cut ...]
Compiling .pioenvs/new1/ln882h_sdk/mcu/driver_ln882h/hal/hal_qspi.c.o
Compiling .pioenvs/new1/ln882h_sdk/mcu/driver_ln882h/hal/hal_rtc.c.o
Compiling .pioenvs/new1/ln882h_sdk/mcu/driver_ln882h/hal/hal_sdio_device.c.o
Compiling .pioenvs/new1/ln882h_sdk/mcu/driver_ln882h/hal/hal_sdio_room.c.o
Compiling .pioenvs/new1/ln882h_sdk/mcu/driver_ln882h/hal/hal_spi.c.o
Compiling .pioenvs/new1/ln882h_sdk/mcu/driver_ln882h/hal/hal_timer.c.o
Compiling .pioenvs/new1/ln882h_sdk/mcu/driver_ln882h/hal/hal_trng.c.o
Compiling .pioenvs/new1/ln882h_sdk/mcu/driver_ln882h/hal/hal_uart.c.o
Compiling .pioenvs/new1/ln882h_sdk/mcu/driver_ln882h/hal/hal_wdt.c.o
Compiling .pioenvs/new1/ln882h_sdk/mcu/driver_ln882h/hal/hal_ws2811.c.o
Compiling .pioenvs/new1/ln882h_sdk/mcu/ln882h/system_ln882h.c.o
Archiving .pioenvs/new1/libln882h_sdk.a
Linking .pioenvs/new1/raw_firmware.elf
|-- Image 1: raw_firmware.elf
/data/cache/platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld: .pioenvs/new1/libln882h_sdk.a(cm_backtrace.c.o): in function `get_cur_thread_stack_info':
/data/cache/platformio/packages/framework-lightning-ln882h/components/utils/debug/CmBacktrace/cm_backtrace.c:211: undefined reference to `vTaskStackAddr'
/data/cache/platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld: /data/cache/platformio/packages/framework-lightning-ln882h/components/utils/debug/CmBacktrace/cm_backtrace.c:212: undefined reference to `vTaskStackSize'
/data/cache/platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld: .pioenvs/new1/libln882h_sdk.a(cm_backtrace.c.o): in function `get_cur_thread_name':
/data/cache/platformio/packages/framework-lightning-ln882h/components/utils/debug/CmBacktrace/cm_backtrace.c:244: undefined reference to `vTaskName'
/data/cache/platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld: /data/cache/platformio/packages/framework-lightning-ln882h/components/utils/debug/CmBacktrace/cm_backtrace.c:244: undefined reference to `vTaskName'
collect2: error: ld returned 1 exit status
|-- Generated file not found: .pioenvs/new1/raw_firmware.elf
*** [.pioenvs/new1/raw_firmware.elf] Error 1
========================= [FAILED] Took 317.54 seconds =========================
```

bdraco, 2025-07-16:
links locally with the above config on dev

bdraco, 2025-07-16:
also on using Home Assistant addon to compile
```
INFO ESPHome 2025.7.0
INFO Reading configuration /config/esphome/ln02.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing new1 (board: ln-02; framework: arduino; platform: libretiny @ 1.9.1)
--------------------------------------------------------------------------------
Tool Manager: Installing git+https://github.com/libretiny-eu/framework-lightning-ln882h
INFO Installing git+https://github.com/libretiny-eu/framework-lightning-ln882h
git version 2.39.5
Cloning into '/data/cache/platformio/cache/tmp/pkg-installing-kskmqina'...
Tool Manager: framework-lightning-ln882h@0.0.0+v2025.01.18.sha.8794085 has been installed!
INFO framework-lightning-ln882h@0.0.0+v2025.01.18.sha.8794085 has been installed!
Tool Manager: Installing git+https://github.com/libretiny-eu/lwip#2.1.3-ln882h
INFO Installing git+https://github.com/libretiny-eu/lwip#2.1.3-ln882h
git version 2.39.5
Cloning into '/data/cache/platformio/cache/tmp/pkg-installing-dcrj8eth'...
Tool Manager: library-lwip@2.1.3-ln882h+sha.7ef5aac has been installed!
INFO library-lwip@2.1.3-ln882h+sha.7ef5aac has been installed!
Library Manager: Installing ESP32Async/AsyncTCP @ 3.4.5
INFO Installing ESP32Async/AsyncTCP @ 3.4.5
Unpacking  [####################################]  100%
Library Manager: AsyncTCP@3.4.5 has been installed!
INFO AsyncTCP@3.4.5 has been installed!
Library Manager: Installing ESP32Async/ESPAsyncWebServer @ 3.7.10
INFO Installing ESP32Async/ESPAsyncWebServer @ 3.7.10
Unpacking  [####################################]  100%
Library Manager: ESPAsyncWebServer@3.7.10 has been installed!
INFO ESPAsyncWebServer@3.7.10 has been installed!
Library Manager: Resolving dependencies...
INFO Resolving dependencies...
Warning! Non-Git installations are NOT SUPPORTED.
HARDWARE: LN882HK 160MHz, 295KB RAM, 1.17MB Flash
 - framework-arduino-api @ 2022.8.24+sha.237b10a 
 - framework-lightning-ln882h @ 0.0.0+v2025.01.18.sha.8794085 
 - library-flashdb @ 1.2.0+sha.d5c892f 
 - library-freertos @ 9.0.0+sha.ff31b7d 
 - library-freertos-port @ 2025.1.18+sha.dbfc408 
 - library-lwip @ 2.1.3-ln882h+sha.7ef5aac 
 - library-printf @ 6.1.0+sha.28a79bd 
 - library-uf2ota @ 5.0.0+sha.f955412 
PLATFORM VERSIONS:
 - libretiny @ 1.9.1
 - ltchiptool @ 4.12.0
CUSTOM OPTIONS:
 - fw_name = esphome
 - fw_version = 2025.7.0
Library Manager: Installing DNSServer
INFO:Library Manager:Installing DNSServer
Unpacking 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%
Library Manager: DNSServer@1.1.0 has been installed!
INFO:Library Manager:DNSServer@1.1.0 has been installed!
Dependency Graph
|-- AsyncTCP @ 3.4.5
|-- ESPAsyncWebServer @ 3.7.10
|-- DNSServer @ 1.1.0
Compiling .pioenvs/new1/src/esphome/components/api/api_connection.cpp.o
Compiling .pioenvs/new1/src/esphome/components/api/api_frame_helper.cpp.o
Compiling .pioenvs/new1/src/esphome/components/api/api_pb2.cpp.o
Compiling .pioenvs/new1/src/esphome/components/api/api_pb2_service.cpp.o
Compiling .pioenvs/new1/src/esphome/components/api/api_server.cpp.o
Compiling .pioenvs/new1/src/esphome/components/api/list_entities.cpp.o
Compiling .pioenvs/new1/src/esphome/components/api/proto.cpp.o
Compiling .pioenvs/new1/src/esphome/components/api/subscribe_state.cpp.o
Compiling .pioenvs/new1/src/esphome/components/captive_portal/captive_portal.cpp.o
Compiling .pioenvs/new1/src/esphome/components/esphome/ota/ota_esphome.cpp.o
Compiling .pioenvs/new1/src/esphome/components/libretiny/core.cpp.o
Compiling .pioenvs/new1/src/esphome/components/libretiny/gpio_arduino.cpp.o
Compiling .pioenvs/new1/src/esphome/components/libretiny/helpers.cpp.o
[... 498 lines cut ...]
Compiling .pioenvs/new1/ln882h_sdk/mcu/driver_ln882h/hal/hal_spi.c.o
Compiling .pioenvs/new1/ln882h_sdk/mcu/driver_ln882h/hal/hal_timer.c.o
Compiling .pioenvs/new1/ln882h_sdk/mcu/driver_ln882h/hal/hal_trng.c.o
Compiling .pioenvs/new1/ln882h_sdk/mcu/driver_ln882h/hal/hal_uart.c.o
Compiling .pioenvs/new1/ln882h_sdk/mcu/driver_ln882h/hal/hal_wdt.c.o
Compiling .pioenvs/new1/ln882h_sdk/mcu/driver_ln882h/hal/hal_ws2811.c.o
Compiling .pioenvs/new1/ln882h_sdk/mcu/ln882h/system_ln882h.c.o
Archiving .pioenvs/new1/libln882h_sdk.a
Linking .pioenvs/new1/raw_firmware.elf
|-- Image 1: raw_firmware.elf
|   |-- image_firmware.0x000000.bin
|   |   |-- raw_firmware.bin
PASS -- checking .pioenvs/new1/boot.bin
PASS -- checking .pioenvs/new1/raw_firmware.bin
RAM:   [====      ]  37.7% (used 113888 bytes from 302080 bytes)
Flash: [====      ]  38.4% (used 472176 bytes from 1228800 bytes)
Building UF2 OTA image
|-- esphome_2025.7.0_ln-02_ln882hk_lt1.9.1.uf2
|-- firmware.uf2
|-- firmware.bin
========================= [SUCCESS] Took 34.51 seconds =========================
Warning! Non-Git installations are NOT SUPPORTED.
INFO Successfully compiled program.
INFO Resolving IP address of new1.local in mDNS
```

wiltet, 2025-07-17:
> Did you try clean build files ?

yes, and more then one time.

> [@wiltet](https://github.com/wiltet) can you try again with this patch?

sorry, but I don't know how to. have to check this out first and have a try.

...with the "file editor" addon I can not see this folder or I just didn't find it.
I have HAsso installed on a RPi4.

wiltet, 2025-07-17:
When I searched the detached system drive I found this file only under `/hassos-data/supervisor/addons/data/5c53de3b_esphome/cache/platformio/platforms/libretiny/builder/family/lightning-ln882h.py` and that isn't it.
Sorry but I don't understand the HAsso docker filesystem and don't want to mess it up.

lamauny, 2025-07-17:
@wiltet your yaml also compiles on my 2025.7.0 HA Add-on.
FREERTOS_PORT_LIGHTNING_LN882H is auto defined by LibreTiny.

lamauny, 2025-07-17:
@wiltet Don't be defeatist, wait a few days, maybe someone will find a solution.

can you give more information about your HASSOS environment : platform, version, etc... ?

ssieb, 2025-07-17:
> When I searched the detached system drive I found this file only under `/hassos-data/supervisor/addons/data/5c53de3b_esphome/cache/platformio/platforms/libretiny/builder/family/lightning-ln882h.py` [...]

That's the right file.
Try re-installing the add-on.

wiltet, 2025-07-17:
Okay, now I removed the addon with all its datas, rebootet and reinstalled it.
LN882x still not selectable...   aaand...   SUCCESS!!! 

[log.txt](https://github.com/user-attachments/files/21305142/log.txt)

ssieb, 2025-07-17:
You could file an issue to get the `LN882X` option added.

ssieb, 2025-07-17:
File it in the dashboard issues.  I see it's available in the cli wizard.

closed 2025-07-17 by ssieb