esphome/esphome #9592: Compile error in void setup() no matching function for call to 'esphome::Automation<int, const char*, const char*>::Automation(esphome::logger::LoggerMessageTrigger*&)'

jandechent, 2025-07-17:
### The problem

My esphome was working, after upgrade to 2025.7, I now get (on one of 5 esphome devices)  this error
```
Archiving .pioenvs/heizung/lib64d/libWiFi.a
Compiling .pioenvs/heizung/libbc6/Update/Updater.cpp.o
src/main.cpp: In function 'void setup()':
src/main.cpp:256:97: error: no matching function for call to 'esphome::Automation<int, const char*, const char*>::Automation(esphome::logger::LoggerMessageTrigger*&)'
  256 |   automation_id = new Automation<int, const char *, const char *>(logger_loggermessagetrigger_id);
      |                                                                                                 ^
In file included from src/esphome/components/binary_sensor/filter.h:3,
                 from src/esphome/components/binary_sensor/binary_sensor.h:5,
                 from src/esphome/core/application.h:26,
                 from src/esphome/components/api/api_frame_helper.h:17,
                 from src/esphome/components/api/api_connection.h:5,
                 from src/esphome.h:3,
                 from src/main.cpp:3:
src/esphome/core/automation.h:286:12: note: candidate: 'esphome::Automation<Ts>::Automation(esphome::Trigger<Ts ...>*) [with Ts = {int, const char*, const char*}]'
  286 |   explicit Automation(Trigger<Ts...> *trigger) : trigger_(trigger) { this->trigger_->set_automation_parent(this); }
      |            ^~~~~~~~~~
src/esphome/core/automation.h:286:39: note:   no known conversion for argument 1 from 'esphome::logger::LoggerMessageTrigger*' to 'esphome::Trigger<int, const char*, const char*>*'
  286 |   explicit Automation(Trigger<Ts...> *trigger) : trigger_(trigger) { this->trigger_->set_automation_parent(this); }
      |                       ~~~~~~~~~~~~~~~~^~~~~~~
src/esphome/core/automation.h:284:32: note: candidate: 'constexpr esphome::Automation<int, const char*, const char*>::Automation(const esphome::Automation<int, const char*, const char*>&)'
  284 | template<typename... Ts> class Automation {
      |                                ^~~~~~~~~~
src/esphome/core/automation.h:284:32: note:   no known conversion for argument 1 from 'esphome::logger::LoggerMessageTrigger*' to 'const esphome::Automation<int, const char*, const char*>&'
src/esphome/core/automation.h:284:32: note: candidate: 'constexpr esphome::Automation<int, const char*, const char*>::Automation(esphome::Automation<int, const char*, const char*>&&)'
src/esphome/core/automation.h:284:32: note:   no known conversion for argument 1 from 'esphome::logger::LoggerMessageTrigger*' to 'esphome::Automation<int, const char*, const char*>&&'
Compiling .pioenvs/heizung/lib942/ESPAsyncWebServer/AsyncEventSource.cpp.o
Compiling .pioenvs/heizung/lib942/ESPAsyncWebServer/AsyncJson.cpp.o
Compiling .pioenvs/heizung/lib942/ESPAsyncWebServer/AsyncMessagePack.cpp.o
Compiling .pioenvs/heizung/lib942/ESPAsyncWebServer/AsyncWebHeader.cpp.o
Compiling .pioenvs/heizung/lib942/ESPAsyncWebServer/AsyncWebServerRequest.cpp.o
Archiving .pioenvs/heizung/libbc6/libUpdate.a
Compiling .pioenvs/heizung/lib942/ESPAsyncWebServer/AsyncWebSocket.cpp.o
*** [.pioenvs/heizung/src/main.cpp.o] Error 1
```

### Which version of ESPHome has the issue?

2025.7.0

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

### YAML Config

```yaml
#
#     ┌───────────┐
#     │  HEIZUNG  │                                              ┌─Y─ Ground
#     |           |                                              ├─2─ Nullleiter───────────────   to USB GND
#     │    L,N,PE ├3─────────────────────────────────────────────┴─1─ Phase 230V─────────────┬─   to USB 5V
#     │      Heat ├────────────────────────────────────────────────4─ relais channel 1 ──────┤    Wärmeforderung 
#     │       Umw ├────────────────────────────────────────────────3─ relais channel 3 ──────┤    Umwälzen
#     │         N ├────────────────────────────────────┬──┐                                  │
#     └───╥───╥───┘    ╔══════════════════════════════════┴────────5─ relais channel 4 ──────┤    Boiler Ventil 
#    Rückl║   ║Vorl    ║    ┌────────┐──dallas_temp    │              0x307673201864ff28     │    Boiler Temperatur
#         ║   ╠═══════VNT═══│ BOILER │════════════╗    │                                     │
#         ║   ║             └─╥────╥─┘            ║    │                                     │
#         ║   ║               ╚ZIRK╝              ║    │N                                    │
#         ║   ║                 ╚════════════════ ║ ═══┴───────────6─ relais channel 2 ──────┘    Zirkulieren   
#         ║   ║                                   ║
#         ║   ║dalles_temp                        ║                   0x14b84d201864ff28          Vorlauf Heizungen
[... 545 lines cut ...]
              value: !lambda |-
                return std::max({
                  id(vorlauf_max).state,
                  id(vorlauf_heizungen).state
                });
          - number.set:
              id: rucklauf_max
              value: !lambda |-
                return std::max({
                  id(rucklauf_max).state,
                  id(rucklauf_heizung_eingang).state, 
                  id(rucklauf_heizung_garten).state, 
                  id(rucklauf_wasser).state
                });
          - logger.log:
                      level: INFO
                      format: "Rücklauf Max: %.1f"
                      args: [ 'id(rucklauf_max).state' ]
```

### Anything in the logs that might be useful for us?

```txt

```

ssieb, 2025-07-18:
It's the `on_message:` trigger in the logger.

jb1228, 2025-07-18:
> It's the `on_message:` trigger in the logger.

Confirmed. Removing this trigger did it.

I swear I saw a PR that fixed this, which is why I waited until 2025.7.1; but I could be wrong.

closed 2025-07-18 by bdraco