esphome/esphome #9599: Restarting a script no longer works properly.

ThomasMatern, 2025-07-17:
### The problem

I have two ESPhome devices which get receive an external "ping" (API call) in regular intervals, and take an action if the ping was not received for some time. This is implemented via a 'timeout' script, which waits for the timeout and then takes the action. The ping restarts the script, avoiding the timeout-script from ever making it past the initial wait statement.
The above works in 2025.6.3 and does not work in 2025.7.0 - restarting a script (with mode: restart) does not seem to restart the script, instead it seems that for some time, two (or more) instances of the script run in parallel.

### Which version of ESPHome has the issue?

2025.7.0

### What type of installation are you using?

Docker

### What platform are you using?

ESP8266

### Component causing the issue

script

### YAML Config

```yaml
# Sample config
api:
  services:
    - service: launch
      then:
        - logger.log: "Launching launcher..."
        - script.execute: launcher

script:
  - id: launcher
    mode: restart
    then:
      - logger.log: "Launcher: start timeout handler"
      - script.execute: timeout_handler
      - delay: 6s
      - logger.log: "Launcher: restart timeout handler"
      - script.execute: timeout_handler
      - delay: 6s
      - logger.log: "Launcher: restart timeout handler again"
      - script.execute: timeout_handler

  - id: timeout_handler
    mode: restart
    then:
      - logger.log: "Timeout handler: started"
      - delay: 10s
      - logger.log: "Timeout handler: still running after 10 seconds"
      - delay: 10s
      - logger.log: "Timeout handler: still running after 20 seconds"
```

### Anything in the logs that might be useful for us?

```txt
In ESPhome 2025.6.3, the above script produces this output:

[19:48:54][D][main:185]: Launching launcher...
[19:48:54][D][main:287]: Launcher: start timeout handler
[19:48:54][D][main:312]: Timeout handler: started
[19:49:00][D][main:296]: Launcher: restart timeout handler
[19:49:00][D][script:077]: Script 'timeout_handler' restarting (mode: restart)
[19:49:00][D][main:312]: Timeout handler: started
[19:49:06][D][main:305]: Launcher: restart timeout handler again
[19:49:06][D][script:077]: Script 'timeout_handler' restarting (mode: restart)
[19:49:06][D][main:312]: Timeout handler: started
[19:49:16][D][main:319]: Timeout handler: still running after 10 seconds
[19:49:26][D][main:326]: Timeout handler: still running after 20 seconds

... which is correct. The timeout-script is restarted and only reaches the "still running" part after the launcher stops restarting the script.

In 2025.7.0, the same script produces this output:
[20:38:51][D][main:189]: Launching launcher...
[20:38:51][D][main:291]: Launcher: start timeout handler
[20:38:51][D][main:316]: Timeout handler: started
[20:38:57][D][main:300]: Launcher: restart timeout handler
[20:38:57][D][script:077]: Script 'timeout_handler' restarting (mode: restart)
[20:38:57][D][main:316]: Timeout handler: started
[20:39:01][D][main:323]: Timeout handler: still running after 10 seconds
[20:39:03][D][main:309]: Launcher: restart timeout handler again
[20:39:03][D][script:077]: Script 'timeout_handler' restarting (mode: restart)
[20:39:03][D][main:316]: Timeout handler: started
[20:39:07][D][main:323]: Timeout handler: still running after 10 seconds
[20:39:11][D][main:330]: Timeout handler: still running after 20 seconds

... which is wrong. The "still running" message appears 10s after the script was started, and only 4s after it was restarted, so it appears as if the script was not restarted. However, 10s after the restart, another "still running" message appears, so it appears as if the script runs in parallel? But then the final "still running after 20s" message appears only once, which means that it did not run in parallel.
```

Sleeper85, 2025-07-17:
Same issue observed, worked fine before esphome `2025.7`.

ssieb, 2025-07-17:
Yes, we know it's an issue.  It will be fixed for the next point release.

bdraco, 2025-07-17:
I'll work on a fix after lunch

bdraco, 2025-07-18:
even with the above fix I'm still seeing it execute
```yaml
esphome:
  name: test-delay-action

host:
api:
  actions:
    - action: start_delay_then_restart
      then:
        - logger.log: "Starting first script execution"
        - script.execute: test_delay_script
        - delay: 500ms  # Give first script time to start delay
        - logger.log: "Restarting script (should cancel first delay)"
        - script.execute: test_delay_script

logger:
  level: DEBUG

script:
  - id: test_delay_script
    mode: restart
    then:
      - logger.log: "Script started, beginning delay"
      - delay: 5s  # Long enough that it won't complete before restart
      - logger.log: "ERROR: First delay completed - this should NOT happen with restart mode!"
```

bdraco, 2025-07-18:
nevermind, I forgot to reinstall.. long day.. fix works

closed 2025-07-18 by bdraco

ThomasMatern, 2025-07-18:
Thanks guys for the quick fix.

Sleeper85, 2025-07-18:
@bdraco many thanks 😉

PR #9646 (2025-07-18): [scheduler] Fix DelayAction cancellation in restart mode scripts
# What does this implement/fix?

This PR fixes DelayAction cancellation when used within scripts that have `mode: restart`. Previously, when a script with a delay action was restarted, the delay from the first execution would continue running even though the script had been restarted, causing both delays to complete.

The issue was that DelayAction was using an unnamed timeout (`""`) which couldn't be properly cancelled because it conflicted with the scheduler's handling of empty string names vs nullptr.  https://github.com/espho [... cut]