esphome/esphome #9601: ESP32-Cam http does not work after update

DerEmder, 2025-07-17:
### The problem

Installationsmethode Home Assistant OS
Core 2025.7.2
Supervisor 2025.07.1
Operating System 16.0
Frontend 20250702.2

I installed the yaml config with arduino and tried esp-idf as well. All versions of ESPHome before 2025.5.x worked almost perfectly. After the update neither the cam nor the attached LD2410 was working. With the IDF framework the HA API could not even connect to ESPHome on that board.

I ordered a new board wich arrived yetserday and, in accordance with one of the devs I will try to find the problem step-by-step.

This is the yaml config I tried in this first step.

```
esphome:
  name: max_test                        # neuer Name ⇒ keine HA-Kollisionen

esp32:
  board: esp32cam                       # gleiche HW
  framework:
    type: esp-idf                       # Framework beibehalten
  flash_size: 4MB

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  power_save_mode: none
  manual_ip:                             # freie IP wählen
    static_ip: 10.42.0.179
    gateway: 10.42.0.1
    subnet: 255.255.255.0
    dns1: 10.42.0.1

logger:
  level: VERY_VERBOSE                   # komplette Ausgabe
  baud_rate: 115200                     # USB-Log

debug:                                  # Heap/Stack alle 10 s
  update_interval: 10s

api:                                    # ohne Verschlüsselung → Fehler klarer
ota:                                    # OTA drin lassen, schadet nicht
```

Log:

```
INFO ESPHome 2025.6.3
INFO Reading configuration /config/esphome/max.yaml...
WARNING 'max_test': Using the '_' (underscore) character in the hostname is discouraged as it can cause problems with some DHCP and local name services. For more information, see https://esphome.io/guides/faq.html#why-shouldn-t-i-use-underscores-in-my-device-name
INFO Starting log output from 10.42.0.179 using esphome API
INFO Successfully resolved max_test @ 10.42.0.179 in 0.000s
INFO Successfully connected to max_test @ 10.42.0.179 in 0.049s
INFO Successful handshake with max_test @ 10.42.0.179 in 0.081s
[18:30:29][I][app:137]: ESPHome version 2025.6.3 compiled on Jul 12 2025, 12:57:37
[18:30:29][C][wifi:613]: WiFi:
[18:30:29][C][wifi:434]:   Local MAC: 
[18:30:29][C][wifi:439]:   SSID: 
[18:30:29][C][wifi:442]:   IP Address: 10.42.0.179
[18:30:29][C][wifi:446]:   BSSID: 
[18:30:29][C][wifi:446]:   Hostname: 'max_test'
[18:30:29][C][wifi:446]:   Signal strength: -69 dB ▂▄▆█
[18:30:29][V][wifi:453]:   Priority: 0.0
[18:30:29][C][wifi:455]:   Channel: 11
[18:30:29][C][wifi:455]:   Subnet: 255.255.255.0
[18:30:29][C][wifi:455]:   Gateway: 10.42.0.1
[18:30:29][C][wifi:455]:   DNS1: 10.42.0.1
[18:30:29][C][wifi:455]:   DNS2: 0.0.0.0
[18:30:29][C][logger:211]: Logger:
[18:30:29][C][logger:211]:   Max Level: VERY_VERBOSE
[18:30:29][C][logger:211]:   Initial Level: VERY_VERBOSE
[18:30:29][C][logger:217]:   Log Baud Rate: 115200
[18:30:29][C][logger:217]:   Hardware UART: UART0
[18:30:29][C][logger:224]:   Task Log Buffer Size: 768
[18:30:29][C][esphome.ota:073]: Over-The-Air updates:
[18:30:29][C][esphome.ota:073]:   Address: 10.42.0.179:3232
[18:30:29][C][esphome.ota:073]:   Version: 2
[18:30:29][C][safe_mode:018]: Safe Mode:
[18:30:29][C][safe_mode:019]:   Boot considered successful after 60 seconds
[18:30:29][C][safe_mode:019]:   Invoke after 10 boot attempts
[18:30:29][C][safe_mode:019]:   Remain for 300 seconds
[18:30:29][C][api:182]: API Server:
[18:30:29][C][api:182]:   Address: 10.42.0.179:6053
[18:30:29][C][api:192]:   Using noise encryption: NO
[18:30:29][C][mdns:122]: mDNS:
[18:30:29][C][mdns:122]:   Hostname: max_test
[18:30:29][V][mdns:126]:   Services:
[18:30:29][V][mdns:128]:   - _esphomelib, _tcp, 6053
[18:30:29][V][mdns:131]:     TXT: version = 2025.6.3
[18:30:29][V][mdns:131]:     TXT: mac = 6825dd2df358
[18:30:29][V][mdns:131]:     TXT: platform = ESP32
[18:30:29][V][mdns:131]:     TXT: board = esp32cam
[18:30:29][V][mdns:131]:     TXT: network = wifi
[18:30:29][C][debug:018]: Debug component:
[18:30:29][D][debug:033]: ESPHome version 2025.6.3
[18:30:29][D][debug:037]: Free Heap Size: 232760 bytes
[18:30:29][D][debug:180]: Chip: Model=ESP32, Features=2.4GHz WiFi, BLE, BT,  Cores=2, Revision=301
[18:30:29][D][debug:189]: CPU Frequency: 160 MHz
[18:30:29][D][debug:197]: Framework: ESP-IDF
[18:30:29][D][debug:204]: ESP-IDF Version: 5.3.2
[18:30:29][D][debug:209]: EFuse MAC: 
[18:30:29][D][debug:077]: Reset Reason: power-on event
[18:30:29][D][debug:105]: Wakeup Reason: undefined
[18:30:29][C][debug:110]: Partition table:
[18:30:29][C][debug:110]:   Name         Type Subtype  Address    Size      
[18:30:29][C][debug:117]:   otadata      1    0        0x00009000 0x00002000
[18:30:29][C][debug:117]:   phy_init     1    1        0x0000B000 0x00001000
[18:30:29][C][debug:117]:   app0         0    16       0x00010000 0x001C0000
[18:30:29][C][debug:117]:   app1         0    17       0x001D0000 0x001C0000
[18:30:29][C][debug:117]:   nvs          1    2        0x00390000 0x0006D000
[18:31:09][VV][api.service:059]: on_ping_request: PingRequest {}
[18:31:09][VV][api.service:013]: send_message ping_response: PingResponse {}
[18:31:09][I][safe_mode:042]: Boot seems successful; resetting boot loop counter
[18:31:09][VV][esp32.preferences:040]: s_pending_save: key: 233825507, len: 4
[18:31:10][V][esp32.preferences:114]: Saving 1 items...
[18:31:10][VV][esp32.preferences:123]: Checking if NVS data 233825507 has changed
[18:31:10][V][esp32.preferences:126]: sync: key: 233825507, len: 4
[18:31:10][D][esp32.preferences:142]: Writing 1 items: 0 cached, 1 written, 0 failed

```

### Which version of ESPHome has the issue?

2025.5.x and higher

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

### YAML Config

```yaml

```

### Anything in the logs that might be useful for us?

```txt

```

DerEmder, 2025-07-17:
Out of boredome yesterday evening I added the esp32cam to the above yaml and flashed it. Worked! Yay.

Log:

```
[21:57:32]ets Jul 29 2019 12:21:46
[21:57:32]
[21:58:00]rst:0x1 (POWERON_RESET),boot:0x13 (Sets Jul 29 2019 12:21:46
[21:58:00]
[21:58:00]rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[21:58:00]configsip: 0, SPIWP:0xee
[21:58:00]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[21:58:00]mode:DIO, clock div:2
[21:58:00]load:0x3fff0030,len:7176
[21:58:00]load:0x40078000,len:15796
[21:58:00]load:0x40080400,len:4
[21:58:00]load:0x40080404,len:3924
[21:58:00]entry 0x40080638
[21:58:00]I (29) boot: ESP-IDF 5.3.2 2nd stage bootloader
[21:58:00]I (29) boot: compile time Jul 16 2025 21:54:49
[21:58:00]I (29) boot: Multicore bootloader
[21:58:00]I (34) boot: chip revision: v3.1
[21:58:00]I (37) boot.esp32: SPI Speed      : 40MHz
[21:58:00]I (42) boot.esp32: SPI Mode       : DIO
[21:58:00]I (46) boot.esp32: SPI Flash Size : 4MB
[21:58:00]I (51) boot: Enabling RNG early entropy source...
[21:58:00]I (56) boot: Partition Table:
[21:58:00]I (60) boot: ## Label            Usage          Type ST Offset   Length
[21:58:00]I (67) boot:  0 otadata          OTA data         01 00 00009000 00002000
[21:58:00]I (75) boot:  1 phy_init         RF data          01 01 0000b000 00001000
[21:58:00]I (82) boot:  2 app0             OTA app          00 10 00010000 001c0000
[21:58:00]I (90) boot:  3 app1             OTA app          00 11 001d0000 001c0000
[21:58:00]I (97) boot:  4 nvs              WiFi data        01 02 00390000 0006d000
[21:58:00]I (105) boot: End of partition table
[21:58:00]I (109) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=236d4h (145108) map
[21:58:00]I (169) esp_image: segment 1: paddr=000336fc vaddr=3ffb0000 size=04fd4h ( 20436) load
[21:58:00]I (177) esp_image: segment 2: paddr=000386d8 vaddr=40080000 size=07940h ( 31040) load
[21:58:00]I (189) esp_image: segment 3: paddr=00040020 vaddr=400d0020 size=9fe94h (654996) map
[21:58:00]I (420) esp_image: segment 4: paddr=000dfebc vaddr=40087940 size=129fch ( 76284) load
[21:58:00]I (464) boot: Loaded app from partition at offset 0x10000
[21:58:00]I (464) boot: Disabling RNG early entropy source...
[21:58:00]I (476) quad_psram: This chip is ESP32-D0WD
[21:58:00]I (476) esp_psram: Found 8MB PSRAM device
[21:58:00]I (477) esp_psram: Speed: 40MHz
[21:58:00]I (480) esp_psram: PSRAM initialized, cache is in low/high (2-core) mode.
[21:58:00]W (487) esp_psram: Virtual address not enough for PSRAM, map as much as we can. 4MB is mapped
[21:58:00]I (496) cpu_start: Multicore app
[21:58:01]I (1381) esp_psram: SPI SRAM memory test OK
[21:58:01]I (1389) cpu_start: Pro cpu start user code
[21:58:01]I (1389) cpu_start: cpu freq: 160000000 Hz
[21:58:01]I (1389) app_init: Application information:
[21:58:01]I (1392) app_init: Project name:     max_test
[21:58:01]I (1397) app_init: App version:      2025.6.3
[21:58:01]I (1403) app_init: Compile time:     Jul 16 2025 21:51:35
[21:58:01]I (1409) app_init: ELF file SHA256:  b8c6c1ebe...
[21:58:01]I (1414) app_init: ESP-IDF:          5.3.2
[21:58:01]I (1419) efuse_init: Min chip rev:     v0.0
[21:58:01]I (1424) efuse_init: Max chip rev:     v3.99 
[21:58:01]I (1429) efuse_init: Chip rev:         v3.1
[21:58:01]I (1434) heap_init: Initializing. RAM available for dynamic allocation:
[21:58:01]I (1441) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
[21:58:01]I (1447) heap_init: At 3FFB9758 len 000268A8 (154 KiB): DRAM
[21:58:01]I (1453) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
[21:58:01]I (1460) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
[21:58:01]I (1466) heap_init: At 4009A33C len 00005CC4 (23 KiB): IRAM
[21:58:01]I (1473) esp_psram: Adding pool of 4096K of PSRAM memory to heap allocator
[21:58:01]I (1481) spi_flash: detected chip: generic
[21:58:01]I (1485) spi_flash: flash io: dio
[21:58:01]W (1489) i2c: This driver is an old driver, please migrate your application code to adapt `driver/i2c_master.h`
[21:58:01]I (1501) main_task: Started on CPU0
[... 319 lines cut ...]
[21:58:06][D][debug:197]: Framework: ESP-IDF
[21:58:06][D][debug:204]: ESP-IDF Version: 5.3.2
[21:58:06][D][debug:209]: EFuse MAC: 68:25:DD:2D:F3:58
[21:58:06][D][debug:077]: Reset Reason: power-on event
[21:58:06][D][debug:105]: Wakeup Reason: undefined
[21:58:06][C][debug:110]: Partition table:
[21:58:06][C][debug:110]:  Name         Type Subtype  Address    Size      
[21:58:06][C][debug:117]:   otadata      1    0        0x00009000 0x00002000
[21:58:06][C][debug:117]:   phy_init     1    1        0x0000B000 0x00001000
[21:58:06][C][debug:117]:   app0         0    16       0x00010000 0x001C0000
[21:58:06][C][debug:117]:   app1         0    17       0x001D0000 0x001C0000
[21:58:06][C][debug:117]:   nvs          1    2        0x00390000 0x0006D000
[21:58:21][D][esp32_camera:201]: Got Image: len=7323
[21:58:41][D][esp32_camera:201]: Got Image: len=8932
[21:58:42][VV][esp-idf:000][wifi]: I (41988) wifi:
[21:58:42][VV][esp-idf:000][wifi]: <ba-add>idx:0 (ifx:0, 24:65:11:3c:5c:96), tid:0, ssn:0, winSize:64
[21:58:42][VV][esp-idf:000][wifi]: 
[21:59:01][D][esp32_camera:201]: Got Image: len=8996
[21:59:01][I][safe_mode:042]: Boot seems successful; resetting boot loop counter
[21:59:01][VV][esp32.preferences:040]: s_pending_save: key: 233825507, len: 4
[21:59:01][V][esp32.preferences:114]: Saving 1 items...
[21:59:01][VV][esp32.preferences:123]: Checking if NVS data 233825507 has changed
[21:59:01][V][esp32.preferences:126]: sync: key: 233825507, len: 4
[21:59:01][D][esp32.preferences:142]: Writing 1 items: 0 cached, 1 written, 0 failed
```

jesserockz, 2025-07-17:
I don't understand your issue. Your original config has almost nothing in it and the logs reflect that correctly.

Your comment above shows you added the camera and the led, but you didnt post your config and didnt really say what part was not working.

RubenKelevra, 2025-07-17:
@DerEmder well, sorry to be so direct, but this bug report is a bit counter productive. Great that your new device is working now, but maybe just write me an email so we can discuss how to proceed from here? 🤔

Because a bug report is only good if it's either actionable, meaning you know what needs to be fixed, or if you provide enough data so a developer can take a look at the issue and try to reason why it might happen.

As far as I can tell your bugreport does not fit in either category and now just references a different, very convoluted issue which is not helpful for anyone to make sense of this.

If you need help debugging, just hit me up via mail.

If you progress further, adding incrementally small sections of your setup and see something which starts to be not working, create a bug report with this small thing you modified which made it stop working (and mention me).

I suggest closing this bug report, and I'll look at the current status of the docs and the code regarding the i2c conflict later, to submit an PR based on the identified issue. 

Hope this helps :)

DerEmder, 2025-07-17:
Hi Ruben,

that is exactly what I wanted to do, step by step from I²C to UART. But ok, today I updated to 2025.7.1 which will take appr. 3 hours from now on. I will close this here and let you know how it went. And only THEN I will submit a bug or issue if something does **not** work. cu around and thanks for the help.

PR #9615 (2025-07-18): esp32_camera: deprecate i2c_pins; throw error if combined with i2c: block
# What does this implement/fix?

If `i2c_pins:` is used in combination with an `i2c:` block, the i2c module does not properly iterate to the next hardware. Leading to both components trying to use the same hardware with different GPIO pins.

This change will make it invalid to use `i2c_pins:` in combination with an `i2c:` block and instead guide the user to use the new `i2c_id:` option instead, which uses the i2c component by reference.

In addition a warning will be printed, if `i2c_pins:` is used reminding the existing [... cut]
closed 2025-07-17 by DerEmder