esphome/esphome #9608: Error compiling, to_string in 2025.7.1

ssunny1081, 2025-07-17:
### The problem

Also on 2025.7.1, esphome::to_string.. nor std::to_string is compiling anymore.

Please check, thanks

### Which version of ESPHome has the issue?

2025.7.1

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP8266

### Component causing the issue

template

### YAML Config

```yaml
- platform: template
...
switch (id(esp32_sprinkler_ctrlr).controller_state()) {
          case 0:
            return std::to_string("IDLE");
...
```

### Anything in the logs that might be useful for us?

```txt
Error: no matching function for call to 'to_string(const char [5])'
  705 |             return std::to_string("IDLE");
```

jesserockz, 2025-07-17:
This overload was removed from ESPHome in 2025.7.0 and never should have existed in the first place.

But why are you trying to convert a string to a string?

You should use `return {"IDLE"};`

closed 2025-07-17 by jesserockz

ssunny1081, 2025-07-17:
Thank you, will try this. 

Just lack of knowledge, took it from some guide and adapted to my needs. 

Now you wrote it, seems so obvious that it makes no sense..