esphome/esphome #9613: PMSX003 sensor stops working after upgrade to ESPHome 2025.7.x with logger baud_rate: 0

DrBlokmeister, 2025-07-17:
### The problem

After an upgrade to ESPHome 2025.7.0 (and .1), I'm no longer getting any updates from my PMSX003 sensors. A minimal version of my config is below. Six boards are affected by this issue.

Compilation and OTA works fine. I also see the setup of the PMSX platform in the logs. However, there are no sensor updates from my PMSX sensor present in the logs (even in verbose mode), and no sensor updates in ESPHome.

If I roll back to ESPHome 2025.6.1, and it works fine again.

I have asked ChatGPT and it told me that it has to do with the new Arduino version does not automatically disable UART logging anymore when setting the baud rate to 0 (I don't have enough knowledge here to judge whether this is correct reasoning or not). I have tried moving the UART logger to UART1 or UART2, but this gives the same issue. 

### Which version of ESPHome has the issue?

2025.7.X

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

### Component causing the issue

pmsx003

### YAML Config

```yaml
substitutions:
  wifi_ssid: "YOUR_SSID"
  wifi_password: "YOUR_PASSWORD"

esphome:
  name: pmsx003-test

esp32:
  board: wemos_d1_mini32          # also fails on six other identical boards
  framework:
    type: arduino                # ← works on 2025.6.1 (Arduino 2.0.5),
                                 #   stops working on 2025.7.x (Arduino 3.1.x)

# Disable serial logging (worked fine for years on ≤2025.6.x)
logger:
  level: DEBUG
  baud_rate: 0                   # ← crucial trigger

wifi:
  ssid: "${wifi_ssid}"
  password: "${wifi_password}"

uart:
  id: pms
  rx_pin: GPIO18
  tx_pin: GPIO19
  baud_rate: 9600

sensor:
  - platform: pmsx003
    type: PMSX003
    uart_id: pms
    pm_2_5:
      name: "PM2.5"
```

### Anything in the logs that might be useful for us?

```txt

```

jesserockz, 2025-07-18:
Why do you disable the serial logging in the first place?

An ESP32 can have 3 UARTs, with the first one being the logger, the pms/uart bus will automatically use the second UART.

We really need logs from the device, although since this component doesnt mark as failed, please enable the [uart debugger](https://esphome.io/components/uart.html#debugging) and post those logs here while it is running.

DZ8888, 2025-07-18:
same issue here! 
saw a message in log as below.
[16:03:52][W][pmsx003:138]: Payload length 4 doesn't match. Are you using the correct PMSX003 type?

paisitw, 2025-07-20:
Same issue. Encountered "[pmsx003:125 Start character 1 mismatch: 0x00 != 0x42" to all of my PMS5003 and 7003 sensors. Any workaround?

jesserockz, 2025-07-20:
As per my above comment @DZ8888, @paisitw please enable the uart debugger and post the logs when there is incoming data and an error. And paste your own configs YAML too. 

**Please edit your comment instead of making a new one too**

DZ8888, 2025-07-23:
There are some command response mixed to data buffer. they make Packet Validation Failures!
following items might be the problem from my observfation
1. ESPhome 2025.7.0 enhance power management introduced more frequent sleep/wake commands
2. or/and passive mode implementation added manual measurement triggers
3. or/and timing changes in command/response handling
4. Original code wasn't robust enough to handle the increased command traffic

but the above is just my observation for reference. still need Jesse's big hand to solve the issue. let my
pmsx003 return back! Thanks!

BTW, I filter the cammand response, the data can be read normally. of course, the filter is just a experiment. not solution.

DrBlokmeister, 2025-08-10:
Has there been any updates to this? Has anyone tried updating to the latest ESPHome version and testing if this has been fixed?

paisitw, 2025-08-10:
My workaround is to downgrade only ESP that connect PMS5003 to 2025.3.3 and it works fine.

DrBlokmeister, 2025-11-14:
Do not close this issue yet. It is unsure if this issue has been fixed or not.

DrBlokmeister, 2025-11-18:
> We really need logs from the device, although since this component doesnt mark as failed serial logs are not specifically required unless the uart itself is failing to set up, please enable the [uart debugger](https://esphome.io/components/uart.html#debugging) and post those logs here while it is running.

@jesserockz Can you remove the Needs More Information label? UART logs have already been added to the original post a long time ago.

stefesciberras, 2026-01-01:
I am also having issues - I have not used the sensor for a year or so, but now on starting it again with previous configuration, I am not getting any outputs. Fan is running, but the esp does not report any readings except for WIFI.

How can I roll back to 2025.6 ESPhome?