esphome/esphome #9621: Modbus Controller: Wrong Error Code

carlessole, 2025-07-17:
### The problem

I am testing Modbus and Modbus Controller components:

- https://esphome.io/components/modbus
- https://esphome.io/components/modbus_controller.html

I have set a device as a Server.
My current YAML configuration regarding Modbus is:

``` YAML
uart:
  - id: uart_modbus_server_1
    tx_pin: 23
    rx_pin: 22
    baud_rate: 115200

modbus:
  - id: modbus1
    uart_id: uart_modbus_server_1
    flow_control_pin: 18
    role: server

modbus_controller:
  - id: modbus_controller_1
    modbus_id: modbus1
    address: 20
    server_registers:
      - address: 0x0001
        value_type: U_WORD
        read_lambda: |-
          return 0x8;
      - address: 0x0002
        value_type: U_WORD
        read_lambda: |-
          return (random_uint32() % 100);
```

I hope configuration is OK.

What I have seen is if I just ask for the configured registers, the behaviour is OK.

ESPHOME logs:
```
[17:03:44][D][modbus_controller:113]: Received read holding/input registers for device 0x14. FC: 0x3. Start address: 0x1. Number of registers: 0x2.
[17:03:44][D][modbus_controller:124]: Matched register. Address: 0x01. Value type: 1. Register count: 1. Value: 8.0.
[17:03:44][D][modbus_controller:124]: Matched register. Address: 0x02. Value type: 1. Register count: 1. Value: 79.0.
[17:03:45][D][modbus_controller:113]: Received read holding/input registers for device 0x14. FC: 0x3. Start address: 0x1. Number of registers: 0x2.
[17:03:45][D][modbus_controller:124]: Matched register. Address: 0x01. Value type: 1. Register count: 1. Value: 8.0.
[17:03:45][D][modbus_controller:124]: Matched register. Address: 0x02. Value type: 1. Register count: 1. Value: 38.0.
[17:03:46][D][modbus_controller:113]: Received read holding/input registers for device 0x14. FC: 0x3. Start address: 0x1. Number of registers: 0x2.
[17:03:46][D][modbus_controller:124]: Matched register. Address: 0x01. Value type: 1. Register count: 1. Value: 8.0.
[17:03:46][D][modbus_controller:124]: Matched register. Address: 0x02. Value type: 1. Register count: 1. Value: 87.0.
```
Modbus logs:
```
17.07.2025 17:03:44.819 'Test Slave' Info: Start polling
17.07.2025 17:03:44.819 'Device' Tx: 14 03 00 01 00 02 97 0E 
17.07.2025 17:03:44.909 'Device' Rx: 14 03 04 00 08 00 4F 7E C4 
17.07.2025 17:03:45.815 'Device' Tx: 14 03 00 01 00 02 97 0E 
17.07.2025 17:03:45.912 'Device' Rx: 14 03 04 00 08 00 26 BE EA 
17.07.2025 17:03:46.815 'Device' Tx: 14 03 00 01 00 02 97 0E 
17.07.2025 17:03:46.916 'Device' Rx: 14 03 04 00 08 00 57 7E CE 
```

The problem is I configure my Modbus Client to ask to some register is not configured in ESPHome device. For example, instead of requesting 2 registers (1 and 2), I request 3 (1, 2 and 3).
ESPHOME logs look ok:
```
[17:07:27][D][modbus_controller:113]: Received read holding/input registers for device 0x14. FC: 0x3. Start address: 0x1. Number of registers: 0x3.
[17:07:27][D][modbus_controller:124]: Matched register. Address: 0x01. Value type: 1. Register count: 1. Value: 8.0.
[17:07:27][D][modbus_controller:124]: Matched register. Address: 0x02. Value type: 1. Register count: 1. Value: 81.0.
[17:07:27][W][modbus_controller:134]: Could not match any register to address 03. Sending exception response.
[17:07:28][D][modbus_controller:113]: Received read holding/input registers for device 0x14. FC: 0x3. Start address: 0x1. Number of registers: 0x3.
[17:07:28][D][modbus_controller:124]: Matched register. Address: 0x01. Value type: 1. Register count: 1. Value: 8.0.
[17:07:28][D][modbus_controller:124]: Matched register. Address: 0x02. Value type: 1. Register count: 1. Value: 61.0.
[17:07:28][W][modbus_controller:134]: Could not match any register to address 03. Sending exception response.
[17:07:29][D][modbus_controller:113]: Received read holding/input registers for device 0x14. FC: 0x3. Start address: 0x1. Number of registers: 0x3.
[17:07:29][D][modbus_controller:124]: Matched register. Address: 0x01. Value type: 1. Register count: 1. Value: 8.0.
[17:07:29][D][modbus_controller:124]: Matched register. Address: 0x02. Value type: 1. Register count: 1. Value: 82.0.
[17:07:29][W][modbus_controller:134]: Could not match any register to address 03. Sending exception response.
```
But Modbus Logs:
```
17.07.2025 17:07:27.158 'Test Slave' Info: Start polling
17.07.2025 17:07:27.158 'Device' Tx: 14 03 00 01 00 03 56 CE 
17.07.2025 17:07:27.266 'Device' Rx: 14 81 02 D0 55 
17.07.2025 17:07:27.266 'Device' Error: Returned Modbus-exception with code 2
17.07.2025 17:07:28.154 'Device' Tx: 14 03 00 01 00 03 56 CE 
17.07.2025 17:07:28.261 'Device' Rx: 14 81 02 D0 55 
17.07.2025 17:07:28.261 'Device' Error: Returned Modbus-exception with code 2
17.07.2025 17:07:29.154 'Device' Tx: 14 03 00 01 00 03 56 CE 
17.07.2025 17:07:29.258 'Device' Rx: 14 81 02 D0 55 
17.07.2025 17:07:29.258 'Device' Error: Returned Modbus-exception with code 2
17.07.2025 17:07:29.635 'Test Slave' Info: Finish polling
```

The response:
`14 81 02 D0 55`
If we parse it (for example using this [online tool](https://rapidscada.net/modbus/).

`81	Error code	0x80 + 0x01 (1) - Read Coils`

Error code should not be 0x81 but 0x83 as the request was 0x03 (Read Holding).
In Modbus Specs ([MODBUS Exception Responses](https://www.modbus.org/docs/Modbus_Application_Protocol_V1_1b3.pdf#page=47)), it can be read:

> Function Code Field: In a normal response, the server echoes the function code of the original request in the function code field of the response. All function codes have a most – significant bit (MSB) of 0 (their values are all below 80 hexadecimal). In an exception response, the server sets the MSB of the function code to 1. This makes the function code value in an exception response exactly 80 hexadecimal higher than the value would be for a normal response. With the function code’s MSB set, the client's application program can recognize the exception response and can examine the data field for the exception code.

See in detail this:
> This makes the function code value in an exception response exactly 80 hexadecimal higher than the value would be for a normal response.

Also regarding Data Field:

> Data Field: In a normal response, the server may return data or statistics in the data field (any information that was requested in the request). In an exception response, the server returns an exception code in the data field. This defines the server condition that caused the exception.

I have the feeling that ESPHome is always responding with:
> Function: 0x81
> Exception Code: 0x02
As they are the values set as an example in Modbus specs (see table ([MODBUS Exception Responses](https://www.modbus.org/docs/Modbus_Application_Protocol_V1_1b3.pdf#page=47))

### Which version of ESPHome has the issue?

Version: 2025.6.3

### What type of installation are you using?

pip

### What platform are you using?

ESP32

### Component causing the issue

Modbus Controller

### YAML Config

```yaml

```

### Anything in the logs that might be useful for us?

```txt

```

carlessole, 2025-07-17:
After taking a look on current **dev** branch. Seem it has been already solved by @jpeletier in this commit: [[modbus_controller] Fix modbus read_lambda precision for non-floats or large integers ](https://github.com/esphome/esphome/commit/2df0ebd895220f8f8fd0b37bb929749f47f1a6ab#diff-2fbaadb42c820845c1fd051ec37181d4dac0f6ea76306aeb3139394a2803ad9aL66)

[image]

[image]

I guess that on next release it will be already solved, right?
Thanks

jpeletier, 2025-07-17:
If it is in `dev ` you could see the release tags being ahead or behind this PR to see when will it be published.