esphome/esphome #9650: Instalation error after update to 2025.7.0 with powerpal-gateway

HSB-collab, 2025-07-18:
### The problem

Since 2025.7.0 I am unable to update the powerpal-gateway instance on ESPhome. Prior to 2025.7.0 it was working. If I revert to 2025.6.2 it works again.

I get an error saying "/data/cache/platformio/packages/framework-arduinoespressif32/cores/esp32/Arduino.h:43:10: fatal error: pins_arduino.h: No such file or directory"

### Which version of ESPHome has the issue?

2025.7.0 and 2025.7.1

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

### Component causing the issue

Powerpal gatway

### YAML Config

```yaml
esphome:
  name: powerpal-gateway
esp32:
  board: m5stack-core-esp32
external_components:
  - source:
      type: git
      url: https://github.com/muneeb1990/esphome
      ref: powerpal_ble
    components: [powerpal_ble]
# Enable Home Assistant API
api:
  encryption:
    key: "********************************"

ota:
  - platform: esphome
    password: "**************************************"
wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

# optional requirement used with daily energy sensor
time:
  - platform: homeassistant
    id: homeassistant_time

esp32_ble_tracker:

ble_client:
  - mac_address: ##:##:##:##:##:##
    id: powerpal

switch:
  - platform: restart
    name: "Powerpal Monitor Reboot"

sensor:
  - platform: powerpal_ble
    ble_client_id: powerpal
    power:
      name: "Powerpal Power"
    daily_energy:
      name: "Powerpal Daily Energy"
    energy:
      name: "Powerpal Total Energy"
    battery_level:
      name: "Powerpal Battery"
    pairing_code: 797375
    notification_interval: 1
    pulses_per_kwh: 3200
    time_id: homeassistant_time # daily energy still works without a time_id, but recommended to include one to properly handle daylight savings, etc.
```

### Anything in the logs that might be useful for us?

```txt
Updating /config/esphome/powerpal-gateway.yaml
------------------------------------------------------------

INFO ESPHome 2025.7.1
INFO Reading configuration /config/esphome/powerpal-gateway.yaml...
INFO Updating https://github.com/muneeb1990/esphome@powerpal_ble
INFO Detected timezone 'Australia/Melbourne'
INFO Generating C++ source...
INFO Compiling app...
Processing powerpal-gateway (board: m5stack-core-esp32; framework: arduino; platform: https://github.com/pioarduino/platform-espressif32/releases/download/53.03.13/platform-espressif32.zip)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - framework-arduinoespressif32 @ 3.1.3 
 - framework-arduinoespressif32-libs @ 5.3.0+sha.489d7a2b3a 
 - tool-esptoolpy @ 4.8.6 
 - tool-mklittlefs @ 3.2.0 
 - tool-riscv32-esp-elf-gdb @ 14.2.0+20240403 
 - tool-xtensa-esp-elf-gdb @ 14.2.0+20240403 
 - toolchain-xtensa-esp-elf @ 13.2.0+20240530
Dependency Graph
|-- Networking @ 3.1.3
|-- WiFi @ 3.1.3
|-- ESPmDNS @ 3.1.3
|-- Update @ 3.1.3
|-- noise-c @ 0.1.10
Compiling .pioenvs/powerpal-gateway/src/esphome/components/api/api_connection.cpp.o
Compiling .pioenvs/powerpal-gateway/src/esphome/components/api/api_frame_helper.cpp.o
In file included from src/esphome/core/macros.h:7,
                 from src/esphome/core/defines.h:2,
                 from src/esphome/components/api/api_connection.h:3,
                 from src/esphome/components/api/api_connection.cpp:1:
/data/cache/platformio/packages/framework-arduinoespressif32/cores/esp32/Arduino.h:43:10: fatal error: pins_arduino.h: No such file or directory

**********************************************************************
* Looking for pins_arduino.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:pins_arduino.h"
* Web  > https://registry.platformio.org/search?q=header:pins_arduino.h
*
**********************************************************************

   43 | #include "pins_arduino.h"
      |          ^~~~~~~~~~~~~~~~
compilation terminated.
*** [.pioenvs/powerpal-gateway/src/esphome/components/api/api_connection.cpp.o] Error 1
In file included from src/esphome/core/macros.h:7,
                 from src/esphome/core/defines.h:2,
                 from src/esphome/components/api/api_frame_helper.h:9,
                 from src/esphome/components/api/api_frame_helper.cpp:1:
/data/cache/platformio/packages/framework-arduinoespressif32/cores/esp32/Arduino.h:43:10: fatal error: pins_arduino.h: No such file or directory

**********************************************************************
* Looking for pins_arduino.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:pins_arduino.h"
* Web  > https://registry.platformio.org/search?q=header:pins_arduino.h
*
**********************************************************************

   43 | #include "pins_arduino.h"
      |          ^~~~~~~~~~~~~~~~
compilation terminated.
*** [.pioenvs/powerpal-gateway/src/esphome/components/api/api_frame_helper.cpp.o] Error 1
========================== [FAILED] Took 3.47 seconds ==========================
====== [ERROR] /config/esphome/powerpal-gateway.yaml ======

======================== [SUMMARY] ========================
  - /config/esphome/powerpal-gateway.yaml: FAILED
```

ssieb, 2025-07-18:
Arduino doesn't support pins for that board for some reason.  Use `esp32dev` instead.

HSB-collab, 2025-07-18:
> Arduino doesn't support pins for that board for some reason. Use `esp32dev` instead.

Thanks for the help.

When I change the configuration to 

```
esp32:`
      board: esp32dev
      framework:
        type: esp-idf
```

I get a different error as per this from the logs:

src/esphome/components/powerpal_ble/powerpal_ble.cpp:4:10: fatal error: WiFi.h: No such file or directory

********************************************************************
* Looking for WiFi.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:WiFi.h"
* Web  > https://registry.platformio.org/search?q=header:%1B%5Bm%1B%5BKWiFi.h
*
********************************************************************

    4 | #include "WiFi.h"
      |          ^~~~~~~~
compilation terminated.
*** [.pioenvs/powerpal-gateway/src/esphome/components/powerpal_ble/powerpal_ble.cpp.o] Error 1
========================== [FAILED] Took 7.46 seconds ==========================
====== [ERROR] /config/esphome/powerpal-gateway.yaml ======

======================== [SUMMARY] ========================
  - /config/esphome/powerpal-gateway.yaml: FAILED

I have tried looking at the library registry but I can't work it out. Any help would be appreciated

ssieb, 2025-07-18:
Try a clean build.

jesserockz, 2025-07-18:
That external component tried to include WiFi.h making it Arduino only. File an issue with the component author to make it compatible with esp-idf