esphome/esphome #9659: From 2025.7.0 problem with compilation

kotbrs, 2025-07-18:
### The problem

From 2025.7.0 (2025.7.1 too) release I have a problem with compilation. I made a backup of version 2025.6.3, and when I restore it, the compilation goes smoothly. I've cleared the cache. I'm installing 2025.7.x, but I'm still facing issues with compiling the device's YAML configuration. On my other devices, the compilation proceeds without any problems.

### Which version of ESPHome has the issue?

2025.7.x

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

### Component causing the issue

I don't know

### YAML Config

```yaml
################################################################################
# Substitution Variables
################################################################################
substitutions:
  device_internal_name: esp32_s3_lcd7_01
  device_wifi_name: esp32-s3-lcd7-01
  device_friendly_name: esp32 s3 lcd7 01
#  device_ip_address: 192.168.1.19
  device_sampling_time: 30s

  # glyphs
  lightbulb:         "\U000F0335"
  ceiling_light:     "\U000F0769"
  lamp:              "\U000F06B5"
  floor_lamp:        "\U000F08DD"
  string_lights:     "\U000F12BA"
  track_light:       "\U000F0914"
  bulkhead_light:    "\U000F1A2F" 
  led_strip:         "\U000F07D6" 
  coach_lamp:        "\U000F1020"
  ceiling_fan_light: "\U000F1798"
  previous:          "\U0000F053"
  next:              "\U0000F054"
  home:              "\U0000F015"

  light_recessed: "\U000F179B"
  wall_sconce_round: "\U000F0748"
  gas_burner: "\U000F1A1B"
  home_icon: "\U000F02DC"
  menu_left: "\U000F0A02"
  menu_right: "\U000F035F"
  close: "\U000F0156"
  delete: "\U000F01B4"
  backspace: "\U000F006E"
  check: "\U000F012C"
  arrow_down: "\U000F004B"

################################################################################
# Globals
################################################################################
globals: ##to set default reboot behavior
  - id: wifi_connection
    type: bool
    restore_value: no
    initial_value: "false"

################################################################################
[... 1328 lines cut ...]
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 512, in command_run
    exit_code = write_cpp(config)
                ^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 212, in write_cpp
    generate_cpp_contents(config)
  File "/esphome/esphome/__main__.py", line 224, in generate_cpp_contents
    CORE.flush_tasks()
  File "/esphome/esphome/core/__init__.py", line 694, in flush_tasks
    self.event_loop.flush_tasks()
  File "/esphome/esphome/coroutine.py", line 245, in flush_tasks
    next(task.iterator)
  File "/esphome/esphome/__main__.py", line 204, in wrapped
    await coro(conf)
  File "/esphome/esphome/components/lvgl/number/__init__.py", line 55, in to_code
    max_value=widget.get_max(),
              ^^^^^^^^^^^^^^^^
  File "/esphome/esphome/components/lvgl/widgets/__init__.py", line 195, in get_max
    return self.type.get_max(self.config)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/components/lvgl/types.py", line 195, in get_max
    return int(config[CONF_MAX_VALUE] or 100)
               ~~~~~~^^^^^^^^^^^^^^^^
KeyError: 'max_value'
```

jesserockz, 2025-07-18:
https://github.com/esphome/esphome/pull/9660#issuecomment-3089131140

jesserockz, 2025-07-18:
I think a simple fix would be to add a min_value: 0 and max_value: 100 to your `lv_bar` widget. The linked PR will fix the case where none are defined

closed 2025-07-18 by clydebarrow