esphome/esphome #9662: #warning "legacy pcnt driver is deprecated"

nliaudat, 2025-07-18:
### The problem

#warning "legacy pcnt driver is deprecated, please migrate to use driver/pulse_cnt.h"

with latest esphome version (framework-espidf @ 3.50105.0 (5.1.5))

### Which version of ESPHome has the issue?

2024.12.2

### What type of installation are you using?

Docker

### What platform are you using?

ESP32-IDF

### Component causing the issue

esp32-devkit-4

### YAML Config

```yaml
sensor:

  - platform: pulse_counter
    pin: GPIO32
    # count_mode: rising_edge 
    use_pcnt: true
    internal_filter: 1us
    name: "Pulse Meter 2"
    total:
        name: "total pulse_meter_2"
    id : ${id_prefix}pulse_meter_2

  - platform: pulse_meter
    pin: GPIO33
    name: "pulse_meter_1"
    internal_filter: 1us
    internal_filter_mode: EDGE  
    # timeout: 10s
    total:
        name: "total pulse_meter_1"
    id : ${id_prefix}pulse_meter_1
```

BlueForrest, 2025-08-20:
I have the same warning message.
how the add the "pulse_cnt.h" driver into ESPHome within Home Assistant?

ssieb, 2025-08-21:
Someone needs to migrate the code to the new driver.  It's just a warning for now and everything still works.

alexyao2015, 2025-12-30:
This should probably be fixed before idf becomes the default framework

t0urista, 2026-01-05:
still have the same warning message with ESPHOME 2025.12.04