esphome/esphome #9684: Update to 2025.7 breaks wifi

crudolphy, 2025-07-18:
### The problem

I have an Wemos ESP32 Mini with a SHT3X Temp and Humidity Sensor attached.  During the update (OTA, WiFi) couldn't complete the update.  I attached to my computer via USB, and updated it again via USB and Chrome.  When I look at the logs it scans for the SSID, finds the SSID, tries to connect but can't and the response is unknown.
I have an ESP8266 that has a DS18B20 which I did not update and it is online with no issues.  I can connect to it and see the logs.  The Wemos has been working fine for several years now and nothing has changed with the exception of the update to 2025.7.

### Which version of ESPHome has the issue?

2025.7

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

### Component causing the issue

Wifi

### YAML Config

```yaml
esphome:
  name: humiditytempsensor

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "e4bSbcheKYtqN9nB7YWAONfexDE+U7NqqDf7UsToeXw="

ota:
  platform: esphome
  password: "29b68e6994f6c76db6f8ebc93fbd23c9"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
    
  domain: ".ccrudolphy.com"
  manual_ip:
    static_ip: 192.168.30.81
    gateway: 192.168.30.1
    subnet: 255.255.255.0
    dns1: 192.168.1.1

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Humiditytempsensor"
    password: "VLzTHCQYfZaZ"

captive_portal:
i2c:
  - sda: 21
    scl: 22
    scan: true
    id: bus_a
sensor:
  - platform: sht3xd
    temperature:
      name: "test_temp_1"
    humidity:
      name: "temp_humidity_1"
    i2c_id: bus_a
    address: 0x44
    update_interval: 60s
```

### Anything in the logs that might be useful for us?

```txt
[14:40:30][D][wifi:495]: Starting scan
[14:40:32][D][wifi:516]: Found networks:
[14:40:32][I][wifi:553]: - 'ccr-iot' [redacted]▂▄▆█
[14:40:32][D][wifi:555]:     Channel: 11
[14:40:32][D][wifi:556]:     RSSI: -73 dB
[14:40:32][I][wifi:553]: - 'ccr-iot' [redacted]▂▄▆█
[14:40:32][D][wifi:555]:     Channel: 1
[14:40:32][D][wifi:556]:     RSSI: -49 dB
[14:40:32][I][wifi:553]: - 'ccr-iot' [redacted]▂▄▆█
[14:40:32][D][wifi:555]:     Channel: 6
[14:40:32][D][wifi:556]:     RSSI: -55 dB

[14:40:33][I][wifi:319]: Connecting to 'ccr-iot'
[14:40:35][W][wifi:687]: Unknown connection status 0
[14:40:35][D][wifi:710]: Retrying with hidden networks
[14:40:35][I][wifi:319]: Connecting to 'ccr-iot'
[14:40:37][W][wifi:687]: Unknown connection status 0
[14:40:37][D][wifi:710]: Retrying with hidden networks
[14:40:42][I][wifi:319]: Connecting to 'ccr-iot'
[14:40:45][W][wifi_esp32:555][arduino_events]: Disconnected ssid='ccr-iot' bssid=[redacted] reason='Unspecified'
[14:40:45][W][wifi:687]: Unknown connection status 0
[14:40:45][W][wifi:703]: Restarting adapter
[14:40:45][I][wifi:319]: Connecting to 'ccr-iot'
[14:40:47][W][wifi_esp32:555][arduino_events]: Disconnected ssid='ccr-iot' bssid=[redacted] reason='Unspecified'
[14:40:47][W][wifi:687]: Unknown connection status 0

ESPHome Device Builder
Current version: 2025.7.1 (Changelog)
```

bdraco, 2025-07-18:
can you try https://github.com/esphome/esphome/pull/9689#issuecomment-3090654239

bdraco, 2025-07-18:
In the mean time, the likely solution is to switch to esp-idf platform as it uses event driven wifi instead of polling

crudolphy, 2025-07-19:
Just a note on this.  I updated to 2025.7.2 and tried again.  It didn't help.  I looked at your changelog and didn't see anything that referenced this problem so I didn't think it would do anything, but I thought I would try it just in case.

crudolphy, 2025-07-24:
> Did you try with esp-idf?

No, don't know what esp-idf is.  How should I implement that?

ssieb, 2025-07-24:
Under `framework:`, set `type: esp-idf` instead of `arduino`.

crudolphy, 2025-07-25:
So change that re-install using the 2025.7.3 implementation of the esp home
add-on and the see if it connects to WiFi.  Correct?

Chuck Rudolphy
346.383.6920
Mobile from my Pixel 6

ssieb, 2025-07-25:
yes

crudolphy, 2025-07-25:
Doesn't work here is a snippet from the logs. Seems to keep repeating this
endlessly.

[19:22:09]I (450) esp_image: segment 4: paddr=000e41d0 vaddr=40084558
size=12908h ( 76040) load [19:22:09]I (491) boot: Loaded app from partition
at offset 0x10000 [19:22:09]I (491) boot: Disabling RNG early entropy
source... [19:22:09]I (504) cpu_start: Multicore app [19:22:09]I (512)
cpu_start: Pro cpu start user code [19:22:09]I (512) cpu_start: cpu freq:
160000000 Hz [19:22:09]I (512) app_init: Application information: [19:22:09]I
(515) app_init: Project name: humiditytempsensor [19:22:09]I (521)
app_init: App version: 2025.7.3 [19:22:09]I (526) app_init: Compile time:
Jul 24 2025 19:18:45 [19:22:09]I (532) app_init: ELF file SHA256:
0039da9fd... [19:22:09]I (537) app_init: ESP-IDF: 5.3.2 [19:22:09]I (542)
efuse_init: Min chip rev: v0.0 [19:22:09]I (546) efuse_init: Max chip rev:
v3.99 [19:22:09]I (551) efuse_init: Chip rev: v1.0 [19:22:09]I (556)
heap_init: Initializing. RAM available for dynamic allocation: [19:22:10]I
(564) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM [19:22:10]I (569)
heap_init: At 3FFB8558 len 00027AA8 (158 KiB): DRAM [19:22:10]I (576)
heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM [19:22:10]I (582)
heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM [19:22:10]I (588)
heap_init: At 40096E60 len 000091A0 (36 KiB): IRAM [19:22:10]I (596)
spi_flash: detected chip: generic [19:22:10]I (599) spi_flash: flash io: dio
[19:22:10]W (604) i2c: This driver is an old driver, please migrate your
application code to adapt `driver/i2c_master.h` [19:22:10]I (615)
main_task: Started on CPU0 [19:22:10]I (619) main_task: Calling app_main()
[19:22:10]0;32m[I][logger:169]: Log initialized [19:22:10][C][safe_mode:082]:
There have been 3 suspected unsuccessful boot attempts
[19:22:10][D][esp32.preferences:142]:
Writing 1 items: 0 cached, 1 written, 0 failed [19:22:10][I][app:049]:
Running through setup() [19:22:10][C][i2c.idf:021]: Running setup
[19:22:10][I][i2c.idf:262]:
Performing bus recovery [19:22:10][D][esp-idf:000]: I (783) gpio: GPIO[22]|
InputEn: 1| OutputEn: 1| OpenDrain: 1| Pullup: 1| Pulldown: 0| Intr:0
[19:22:10][D][esp-idf:000]: I (785) gpio: GPIO[21]| InputEn: 1| OutputEn:
1| OpenDrain: 1| Pullup: 1| Pulldown: 0| Intr:0 [19:22:10][C][sht3xd:028]:
Running setup [19:22:10][C][wifi:048]: Running setup
[19:22:10][D][esp-idf:000]:
I (836) wifi: [19:22:10][D][esp-idf:000]: wifi driver task: 3ffc53e4,
prio:23, stack:6656, core=0 [19:22:10][D][esp-idf:000]: [19:22:10]
[D][esp-idf:000][wifi]: Init data frame dynamic rx buffer num: 32 [19:22:10]
[D][esp-idf:000][wifi]: I (842) wifi: [19:22:10][D][esp-idf:000][wifi]:
Init static rx mgmt buffer num: 5 [19:22:10][D][esp-idf:000][wifi]:
[19:22:10][D][esp-idf:000][wifi]: I (849) wifi: [19:22:10][D][esp-idf:000]
[wifi]: Init management short buffer num: 32 [19:22:10][D][esp-idf:000]
[wifi]: [19:22:10][D][esp-idf:000][wifi]: I (869) wifi: [19:22:10]
[D][esp-idf:000][wifi]: Init dynamic tx buffer num: 32 [19:22:10]
[D][esp-idf:000][wifi]: [19:22:10][D][esp-idf:000][wifi]: I (880) wifi:
[19:22:10][D][esp-idf:000][wifi]: Init static rx buffer size: 1600
[19:22:10][D][esp-idf:000][wifi]: [19:22:10][D][esp-idf:000][wifi]: I (900)
wifi: [19:22:10][D][esp-idf:000][wifi]: Init static rx buffer num: 10
[19:22:10][D][esp-idf:000][wifi]: [19:22:10][D][esp-idf:000][wifi]: I (911)
wifi: [19:22:10][D][esp-idf:000][wifi]: Init dynamic rx buffer num: 32
[19:22:10][D][esp-idf:000][wifi]: [19:22:10][D][esp-idf:000]: I (932)
wifi_init: rx ba win: 6 [19:22:10][D][esp-idf:000]: I (941) wifi_init:
accept mbox: 6 [19:22:10][D][esp-idf:000]: I (951) wifi_init: tcpip mbox: 32
[19:22:10][D][esp-idf:000]: I (951) wifi_init: udp mbox: 6
[19:22:10][D][esp-idf:000]:
I (962) wifi_init: tcp mbox: 6 [19:22:10][D][esp-idf:000]: I (962)
wifi_init: tcp tx win: 5760 [19:22:10][D][esp-idf:000]: I (972) wifi_init:
tcp rx win: 5760 [19:22:10][D][esp-idf:000]: I (982) wifi_init: tcp mss:
1440 [19:22:10][D][esp-idf:000]: I (982) wifi_init: WiFi IRAM OP enabled
[19:22:10][D][esp-idf:000]: I (992) wifi_init: WiFi RX IRAM OP enabled
[19:22:10][C][wifi:061]: Starting [19:22:10] Local MAC: 30:C6:F7:99:17:A8
[19:22:10][D][esp-idf:000][wifi]: I (1004) phy_init: phy_version
4840,02e0d70,Sep 2 2024,19:39:07 [19:22:10][D][esp-idf:000][wifi]: W (1013)
phy_init: failed to load RF calibration data (0x1102), falling back to full
calibr [19:22:10]�ets Jun 8 2016 00:22:57 [19:22:10] [19:22:10]rst:0x3
(SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT) [19:22:10]configsip: 0,
SPIWP:0xee [19:22:10]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[19:22:10]mode:DIO, clock div:2 [19:22:10]load:0x3fff0030,len:7176
[19:22:10]load:0x40078000,len:15564
[19:22:10]ho 0 tail 12 room 4 [19:22:10]load:0x40080400,len:4
[19:22:10]load:0x40080404,len:3904
[19:22:10]entry 0x40080638

bdraco, 2025-07-25:
Please use code fences in logs to make them readable https://docs.github.com/en/get-started/writing-on-github/working-with-advanced-formatting/creating-and-highlighting-code-blocks

crudolphy, 2025-07-25:
Hopefully this is better???

`[19:25:16][D][esp-idf:000][wifi]: I (61003) wifi:
[19:25:16][D][esp-idf:000][wifi]: mode : sta (30:c6:f7:99:17:a8) + softAP (30:c6:f7:99:17:a9)
[19:25:16][D][esp-idf:000][wifi]: 
[19:25:16][D][esp-idf:000][wifi]: I (61005) wifi:
[19:25:16][D][esp-idf:000][wifi]: Total power save buffer number: 16
[19:25:16][D][esp-idf:000][wifi]: 
[19:25:16][D][esp-idf:000][wifi]: I (61015) wifi:
[19:25:16][D][esp-idf:000][wifi]: Init max length of beacon: 752/752
[19:25:16][D][esp-idf:000][wifi]: 
[19:25:16][D][esp-idf:000][wifi]: I (61035) wifi:
[19:25:16][D][esp-idf:000][wifi]: Init max length of beacon: 752/752
[19:25:16][D][esp-idf:000][wifi]: 
[19:25:16][C][wifi:265]: Setting up AP[D][esp-idf:000][tiT]: I (61047) esp_netif_lwip: DHCP server started on interface WIFI_AP_DEF with IP: 192.168.4.1
[19:25:16]
[19:25:16][C][wifi:267]:   AP SSID: 'Humiditytempsensor'
[19:25:16]  AP Password: 'VLzTHCQYfZaZ'
[19:25:16][D][esp-idf:000][wifi]: I (61319) wifi:
[19:25:16][D][esp-idf:000][wifi]: Total power save buffer number: 16
[19:25:16][D][esp-idf:000][wifi]: 
[19:25:16][D][esp-idf:000][tiT]: I (61322) esp_netif_lwip: DHCP server started on interface WIFI_AP_DEF with IP: 192.168.4.1
[19:25:16][C][wifi:281]:   IP Address: 192.168.4.1
[19:25:16][C][api:034]: Running setup
[19:25:16][W][component:279]: api set Warning flag: unspecified
[19:25:16][D][esp-idf:000]: I (61349) mdns_mem: mDNS task will be created from internal RAM
[19:25:16][I][app:091]: setup() finished successfully!
[19:25:16][I][safe_mode:042]: Boot seems successful; resetting boot loop counter
[19:25:16][D][esp32.preferences:142]: Writing 1 items: 0 cached, 1 written, 0 failed
[19:25:16][I][app:149]: ESPHome version 2025.7.3 compiled on Jul 24 2025, 19:18:33
[19:25:16][C][wifi:613]: WiFi:
[19:25:16][D][esp-idf:000][wifi]: W (61376) wifi:
[19:25:16][D][esp-idf:000][wifi]: Haven't to connect to a suitable AP now!
[19:25:16][D][esp-idf:000][wifi]: 
[19:25:16][W][wifi_esp32:965]: esp_wifi_sta_get_ap_info failed: ESP_ERR_WIFI_NOT_CONNECT
[19:25:16][C][wifi:434]:   Local MAC: 30:C6:F7:99:17:A8
[19:25:16][D][esp-idf:000][wifi]: W (61406) wifi:
[19:25:16][D][esp-idf:000][wifi]: Haven't to connect to a suitable AP now!
[19:25:16][D][esp-idf:000][wifi]: 
[19:25:16][W][wifi_esp32:975]: esp_wifi_sta_get_ap_info failed: ESP_ERR_WIFI_NOT_CONNECT
[19:25:16][C][wifi:439]:   SSID: [redacted]
[19:25:16][C][wifi:442]:   IP Address: 192.168.30.81
[19:25:16][D][esp-idf:000][wifi]: W (61437) wifi:
[19:25:16][D][esp-idf:000][wifi]: Haven't to connect to a suitable AP now!
[19:25:16][D][esp-idf:000][wifi]: 
[19:25:16][W][wifi_esp32:986]: esp_wifi_sta_get_ap_info failed: ESP_ERR_WIFI_NOT_CONNECT
[19:25:16][C][wifi:446]:   BSSID: [redacted]
[19:25:16]  Hostname: 'humiditytempsensor'
[19:25:16]  Signal strength: 0 dB ▂▄▆█
[19:25:16][C][wifi:455]:   Channel: 1
[19:25:16]  Subnet: 255.255.255.0
[19:25:16]  Gateway: 192.168.30.1
[19:25:16]  DNS1: 192.168.1.1
[19:25:16]  DNS2: 0.0.0.0
[19:25:16][C][logger:246]: Logger:
[19:25:16]  Max Level: DEBUG
[19:25:16]  Initial Level: DEBUG
[19:25:16][C][logger:252]:   Log Baud Rate: 115200
[19:25:16]  Hardware UART: UART0
[19:25:16][C][logger:259]:   Task Log Buffer Size: 768
[19:25:16][C][i2c.idf:083]: I2C Bus:
[19:25:16][C][i2c.idf:084]:   SDA Pin: GPIO21
[19:25:16]  SCL Pin: GPIO22
[19:25:16]  Frequency: 50000 Hz
[19:25:16][C][i2c.idf:094]:   Recovery: bus successfully recovered
[19:25:16][I][i2c.idf:104]: Results from bus scan:
[19:25:16][I][i2c.idf:110]: Found device at address 0x44
[19:25:16][C][sht3xd:049]: SHT3xD:
[19:25:16][D][sht3xd:064]:   Setup successful
[19:25:16][D][sht3xd:065]:   Serial Number: 0x122F7BFD
[... 243 lines cut ...]
[19:28:21][D][esp-idf:000][wifi]: Password length matches WPA2 standards, authmode threshold changes from OPEN to WPA2
[19:28:21][D][esp-idf:000][wifi]: 
[19:28:23][D][sht3xd:097]: Got temperature=27.76°C humidity=42.32%
[19:28:23][D][sensor:104]: 'test_temp_1': Sending state 27.75578 °C with 1 decimals of accuracy
[19:28:23][D][sensor:104]: 'temp_humidity_1': Sending state 42.31784 % with 1 decimals of accuracy
[19:28:24][W][wifi_esp32:710]: Disconnected ssid='ccr-iot' bssid=[redacted] reason='Unspecified'
[19:28:24][W][wifi:666]: Connecting to network failed
[19:28:24][I][wifi:319]: Connecting to 'ccr-iot'
[19:28:24][D][esp-idf:000][wifi]: W (249248) wifi:
[19:28:24][D][esp-idf:000][wifi]: Password length matches WPA2 standards, authmode threshold changes from OPEN to WPA2
[19:28:24][D][esp-idf:000][wifi]: 
[19:28:27][W][wifi_esp32:710]: Disconnected ssid='ccr-iot' bssid=[redacted] reason='Unspecified'
[19:28:27][W][wifi:666]: Connecting to network failed
[19:28:32][I][wifi:319]: Connecting to 'ccr-iot'
[19:28:32][D][esp-idf:000][wifi]: W (257138) wifi:
[19:28:32][D][esp-idf:000][wifi]: Password length matches WPA2 standards, authmode threshold changes from OPEN to WPA2
[19:28:32][D][esp-idf:000][wifi]: 
[19:28:35][W][wifi_esp32:710]: Disconnected ssid='ccr-iot' bssid=[redacted] reason='Unspecified'
[19:28:35][W][wifi:666]: Connecting to network failed
[19:28:35][I][wifi:319]: Connecting to 'ccr-iot'
[19:28:35][D][esp-idf:000][wifi]: W (260011) wifi:
[19:28:35][D][esp-idf:000][wifi]: Password length matches WPA2 standards, authmode threshold changes from OPEN to WPA2
[19:28:35][D][esp-idf:000][wifi]: 
[19:28:38][W][wifi_esp32:710]: Disconnected ssid='ccr-iot' bssid=[redacted] reason='Unspecified'
[19:28:38][W][wifi:666]: Connecting to network failed`

ssieb, 2025-07-25:
How are you copying it that it loses all the line endings?

crudolphy, 2025-07-25:
It is getting to dinner time here.  I will pick this up in the morning.  Thanks for your attention to this.
I highlight with my mouse and the right click then click copy.  Then I went to the email body in gmail and right clicked again and said paste.

Here I just clicked on the <> code icon then pasted in between.  So it needs three back ticks on each end.  Lets try this
```[19:25:16][D][esp-idf:000][wifi]: I (61003) wifi:
[19:25:16][D][esp-idf:000][wifi]: mode : sta (30:c6:f7:99:17:a8) + softAP (30:c6:f7:99:17:a9)
[19:25:16][D][esp-idf:000][wifi]: 
[19:25:16][D][esp-idf:000][wifi]: I (61005) wifi:
[19:25:16][D][esp-idf:000][wifi]: Total power save buffer number: 16
[19:25:16][D][esp-idf:000][wifi]: 
[19:25:16][D][esp-idf:000][wifi]: I (61015) wifi:
[19:25:16][D][esp-idf:000][wifi]: Init max length of beacon: 752/752
[19:25:16][D][esp-idf:000][wifi]: 
[19:25:16][D][esp-idf:000][wifi]: I (61035) wifi:
[19:25:16][D][esp-idf:000][wifi]: Init max length of beacon: 752/752
[19:25:16][D][esp-idf:000][wifi]: 
[19:25:16][C][wifi:265]: Setting up AP[D][esp-idf:000][tiT]: I (61047) esp_netif_lwip: DHCP server started on interface WIFI_AP_DEF with IP: 192.168.4.1
[19:25:16]
[19:25:16][C][wifi:267]:   AP SSID: 'Humiditytempsensor'
[19:25:16]  AP Password: 'VLzTHCQYfZaZ'
[19:25:16][D][esp-idf:000][wifi]: I (61319) wifi:
[19:25:16][D][esp-idf:000][wifi]: Total power save buffer number: 16
[19:25:16][D][esp-idf:000][wifi]: 
[19:25:16][D][esp-idf:000][tiT]: I (61322) esp_netif_lwip: DHCP server started on interface WIFI_AP_DEF with IP: 192.168.4.1
[19:25:16][C][wifi:281]:   IP Address: 192.168.4.1
[19:25:16][C][api:034]: Running setup
[19:25:16][W][component:279]: api set Warning flag: unspecified
[19:25:16][D][esp-idf:000]: I (61349) mdns_mem: mDNS task will be created from internal RAM
[19:25:16][I][app:091]: setup() finished successfully!
[19:25:16][I][safe_mode:042]: Boot seems successful; resetting boot loop counter
[19:25:16][D][esp32.preferences:142]: Writing 1 items: 0 cached, 1 written, 0 failed
[19:25:16][I][app:149]: ESPHome version 2025.7.3 compiled on Jul 24 2025, 19:18:33
[19:25:16][C][wifi:613]: WiFi:
[19:25:16][D][esp-idf:000][wifi]: W (61376) wifi:
[19:25:16][D][esp-idf:000][wifi]: Haven't to connect to a suitable AP now!
[19:25:16][D][esp-idf:000][wifi]: 
[19:25:16][W][wifi_esp32:965]: esp_wifi_sta_get_ap_info failed: ESP_ERR_WIFI_NOT_CONNECT
[19:25:16][C][wifi:434]:   Local MAC: 30:C6:F7:99:17:A8
[19:25:16][D][esp-idf:000][wifi]: W (61406) wifi:
[19:25:16][D][esp-idf:000][wifi]: Haven't to connect to a suitable AP now!
[19:25:16][D][esp-idf:000][wifi]: 
[19:25:16][W][wifi_esp32:975]: esp_wifi_sta_get_ap_info failed: ESP_ERR_WIFI_NOT_CONNECT
[19:25:16][C][wifi:439]:   SSID: [redacted]
[19:25:16][C][wifi:442]:   IP Address: 192.168.30.81
[19:25:16][D][esp-idf:000][wifi]: W (61437) wifi:
[19:25:16][D][esp-idf:000][wifi]: Haven't to connect to a suitable AP now!
[19:25:16][D][esp-idf:000][wifi]: 
[19:25:16][W][wifi_esp32:986]: esp_wifi_sta_get_ap_info failed: ESP_ERR_WIFI_NOT_CONNECT
[19:25:16][C][wifi:446]:   BSSID: [redacted]
[19:25:16]  Hostname: 'humiditytempsensor'
[19:25:16]  Signal strength: 0 dB ▂▄▆█
[19:25:16][C][wifi:455]:   Channel: 1
[19:25:16]  Subnet: 255.255.255.0
[19:25:16]  Gateway: 192.168.30.1
[19:25:16]  DNS1: 192.168.1.1
[19:25:16]  DNS2: 0.0.0.0
[19:25:16][C][logger:246]: Logger:
[19:25:16]  Max Level: DEBUG
[19:25:16]  Initial Level: DEBUG
[19:25:16][C][logger:252]:   Log Baud Rate: 115200
[19:25:16]  Hardware UART: UART0
[19:25:16][C][logger:259]:   Task Log Buffer Size: 768
[19:25:16][C][i2c.idf:083]: I2C Bus:
[19:25:16][C][i2c.idf:084]:   SDA Pin: GPIO21
[19:25:16]  SCL Pin: GPIO22
[19:25:16]  Frequency: 50000 Hz
[19:25:16][C][i2c.idf:094]:   Recovery: bus successfully recovered
[19:25:16][I][i2c.idf:104]: Results from bus scan:
[19:25:16][I][i2c.idf:110]: Found device at address 0x44
[19:25:16][C][sht3xd:049]: SHT3xD:
[... 245 lines cut ...]
[19:28:21][D][esp-idf:000][wifi]: Password length matches WPA2 standards, authmode threshold changes from OPEN to WPA2
[19:28:21][D][esp-idf:000][wifi]: 
[19:28:23][D][sht3xd:097]: Got temperature=27.76°C humidity=42.32%
[19:28:23][D][sensor:104]: 'test_temp_1': Sending state 27.75578 °C with 1 decimals of accuracy
[19:28:23][D][sensor:104]: 'temp_humidity_1': Sending state 42.31784 % with 1 decimals of accuracy
[19:28:24][W][wifi_esp32:710]: Disconnected ssid='ccr-iot' bssid=[redacted] reason='Unspecified'
[19:28:24][W][wifi:666]: Connecting to network failed
[19:28:24][I][wifi:319]: Connecting to 'ccr-iot'
[19:28:24][D][esp-idf:000][wifi]: W (249248) wifi:
[19:28:24][D][esp-idf:000][wifi]: Password length matches WPA2 standards, authmode threshold changes from OPEN to WPA2
[19:28:24][D][esp-idf:000][wifi]: 
[19:28:27][W][wifi_esp32:710]: Disconnected ssid='ccr-iot' bssid=[redacted] reason='Unspecified'
[19:28:27][W][wifi:666]: Connecting to network failed
[19:28:32][I][wifi:319]: Connecting to 'ccr-iot'
[19:28:32][D][esp-idf:000][wifi]: W (257138) wifi:
[19:28:32][D][esp-idf:000][wifi]: Password length matches WPA2 standards, authmode threshold changes from OPEN to WPA2
[19:28:32][D][esp-idf:000][wifi]: 
[19:28:35][W][wifi_esp32:710]: Disconnected ssid='ccr-iot' bssid=[redacted] reason='Unspecified'
[19:28:35][W][wifi:666]: Connecting to network failed
[19:28:35][I][wifi:319]: Connecting to 'ccr-iot'
[19:28:35][D][esp-idf:000][wifi]: W (260011) wifi:
[19:28:35][D][esp-idf:000][wifi]: Password length matches WPA2 standards, authmode threshold changes from OPEN to WPA2
[19:28:35][D][esp-idf:000][wifi]: 
[19:28:38][W][wifi_esp32:710]: Disconnected ssid='ccr-iot' bssid=[redacted] reason='Unspecified'
[19:28:38][W][wifi:666]: Connecting to network failed```

ssieb, 2025-07-25:
Add the following to your yaml, install it, then paste just around the disconnect line:
```yaml
external_components:
  - source:
      type: git
      url: https://github.com/ssieb/esphome
      ref: wifi
    components: [ wifi ]
    refresh: 1min
```

crudolphy, 2025-07-29:
> So that reason is:
> WIFI_REASON_NO_AP_FOUND_IN_AUTHMODE_THRESHOLD = 211, /**< No AP found in
> auth mode threshold */
>
> Possible reasons for this:
>
> NO_AP_FOUND_IN_AUTHMODE_THRESHOLD
>
> Received WPA3-Enterprise disable transition bit, required authmode is greater than WPA2.(disable wpa2 transition)
> Received WPA3-Personal disable transition bit, required authmode is greater than WPA2.(disable wpa2 transition)
> OWE transition is disabled, reject legacy open bss profile.(disable owe transition)
> authmode threshold failure.
>
> You'll need to check your AP settings.

Dear Sir:

First I want to say thank you very much for your help.  Your investigation
and subsequently pointing me to my AP's settings were right on the money.

I had my ap set to "WPA-Personal" which allowed two different ciphers
WPA-PSK:Auto and WPA-PSK2: AES.  I used Google Gemini to get an explanation
first of the log messages which confirmed your instructions to look at my
AP settings.  Then when I put Gemini to work with the different choices I
had available to me it suggested I change from the WPA-PSK:Auto  to the
WPA-PSK2:AES.

I did this and then re-installed the configuration via USB and it connected
to the wifi.  I then updated the ESPHome Device Builder to the latest
version 2024.7.4 and then updated both this ESP32 device but also an
ESP8266 device without issue.

I have to tell you that this also solved another problem.  I have several
shelly 1PM in use.  (4) are older models and the 5th is a new model.  When
I re-configured my AP's a while ago I could get the (4) older ones to login
to the AP but couldn't get the 5th.  No matter what I tried it wouldn't
work.  So I gave up.  As soon as I switched to WPA2-PSK:AES it logged into
the AP and became functional and also is now available again in my Home
Assistant instance.

Can't tell you how much I appreciate your help with this.

Regards
Chuck Rudolphy

ssieb, 2025-07-29:
You're welcome.

closed 2025-07-29 by ssieb