esphome/esphome #9695: ESPHome CLI fails with Protobuf version error

donburch888, 2025-07-18:
### The problem

I normally use HA ESPHome integration, and Device Builder 2025.7.0 is still working correctly ... except that the Wi-Fi at my test bench is unreliable ... so sometimes I use ESPHome CLI on my desktop PC with USB cable. USB is also necessary to see logs during in_boot and on_shutdown.

This issue relates to the CLI version of ESPHome, running on Ubuntu Linux.

Yesterday one of the platformio files got corrupted, and I ended up removing and reinstalling ESPHome and platformio.

However compiling and running a simple test program gives me:

(ESPHome) don@Ponder:~/ESPHome$ esphome run s3-mini.yaml
INFO ESPHome 2025.7.1
INFO Reading configuration s3-mini.yaml...
INFO Detected timezone 'Australia/Sydney'
INFO Generating C++ source...
INFO Core config, version changed, cleaning build files...
INFO Deleting /home/don/ESPHome/.esphome/build/s3-mini/.pioenvs
INFO Compiling app...
Processing s3-mini (board: esp32-s3-devkitm-1; framework: espidf; platform: https://github.com/pioarduino/platform-espressif32/releases/download/53.03.13/platform-espressif32.zip)
--------------------------------------------------------------------------------------------------------------------------------
Library Manager: Installing esphome/noise-c @ 0.1.10
INFO Installing esphome/noise-c @ 0.1.10
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Library Manager: noise-c@0.1.10 has been installed!
INFO noise-c@0.1.10 has been installed!
Library Manager: Resolving dependencies...
INFO Resolving dependencies...
Library Manager: Installing esphome/libsodium @ 1.10020.7
INFO Installing esphome/libsodium @ 1.10020.7
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Library Manager: libsodium@1.10020.7 has been installed!
INFO libsodium@1.10020.7 has been installed!
Library Manager: Installing bblanchon/ArduinoJson @ 7.4.2
INFO Installing bblanchon/ArduinoJson @ 7.4.2
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Library Manager: ArduinoJson@7.4.2 has been installed!
INFO ArduinoJson@7.4.2 has been installed!
HARDWARE: ESP32S3 240MHz, 320KB RAM, 8MB Flash
 - framework-espidf @ 3.50302.0 (5.3.2) 
 - tool-cmake @ 3.30.2 
 - tool-esptoolpy @ 4.8.6 
 - tool-mklittlefs @ 3.2.0 
 - tool-ninja @ 1.7.1 
 - tool-riscv32-esp-elf-gdb @ 14.2.0+20240403 
 - tool-xtensa-esp-elf-gdb @ 14.2.0+20240403 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-riscv32-esp @ 13.2.0+20240530 
 - toolchain-xtensa-esp-elf @ 13.2.0+20240530
Reading CMake configuration...
Generating assembly for certificate bundle...
Dependency Graph
|-- noise-c @ 0.1.10
|-- ArduinoJson @ 7.4.2
Compiling .pioenvs/s3-mini/src/esphome/components/api/api_connection.cpp.o
Compiling .pioenvs/s3-mini/src/esphome/components/api/api_frame_helper.cpp.o
Compiling .pioenvs/s3-mini/src/esphome/components/api/api_pb2.cpp.o
Compiling .pioenvs/s3-mini/src/esphome/components/api/api_pb2_service.cpp.o
Compiling .pioenvs/s3-mini/src/esphome/components/api/api_server.cpp.o

then lots of compiling and

Compiling .pioenvs/s3-mini/lib927/noise-c/protocol/symmetricstate.o
Compiling .pioenvs/s3-mini/lib927/noise-c/protocol/util.o
Archiving .pioenvs/s3-mini/lib927/libnoise-c.a
Indexing .pioenvs/s3-mini/lib927/libnoise-c.a
Linking .pioenvs/s3-mini/firmware.elf
RAM:   [=         ]  11.3% (used 37080 bytes from 327680 bytes)
[... 238 lines cut ...]
    humidity:
      name: $devicename Control box Humidity
    update_interval: $update_interval_sensor
```

### Anything in the logs that might be useful for us?

```txt

```

### Additional information

I have verified that the error occurs in the esphome logs s3-mini.yaml phase by running each phase separately.

Following the link in the error message reveals:

    Starting with the 2025Q1 release, the protobuf project will adopt a rolling compatibility window for major versions. Code generated for a major version V (full version: V.x.y) will be supported by protobuf runtimes of version V and V+1.

I note a number of Home Assistant github issues in the past few months relating to protobuf version issues, including:

- Protobuf py version warnings - https://github.com/home-assistant/core/issues/148296
- protobuf-5.29.2 to protobuf-5.29.3 Update needed - https://github.com/home-assistant/core/issues/144679
- /usr/local/lib/python3.13/site-packages/google/protobuf/runtime_version - https://github.com/home-assistant/core/issues/146856
- Python deprecation warning for integration: androidtv_remote - https://github.com/home-assistant/core/issues/146713

donburch888, 2025-07-19:
Just saw ESPHome Device Builder upgrade to 2025.7.2 and upgraded.  Still able to compile, upload, and log my S3-mini.yaml from within Home Assistant without error. 

Went back to my Ubuntu linux CLI and yes, there is an update to ESPHome 2025.7.2 and `pip install -U esphome` did the upgrade.  
Trying `(ESPHome) don@Ponder:~/ESPHome$ esphome run s3-mini.yaml` gave `INFO ESPHome 2025.7.2` and after all the compiling finished again with
```
INFO Successfully uploaded program.
Traceback (most recent call last):
  File "/home/don/ESPHome/bin/esphome", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/home/don/ESPHome/lib/python3.12/site-packages/esphome/__main__.py", line 1087, in main
    return run_esphome(sys.argv)
           ^^^^^^^^^^^^^^^^^^^^^
  File "/home/don/ESPHome/lib/python3.12/site-packages/esphome/__main__.py", line 1074, in run_esphome
    rc = POST_CONFIG_ACTIONS[args.command](args, config)
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/don/ESPHome/lib/python3.12/site-packages/esphome/__main__.py", line 550, in command_run
    return show_logs(config, args, port)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/don/ESPHome/lib/python3.12/site-packages/esphome/__main__.py", line 401, in show_logs
    return run_miniterm(config, port, args)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/don/ESPHome/lib/python3.12/site-packages/esphome/__main__.py", line 135, in run_miniterm
    from aioesphomeapi import LogParser
  File "/home/don/ESPHome/lib/python3.12/site-packages/aioesphomeapi/__init__.py", line 2, in <module>
    from .api_pb2 import (  # type: ignore[attr-defined] # noqa: F401
  File "/home/don/ESPHome/lib/python3.12/site-packages/aioesphomeapi/api_pb2.py", line 13, in <module>
    _runtime_version.ValidateProtobufRuntimeVersion(
  File "/home/don/ESPHome/lib/python3.12/site-packages/google/protobuf/runtime_version.py", line 113, in ValidateProtobufRuntimeVersion
    _ReportVersionError(
  File "/home/don/ESPHome/lib/python3.12/site-packages/google/protobuf/runtime_version.py", line 50, in _ReportVersionError
    raise VersionError(msg)
google.protobuf.runtime_version.VersionError: Detected incompatible Protobuf Gencode/Runtime versions when loading api.proto: gencode 5.29.5 runtime 5.29.3. Runtime version cannot be older than the linked gencode version. See Protobuf version guarantees at https://protobuf.dev/support/cross-version-runtime-guarantee.
```

bdraco, 2025-07-19:
Thats solved in newer aioesphomeapi which will be in ESPHome 2025.8.x.

In the mean time you could `pip install aioesphomeapi==35.0.0`

https://github.com/esphome/aioesphomeapi/releases/tag/v35.0.0

donburch888, 2025-07-19:
Yes, the `pip install` command has got my CLI ESPHome able to display logs ... despite the error message at end of the install:
```
ERROR: pip's dependency resolver does not currently take into account all the packages that are installed. This behaviour is the source of the following dependency conflicts.
esphome 2025.7.2 requires aioesphomeapi==34.2.1, but you have aioesphomeapi 35.0.0 which is incompatible.
```

Thank you for the prompt attention.

closed 2025-07-19 by donburch888