esphome/esphome #9700: Update to 2025.7 breaks binary_sensor w/ allow_other_uses and duty_cycle

neilbrookins, 2025-07-19:
### The problem

Input binary sensor never changes state when GPIO changes state.
When using a single GPIO pin with two different platforms we specify allow_other_uses: true to permit this case. I'm using a binary_sensor of device_class: running as the platform: gpio for the first use. The same pin gets a second use with platform: duty_cycle This used to work just fine; but after 2025.7.0 the binary_sensor always shows not running, even when the GPIO changes state. Despite this failure, the duty_cycle still shows the correct values. Which is conflicting information when it says 100% duty at the same time as not running. When the GPIO changes state again, the duty goes back to 0% and the binary_sensor still says not running.  If I remove/comment out the duty_cycle YAML and keep only the binary_sensor then it works correctly and shows all the state changes.

### Which version of ESPHome has the issue?

2025.7.0 and 2025.7.1

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP8266

### Component causing the issue

gpio duty_cycle

### YAML Config

```yaml
binary_sensor:
  - platform: gpio
    pin:
      number: D1
#     allow_other_uses: true
      mode: INPUT_PULLUP
    name: "E-one sewer pump"
    filters:
      - delayed_on: 100ms
      - delayed_off: 100ms
    device_class: running
sensor: 
# - platform: duty_cycle
#   pin:
#     number: D1
#     allow_other_uses: True
#     mode: INPUT_PULLUP
#   name: "E-one sewer duty cycle"

#The above example works correctly as shown, with just binary_sensor defined. 
#To reproduce the error, un-comment all the above lines so that you have BOTH the binary_sensor AND the duty_cycle on the SAME pin.
```

### Anything in the logs that might be useful for us?

```txt
no errors that I noticed.
```

### Additional information

[image]

In the above screen shot from the ESPHome web interface the values for "E-one sewer duty cycle" and "E-one sewer pump" both are derived from GPIO pin D1. This pin cannot be "OFF" when the duty cycle is 100% which indicates its "ON".  It should say "ON" here to match the duty_cycle which shows correctly that it is on.  I simulated the ON state by disconnecting the GPIO pin which is set to pull up when open circuit. Meaning there is no device connected to the GPIO pin when this issue is reproduced, proving the issue is not external.

ssieb, 2025-07-19:
You shouldn't really be doing that, but if you must, you will need to disable the interrupt on the binary sensor.

neilbrookins, 2025-07-19:
I added:
    use_interrupt: False
and now its working perfectly. Thanks a million!

PR #9701 (2025-07-19): [gpio] Auto-disable interrupts for shared GPIO pins in binary sensors
# What does this implement/fix?

This PR automatically disables interrupts for GPIO binary sensors when the pin is shared with other components (i.e., when `allow_other_uses: true` is set). This fixes an issue where binary sensors stop working correctly when sharing a pin with components like duty_cycle sensors.

Previously, users had to manually set `use_interrupt: false` to resolve conflicts. This change makes it automatic, improving the user experience and preventing hard-to-diagnose issues.

## Types of changes

- [x] Bu [... cut]
closed 2025-07-19 by kbx81