esphome/esphome #9709: Reset: Hardware Watchdog|Fatal exception:4 flag:1 (Hardware Watchdog) epc1:0x401038f5 epc2:0x00000000 epc3:0x00000

andreadegiovine, 2025-07-19:
### The problem

I have 3 UNICAL air conditioners (rebranded midea), for all of them I receive this error randomly.

For none of the 3 I receive the energy consumption (via midea app for 1 it worked).

For 2 air conditioners I cannot set the ECO preset.

The loop is about 50-60ms.

What data can be used to help the analysis?

### Which version of ESPHome has the issue?

2025.5.2

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

### Component causing the issue

midea

### YAML Config

```yaml
substitutions:
  name: clima-soggiorno
  friendly_name: "Clima soggiorno"

  
esphome:
  name: "${name}"
  friendly_name: "${friendly_name}"
  project:
    name: SMLIGHT.SLWF-01Pro
    version: "2.1"

esp8266:
  board: esp12e

ota:
  - platform: esphome
  
  
wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  power_save_mode: NONE
  reboot_timeout: 0s

api:
  reboot_timeout: 0s
  services:
    - service: follow_me
      variables:
        temperature: float
      then:
        midea_ac.follow_me:
          temperature: !lambda "return temperature;"
          beeper: false

uart:
  tx_pin: 12
  rx_pin: 14
  baud_rate: 9600

remote_transmitter:
  pin: GPIO13
  carrier_duty_percent: 100%
          

climate:
  - platform: midea
    name: ""
    id: midea_climate
    period: 1s
    timeout: 2s
    num_attempts: 3
    autoconf: false
    beeper: false
    visual:
      min_temperature: "17 °C"
      max_temperature: "30 °C"
      temperature_step: "1 °C"
    supported_modes:
      - HEAT_COOL
      - COOL
      - HEAT
      - DRY
    supported_presets:
      - ECO
    supported_swing_modes:
      - VERTICAL
      - HORIZONTAL
      - BOTH
    outdoor_temperature:
      name: Outdoor Temperature

switch:
  - platform: template
    name: Eco
    id: midea_eco
    icon: mdi:sprout
    lambda: 'return id(midea_climate).preset == CLIMATE_PRESET_ECO;'
    turn_on_action:
      remote_transmitter.transmit_toshiba_ac:
        rc_code_1: !lambda |-
            uint8_t temp = (uint8_t) id(midea_climate).target_temperature;
            uint8_t eco_prefix = 0x82;  // ON
            uint8_t temp_byte = 0x40 + (temp - 24) * 0x80;
            uint64_t code = (uint64_t(0xB54AF50A0000) | (eco_prefix << 8) | temp_byte);
            return code;
    turn_off_action:
      remote_transmitter.transmit_toshiba_ac:
        rc_code_1: !lambda |-
            uint8_t temp = (uint8_t) id(midea_climate).target_temperature;
            uint8_t eco_prefix = 0x83;  // OFF
            uint8_t temp_byte = 0x40 + (temp - 24) * 0x80;
            uint64_t code = (uint64_t(0xB54AF50A0000) | (eco_prefix << 8) | temp_byte);
            return code;

button:
  - platform: template
    name: Swing vertical
    id: midea_swing_vertical
    icon: mdi:compare-vertical
    on_press:
      remote_transmitter.transmit_toshiba_ac:
        rc_code_1: 0xB24D0FF0E01F

  - platform: template
    name: Swing horizontal
    id: midea_swing_horizontal
    icon: mdi:compare-horizontal
    on_press:
      remote_transmitter.transmit_toshiba_ac:
        rc_code_1: 0xB54AF50AB24D

  - platform: template
    name: Display
    id: midea_display
    icon: mdi:theme-light-dark
    on_press:
      remote_transmitter.transmit_toshiba_ac:
        rc_code_1: 0xB54AF50AA55A
```

### Anything in the logs that might be useful for us?

```txt

```

### Additional information

https://github.com/dudanov/MideaUART/issues/24

andreadegiovine, 2025-08-26:
@bdraco @jesserockz @clydebarrow 
Can anyone help me?

ssieb, 2025-08-26:
Why are you overriding it instead of letting the component transmit the commands?

andreadegiovine, 2025-08-26:
Eco from component work only in 1 of 3 devices

Display not work

Swing i prefer 2 direction action

closed 2025-12-02 by github-actions[bot]