esphome/esphome #9710: Some i2c devices not working after 2025.7.0

zumbefin, 2025-07-19:
### The problem

Hi, I have SHT40,bme280 and bh1750 connected to esp32. After 2025.7.0 update they remain as unknown devices.

bh1750 gives in log
[W][bh1750.sensor:063]: Set measurement time failed

bme280 gives component mark as failed in some boots and in some it initializes correctly, but does not report any data.

SHT40 works normally.

If I restore the previous version 2025.6.3 everything starts working normally.

The same problem is still in the latest 2025.7.2 version

### Which version of ESPHome has the issue?

2025.7.0

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

### Component causing the issue

i2c bme280, bh1750

### YAML Config

```yaml
esphome:
  name: "takapiha-keskus"
  friendly_name: Takapiha-keskus

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:
  baud_rate: 0

# Enable Home Assistant API
api:  
  #encryption: 
    #key: "*sensored*"

#mqtt:
  #broker: 10.1.1.5
  #username: takapiha

ota:
  - platform: esphome
    password: *sensored*

safe_mode:
  boot_is_good_after: 90s
  num_attempts: 3

preferences:
  flash_write_interval: 10min  

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  power_save_mode: none

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Takapiha Fallback Hotspot"
    password: "*sensored*"
  fast_connect: off

captive_portal:

#web_server:
  #port: 80
  #version: 3    

# i2C pins and setup
i2c:
  sda: GPIO21 # D21
  scl: GPIO22 # D22
  scan: false
  id: bus_a
      

#co2 kommunikaatio
uart:
  rx_pin: GPIO16
  tx_pin: GPIO17
  baud_rate: 9600  

#Valoisuus anturi BH1750
sensor:
  - platform: bh1750
    name: "takapiha valoisuus bh1750"
    address: 0x23
    update_interval: 60s

  - platform: sht4x
    temperature:
      name: "takapiha lämpötila sht40"
      accuracy_decimals: 2
    humidity:
      name: "takapiha kosteus sht40"
      accuracy_decimals: 2
    heater_max_duty: 0.05
    heater_power: High
    heater_time: Long  

  - platform: wifi_signal # Reports the WiFi signal strength/RSSI in dB
    name: "WiFi Signal dB"
    id: wifi_signal_db
    update_interval: 60s
    entity_category: "diagnostic"

  - platform: copy # Reports the WiFi signal strength in %
    source_id: wifi_signal_db
    name: "WiFi Signal Percent"
    filters:
      - lambda: return min(max(2 * (x + 100.0), 0.0), 100.0);
    unit_of_measurement: "Signal %"
    entity_category: "diagnostic"
    device_class: ""   
#CO2
  - platform: senseair
    co2:
      name: "SenseAir CO2 arvo ulkona"
      accuracy_decimals: 1
    update_interval: 60s

          

#sisäinen lämpötila
  - platform: internal_temperature
    name: "takapiha ESP lämpötila"    

  # BME280 Temperature, Pressure, Humidity sensor
  - platform: bme280_i2c
  
    temperature:
      name: "takapiha lämpötila bme280"
      
      accuracy_decimals: 2
    pressure:
      name: "takapiha ilmanpaine bme280"
    
      accuracy_decimals: 1
      filters:
        - clamp:
            min_value: 930
            max_value: 1070
            ignore_out_of_range: true
    humidity:
      name: "takapiha kosteus bme280"
    
      accuracy_decimals: 2
    address: 0x76
    
    

#restart nappi
switch:
  - platform: restart
    name: "Restart ESP32"
    id: restart_switch 
    
#CO2 lämmittimen rele
  - platform: gpio
    name: "CO2 lämmitys rele"
    pin: 
      number: GPIO4
      inverted: true
      mode: 
        output: True
        open_drain: True
    restore_mode: ALWAYS_OFF
```

### Anything in the logs that might be useful for us?

```txt

```

peterd550, 2025-07-20:
Try changing the type:

esp32:
board: esp32dev 
framework:
type: esp-idf

zumbefin, 2025-07-20:
Bme280 is now working, but bh1750 is giving:

[16:10:22][W][bh1750.sensor:091]: Start measurement failed
[16:10:22][W][component:279]: bh1750.sensor set Warning flag: unspecified
[16:11:22][W][bh1750.sensor:063]: Set measurement time failed
[16:11:22][D][sensor:104]: 'takapiha valoisuus bh1750': Sending state nan lx with 1 decimals of accuracy

peterd550, 2025-07-21:
I have resolved my issue.

A simple reboot of my home assistant and cleaned my build and install.
It did a heap of pip install's and compiling but it works now :)

closed 2025-07-21 by ssieb

zumbefin, 2025-07-21:
Not helping for me i did a build clean and full restart to ha bh1750 still not working.

ssieb, 2025-07-21:
Need to see serial logs from boot.

swoboda1337, 2025-09-20:
Is this still a problem @zumbefin