esphome/esphome #9712: 2025.7.2 has broken my TagReader

NeeeeB, 2025-07-19:
### The problem

Been using a tagreader from this github for years : 
https://github.com/adonno/tagreader

Has worked flawlessly until the last ESPHome update.
Now it is somewhat stuck in a bootloop because of what it looks like a wifi issue.
Already tried to reflash it multiple times, always the same result.

this is the log file, that I can access when device is plugged onto the same machine as ESPHomeis running on.

### Which version of ESPHome has the issue?

2025.7.2

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP8266

### YAML Config

```yaml

```

### Anything in the logs that might be useful for us?

```txt
See attached log file
```

### Additional information

[logs_tagreader-462833_run.txt](https://github.com/user-attachments/files/21327590/logs_tagreader-462833_run.txt)

bdraco, 2025-07-19:
It looks like bug in the ESP8266 Arduino core's WiFiUDP implementation

bdraco, 2025-07-19:
The stack trace is missing a few pieces.  It might be that the binary flashed was not the latest one on the local system or the debug systems are missing for where is crashing which makes it hard to figure out whats going on.

Can you try to get a better back trace ? https://esphome.io/guides/troubleshooting.html

NeeeeB, 2025-07-19:
> The stack trace is missing a few pieces. It might be that the binary flashed was not the latest one on the local system or the debug systems are missing for where is crashing which makes it hard to [...]

Followed the link
Installed ESPHome on my computer (alongside python and git)
Downloaded the yaml
Ran the different commands : 
esphome compile xxxx.yaml
esphome upload xxxx.yaml
esphome logs xxxx.yaml

Got exactly the same stack trace....

```
[22:27:29]--------------- CUT HERE FOR EXCEPTION DECODER ---------------

[22:27:29]Exception (28):
WARNING Exception type: Access to invalid address: LOAD (wild pointer?)
[22:27:29]epc1=0x4022847a epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000014 depc=0x00000000
WARNING Decoded 0x4022847a: WiFiUDP::~WiFiUDP()

[22:27:29]>>>stack>>>
WARNING Found stack trace! Trying to decode it

[22:27:29]ctx: cont
[22:27:29]sp: 3ffffdc0 end: 3fffffd0 offset: 0150
[22:27:29]3fffff10:  3fff48e8 3fff48dc 00000000 00000000
[22:27:29]3fffff20:  3ffe9906 3fff1a5c 3fff195c 4021ad54
WARNING Decoded 0x4021ad54: esphome::wifi::WiFiComponent::check_connecting_finished()
[22:27:29]3fffff30:  3fffff38 00000007 2d544f49 006e6542
[22:27:29]3fffff40:  3fff5a44 00000001 4022efa8 4023acb4
WARNING Decoded 0x4022efa8: std::_Function_handler<void (), std::_Bind<bool (esp8266::MDNSImplementation::MDNSResponder::*(esp8266::MDNSImplementation::MDNSResponder*))()> >::_M_manager(std::_Any_data&, std::_Any_data const&, std::_Manager_operation)
WARNING Decoded 0x4023acb4: std::_Function_handler<void (), std::_Bind<bool (esp8266::MDNSImplementation::MDNSResponder::*(esp8266::MDNSImplementation::MDNSResponder*))()> >::_M_invoke(std::_Any_data const&)
[22:27:29]3fffff50:  3fff5b54 00000000 3fff0224 3ffefef8
[22:27:29]3fffff60:  0000f2a5 3fff1a5c 3fff195c 4021af2c
WARNING Decoded 0x4021af2c: esphome::wifi::WiFiComponent::loop()
[22:27:29]3fffff70:  00000010 0000f294 00000001 3fff195c
[22:27:29]3fffff80:  00000000 0000f2a5 3ffefef8 4021c573
WARNING Decoded 0x4021c573: esphome::Application::loop()
[22:27:29]3fffff90:  0000f2a5 3fff195c 00000060 3fff05ec
[22:27:29]3fffffa0:  3fffdad0 00000000 3fff05c0 3fff05ec
[22:27:29]3fffffb0:  3fffdad0 00000000 3fff05c0 40235f00
WARNING Decoded 0x40235f00: loop_wrapper() at core_esp8266_main.cpp
[22:27:29]3fffffc0:  feefeffe feefeffe 3fffdab0 40100569
WARNING Decoded 0x40100569: cont_wrapper
[22:27:29]<<<stack<<<

[22:27:29]--------------- CUT HERE FOR EXCEPTION DECODER ---------------
```

NeeeeB, 2025-07-29:
Is there any way to help resolving this ?
Missing my TagReader is a big problem unfortunately...

NeeeeB, 2025-08-19:
> Without a complete trace there isn't enough information to work on a solution.

I totally understand this, but I did as you requested and could not get a better stacktrace...
So it seems we are stuck unfortunately.

bdraco, 2025-08-19:
Can you try compiling with the logger set to VERBOSE ?

bdraco, 2025-08-19:
You didn't post your config either.  I suspect disable ap: in your config will probably work around the problem

pmannk, 2025-08-20:
If it helps you I have a NodeMCU (esp8266) with an RC522 that's working fine on 2025.7.5. Here's the config I'm using:

```
substitutions:
  hostname: 'nodemcu-d36b6d.local'
  devname: 'nodemcu_d36b6d'
  description: 'RFID NodeMCU'
  fastconnect: 'false'

esp8266:
  board: 'nodemcuv2'

esphome:
  name: ${devname}
  comment: ${description}
  on_boot:
    priority: 700.0
    then:
      - text_sensor.template.publish:
          id: ${devname}_last_read
          state: "00-00-00-00"

wifi:
  ssid: !
  password: !
  reboot_timeout: 0s
  use_address: ${hostname}
  fast_connect: ${fastconnect}
  ap:
    ssid: !
    password: !
    ap_timeout: 240min

captive_portal:

logger:
  level: DEBUG

api:
  reboot_timeout: 0s

ota:
  platform: esphome

web_server:
  port: 80

switch:
  - platform: restart
    name: ${devname} ESP Restart
    id: ${devname}_restart

  - platform: gpio #Standard status led (1)
    pin:
      number: GPIO16
      inverted: true
    name: ${devname} blueled1
    id: ${devname}_blueled1
    internal: yes

  - platform: gpio #Spare led(2)
    pin:
      number: GPIO2
      inverted: true
    name: ${devname} blueled2
    id: ${devname}_blueled2
    internal: yes

text_sensor:    
  - platform: template #Sensor to display last read card
    name: ${devname} Last Read
    id: ${devname}_last_read

spi:
  clk_pin: D5
  miso_pin: D6
  mosi_pin: D7

rc522_spi:
  cs_pin: D8
  update_interval: 1s
  on_tag:
    then:
      - if:
          condition:
            not:
              text_sensor.state:
                id: ${devname}_last_read
                state: !lambda 'return x;'
          then: #Action if a new/different tag was scanned
            - switch.turn_on: ${devname}_blueled1
            - homeassistant.tag_scanned: !lambda 'return x;'
            - text_sensor.template.publish:
                id: ${devname}_last_read
                state: !lambda 'return x;'
            - delay: 30s
            - text_sensor.template.publish:
                id: ${devname}_last_read
                state: "00-00-00-00"
            - switch.turn_off: ${devname}_blueled1
          else: #Action if a repeat tag was scanned
            - repeat:
                count: 10
                then:
                  - switch.toggle: ${devname}_blueled1
                  - delay: 100ms
```

NeeeeB, 2025-08-20:
> > I think I can just replace all the content of my yaml file with the one from the link, replace the substitutions with mine, set wifi to use my secrets.yaml. This I can and know how to do.
> 
> Yes, effectively merge the contents of tagreader.yaml with your config, giving priority to your specific preferences (name, name_add_mac_suffix and wifi ssid/password). Then comment out the ap: and its child ssid: sections. Also comment out the dashboard_import: and packages: sections so you don't pull in bits from the original tagreader.yaml.

Oh damn it worked.
Tagreader is up and running.
Also updated from 2.7.5 to 2.8.0 without problem.
Removing the "ap" section was the solution.
Thanks @bdraco and also thanks to you @pmannk for confirming the steps.

closed 2025-08-20 by NeeeeB