esphome/esphome #9714: Is anyone in control of the whole thing?

jacyker, 2025-07-19:
### The problem

It's hard not to get the impression that the project has become so large that it needs someone to oversee it comprehensively. 2025.7 and 2025.7.2 are another example of the lack of proper testing. I just adjusted the settings on the ESP32 TouchPad and the WS2812B, and now I see that the BME650 is also giving astronomical readings... And it looks like that's not the end of the surprises.

PS. Most of them are on IDF, so it's not like switching from Arduino 2.x to 3.x caused this...

### Which version of ESPHome has the issue?

2025.7

### What type of installation are you using?

Docker

### What platform are you using?

ESP32

### YAML Config

```yaml

```

### Anything in the logs that might be useful for us?

```txt

```

cracrama, 2025-07-19:
I have problem with BME680 reading since April and so far no solution, tried anything. Slowly switching all critical automations to different platform.

closed 2025-07-19 by bdraco

tbrasser, 2025-07-19:
You are not _forced_ to update everything all the time.

cracrama, 2025-07-19:
BME680 issue related problems  #7136 and #6924.