esphome/esphome #9715: [BUG] Undefined reference to LwIPLock when using mqtt and esp-idf (2025.7.2)

akawkilo, 2025-07-19:
### The problem

### The problem

When compiling ESPHome 2025.7.2 firmware using `mqtt:` component with `esp-idf` framework, the build fails at linking phase due to missing symbols related to `LwIPLock`.

### Build output

### Which version of ESPHome has the issue?

2025.7.2

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

### Component causing the issue

mqtt

### YAML Config

```yaml
esphome:
  name: sniffer2b

esp32:
  board: ttgo-lora32-v2
  framework:
    type: esp-idf

mqtt:
  broker: 192.168.1.56
  username: "test"
  password: "test"

logger:
  level: INFO
```

### Anything in the logs that might be useful for us?

```txt
INFO ESPHome 2025.7.2
INFO Reading configuration /config/esphome/sniffer2b.yaml...
INFO Updating https://github.com/SzczepanLeon/esphome-components.git@main
INFO Generating C++ source...
INFO Compiling app...
Processing sniffer2b (board: ttgo-lora32-v2; framework: espidf; platform: https://github.com/pioarduino/platform-espressif32/releases/download/53.03.13/platform-espressif32.zip)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - framework-espidf @ 3.50302.0 (5.3.2) 
 - tool-cmake @ 3.30.2 
 - tool-esptoolpy @ 4.8.6 
 - tool-mklittlefs @ 3.2.0 
 - tool-ninja @ 1.7.1 
 - tool-riscv32-esp-elf-gdb @ 14.2.0+20240403 
 - tool-xtensa-esp-elf-gdb @ 14.2.0+20240403 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-xtensa-esp-elf @ 13.2.0+20240530
Reading CMake configuration...
Dependency Graph
|-- ArduinoJson @ 7.4.2
Linking .pioenvs/sniffer2b/firmware.elf
/data/cache/platformio/packages/toolchain-xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: .pioenvs/sniffer2b/src/esphome/components/mqtt/mqtt_client.cpp.o:(.literal._ZN7esphome4mqtt19MQTTClientComponent16start_dnslookup_Ev+0xc): undefined reference to `_ZN7esphome8LwIPLockC1Ev'
/data/cache/platformio/packages/toolchain-xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: .pioenvs/sniffer2b/src/esphome/components/mqtt/mqtt_client.cpp.o:(.literal._ZN7esphome4mqtt19MQTTClientComponent16start_dnslookup_Ev+0x10): undefined reference to `_ZN7esphome8LwIPLockD1Ev'
/data/cache/platformio/packages/toolchain-xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: .pioenvs/sniffer2b/src/esphome/components/mqtt/mqtt_client.cpp.o: in function `_ZN7esphome4mqtt19MQTTClientComponent16start_dnslookup_Ev':
/data/build/sniffer2b/src/esphome/components/mqtt/mqtt_client.cpp:198:(.text._ZN7esphome4mqtt19MQTTClientComponent16start_dnslookup_Ev+0x24): undefined reference to `_ZN7esphome8LwIPLockC1Ev'
/data/cache/platformio/packages/toolchain-xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: /data/build/sniffer2b/src/esphome/components/mqtt/mqtt_client.cpp:203:(.text._ZN7esphome4mqtt19MQTTClientComponent16start_dnslookup_Ev+0x3e): undefined reference to `_ZN7esphome8LwIPLockD1Ev'
collect2: error: ld returned 1 exit status
*** [.pioenvs/sniffer2b/firmware.elf] Error 1
========================= [FAILED] Took 18.09 seconds =========================
```

jesserockz, 2025-07-19:
You are using an external component that is overriding the esp32 component which is where the lwiplock class is implemented. 
Remove the external component or update it

akawkilo, 2025-07-20:
Thank you for pointing this out.
Indeed, I was using an external component, which I overlooked during my initial analysis.
I should have tested MQTT discovery first using a basic YAML configuration without external components.
As a test, I selected all components except esp32 during compilation, and it compiled successfully.
Thank you for your help and clarification, which helped me identify the root cause.