esphome/esphome #9718: 2025.7.0 and greater:  esp32-s3-devkitc-1 will no longer compile, missing package.json in platform-espressif32.zip

jtrumpio, 2025-07-19:
### The problem

INFO ESPHome 2025.7.2
INFO Reading configuration /config/esphome/rv-fridge-display.yaml...
INFO Updating https://github.com/landonr/lilygo-tdisplays3-esphome.git@None
INFO Generating C++ source...
INFO Compiling app...
Processing rvfridgedisplay (board: esp32-s3-devkitc-1; framework: arduino; platform: https://github.com/pioarduino/platform-espressif32/releases/download/53.03.13/platform-espressif32.zip)
--------------------------------------------------------------------------------
MissingPackageManifestError: Could not find one of 'package.json' manifest files in the package

### Which version of ESPHome has the issue?

2025.7.0 and above

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

### YAML Config

```yaml
esp32:
  board: esp32-s3-devkitc-1
  framework:
    type: arduino
```

### Anything in the logs that might be useful for us?

```txt
INFO ESPHome 2025.7.2
INFO Reading configuration /config/esphome/rv-fridge-display.yaml...
INFO Updating https://github.com/landonr/lilygo-tdisplays3-esphome.git@None
INFO Generating C++ source...
INFO Compiling app...
Processing rvfridgedisplay (board: esp32-s3-devkitc-1; framework: arduino; platform: https://github.com/pioarduino/platform-espressif32/releases/download/53.03.13/platform-espressif32.zip)
--------------------------------------------------------------------------------
MissingPackageManifestError: Could not find one of 'package.json' manifest files in the package
```

peterd550, 2025-07-20:
Try changing the type.

esp32:
  board:esp32dev
    framework:
      type: esp-idf

jtrumpio, 2025-07-20:
Unfortunately that does not work with my board type.  

https://a.co/d/7rrsTok

peterd550, 2025-07-20:
Worth a try.
nice board.
What are you using it for?

jtrumpio, 2025-07-20:
It's a display / historian for the refrigerator/freezer in my rv.  It reads temp off the serial bus from a separate esp8266 that runs the compressors, heaters, and fans.  

[image]

ssieb, 2025-07-21:
If changing the framework doesn't fix it, then try re-installing the add-on.

jtrumpio, 2025-07-21:
Actually in looking back at what I did, changing the framework to esp-idf and back to arduino is what caused this.

Removing the add-on and reinstalling does seem to have resolved this.