esphome/esphome #9721: m5stack-core-esp32 does not compile

jipp, 2025-07-19:
### The problem

using using board: m5stack-core-esp32

Running the yaml config below is not compiling as expected.
Same config does not compile with 2025.7.x anymore. 
It compiles with 2025.6.x.

### Which version of ESPHome has the issue?

2025.7.x

### What type of installation are you using?

Docker

### What platform are you using?

ESP32

### Component causing the issue

board: m5stack-core-esp32

### YAML Config

```yaml
esphome:
  name: woke

esp32:
  board: m5stack-core-esp32
#  framework:
#    type: esp-idf

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: "ota1234"

ota:
  - platform: esphome
    password: "ota1234"

wifi:
  ssid: "wobilix"
  password: "12345678"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Woke Fallback Hotspot"
    password: "yEWAOkTrROeC"

captive_portal:
```

### Anything in the logs that might be useful for us?

```txt
docker run --rm -v "${PWD}":/config -it esphome/esphome compile woke.yaml
INFO ESPHome 2025.7.2
INFO Reading configuration woke.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing woke (board: m5stack-core-esp32; framework: arduino; platform: https://github.com/pioarduino/platform-espressif32/releases/download/53.03.13/platform-espressif32.zip)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - framework-arduinoespressif32 @ 3.1.3 
 - framework-arduinoespressif32-libs @ 5.3.0+sha.489d7a2b3a 
 - tool-esptoolpy @ 4.8.6 
 - tool-mklittlefs @ 3.2.0 
 - tool-riscv32-esp-elf-gdb @ 14.2.0+20240403 
 - tool-xtensa-esp-elf-gdb @ 14.2.0+20240403 
 - toolchain-xtensa-esp-elf @ 13.2.0+20240530
Installing Arduino Python dependencies
Collecting wheel>=0.35.1
  Downloading wheel-0.45.1-py3-none-any.whl.metadata (2.3 kB)
Downloading wheel-0.45.1-py3-none-any.whl (72 kB)
Installing collected packages: wheel
Successfully installed wheel-0.45.1
Dependency Graph
|-- Networking @ 3.1.3
|-- AsyncTCP @ 3.4.5
|-- WiFi @ 3.1.3
|-- FS @ 3.1.3
|-- Update @ 3.1.3
|-- ESPAsyncWebServer @ 3.7.10
|-- ESP32 Async UDP @ 3.1.3
|-- DNSServer @ 3.1.3
|-- ESPmDNS @ 3.1.3
Compiling .pioenvs/woke/src/esphome/components/api/api_connection.cpp.o
Compiling .pioenvs/woke/src/esphome/components/api/api_frame_helper.cpp.o
Compiling .pioenvs/woke/src/esphome/components/api/api_pb2.cpp.o
Compiling .pioenvs/woke/src/esphome/components/api/api_pb2_service.cpp.o
In file included from src/esphome/core/macros.h:7,
                 from src/esphome/core/defines.h:2,
                 from src/esphome/components/api/api_connection.h:3,
                 from src/esphome/components/api/api_connection.cpp:1:
/config/.esphome/platformio/packages/framework-arduinoespressif32/cores/esp32/Arduino.h:43:10: fatal error: pins_arduino.h: No such file or directory

**********************************************************************
* Looking for pins_arduino.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:pins_arduino.h"
* Web  > https://registry.platformio.org/search?q=header:pins_arduino.h
*
**********************************************************************

   43 | #include "pins_arduino.h"
      |          ^~~~~~~~~~~~~~~~
compilation terminated.
In file included from src/esphome/core/macros.h:7,
                 from src/esphome/core/defines.h:2,
                 from src/esphome/components/api/api_pb2.h:5,
                 from src/esphome/components/api/api_pb2.cpp:3:
/config/.esphome/platformio/packages/framework-arduinoespressif32/cores/esp32/Arduino.h:43:10: fatal error: pins_arduino.h: No such file or directory

**********************************************************************
* Looking for pins_arduino.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:pins_arduino.h"
* Web  > https://registry.platformio.org/search?q=header:pins_arduino.h
*
**********************************************************************

   43 | #include "pins_arduino.h"
      |          ^~~~~~~~~~~~~~~~
compilation terminated.
*** [.pioenvs/woke/src/esphome/components/api/api_pb2.cpp.o] Error 1
*** [.pioenvs/woke/src/esphome/components/api/api_connection.cpp.o] Error 1
In file included from src/esphome/core/macros.h:7,
                 from src/esphome/core/defines.h:2,
                 from src/esphome/components/api/api_pb2_service.h:5,
                 from src/esphome/components/api/api_pb2_service.cpp:3:
/config/.esphome/platformio/packages/framework-arduinoespressif32/cores/esp32/Arduino.h:43:10: fatal error: pins_arduino.h: No such file or directory

**********************************************************************
* Looking for pins_arduino.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:pins_arduino.h"
* Web  > https://registry.platformio.org/search?q=header:pins_arduino.h
*
**********************************************************************

   43 | #include "pins_arduino.h"
      |          ^~~~~~~~~~~~~~~~
compilation terminated.
*** [.pioenvs/woke/src/esphome/components/api/api_pb2_service.cpp.o] Error 1
In file included from src/esphome/core/macros.h:7,
                 from src/esphome/core/defines.h:2,
                 from src/esphome/components/api/api_frame_helper.h:9,
                 from src/esphome/components/api/api_frame_helper.cpp:1:
/config/.esphome/platformio/packages/framework-arduinoespressif32/cores/esp32/Arduino.h:43:10: fatal error: pins_arduino.h: No such file or directory

**********************************************************************
* Looking for pins_arduino.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:pins_arduino.h"
* Web  > https://registry.platformio.org/search?q=header:pins_arduino.h
*
**********************************************************************

   43 | #include "pins_arduino.h"
      |          ^~~~~~~~~~~~~~~~
compilation terminated.
*** [.pioenvs/woke/src/esphome/components/api/api_frame_helper.cpp.o] Error 1
================================================================================ [FAILED] Took 3.61 seconds ================================================================================
```

### Additional information

same yaml works with the below command:
docker run --rm -v "${PWD}":/config -it esphome/esphome:2025.6 compile woke.yaml

also uncommenting the framework:
#  framework:
#    type: esp-idf

will result in a successful compile

bdraco, 2025-07-19:
I expect the board def changed in the new arduino version

jesserockz, 2025-07-19:
The board was renamed in Arduino 3 to m5stack_core

jesserockz, 2025-07-20:
I don't know why but the board variants changed in Arduino, and now the platformio code doesn't link to a variant in Arduino properly, causing this error. 

You can safely just change to esp32dev as the board. 

ESPHome 2025.8.0 will remove the requirement to specify board for esp32, and simply allow a variant to be provided (esp32, esp32s3 etc)

It all works in esp-idf because boards don't mean anything

closed 2025-07-20 by jesserockz