esphome/esphome #9732: ESPHome fails to complete startup with poor WiFi

dwmw2, 2025-07-20:
### The problem

I have a blind controller running in a part of the house with very poor wireless coverage. Last updated with git HEAD about February, it was working OK. After an update (to git HEAD again) it just doesn't finish starting up and doesn't even respond to the local open/close buttons.

It repeats this, over and over:
```
[09:46:47][I][wifi:319]: Connecting to 'Lumleys'
[09:46:47][I][esp-idf:000][wifi]: W (158792) wifi:
[09:46:47][I][esp-idf:000][wifi]: Password length matches WPA2 standards, authmode threshold changes from OPEN to WPA2
[09:46:47][I][esp-idf:000][wifi]: 
[09:46:50][I][esp-idf:000][wifi]: I (161666) wifi:
[09:46:50][I][esp-idf:000][wifi]: new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1, snd_ch_cfg:0x0
[09:46:50][I][esp-idf:000][wifi]: 
[09:46:50][I][esp-idf:000][wifi]: I (161668) wifi:
[09:46:50][I][esp-idf:000][wifi]: state: init -> auth (0xb0)
[09:46:50][I][esp-idf:000][wifi]: 
[09:46:50][W][component:417]: as5600.sensor took a long time for an operation (466 ms)
[09:46:50][W][component:418]: Components should block for at most 30 ms
[09:46:51][I][esp-idf:000][wifi]: I (162670) wifi:
[09:46:51][I][esp-idf:000][wifi]: state: auth -> init (0x200)
[09:46:51][I][esp-idf:000][wifi]: 
[09:46:51][I][esp-idf:000][wifi]: I (162672) wifi:
[09:46:51][I][esp-idf:000][wifi]: new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1, snd_ch_cfg:0x0
[09:46:51][I][esp-idf:000][wifi]: 
[09:46:51][W][wifi_esp32:710]: Disconnected ssid='Lumleys' bssid=18:62:2C:5D:94:6C reason='Auth Expired'
[09:46:51][W][wifi:666]: Connecting to network failed
[09:46:51][I][wifi:319]: Connecting to 'Lumleys'

```

### Which version of ESPHome has the issue?

commit 5b3d61b4a6832

### What type of installation are you using?

pip

### What platform are you using?

ESP32

### Component causing the issue

wifi

### YAML Config

```yaml
https://git.infradead.org/?p=users/dwmw2/esp32-pool.git;a=blob;f=landing-blind-3.yaml
```

### Anything in the logs that might be useful for us?

```txt

```

dwmw2, 2025-07-22:
Having the second WiFi config with explicit BSSID doesn't make a difference here. If I remove that and just let it scan, it behaves just the same way.

I've temporarily switched `enable_on_boot` to false and re-enable it with a button press, although I need to experiment with when I can automatically enable it at boot. I'm away again this week but this should be fairly easy to reproduce by just configuring a non-existent SSID so it *never* connects (unless it does require seeing beacons and at least *trying*?).

When I get home I'll do that with a spare unit on my desk and try to bisect. I'm sure this was working correctly in February when I got github.com/esphome/esphome/pull/8325 working, but then was failing by the time that was finally merged this month.

bdraco, 2025-07-30:
This issue got auto closed because https://github.com/esphome/esphome/pull/9963 was merged and it was linked.  However that PR doesn't appear to solve this problem

ssieb, 2025-10-29:
Is this still an issue?  If so, add serial logs.

PR #9963 (2025-07-29): Fix WiFi to prefer strongest AP when multiple APs have same SSID
# What does this implement/fix?

When multiple access points share the same SSID (common in mesh networks and with repeaters), ESPHome now connects to the AP with the strongest signal. Previously, the priority system could cause connections to weaker APs even when stronger ones were available.

The fix ensures that for APs with the same SSID, signal strength (RSSI) is the primary sorting criterion, overriding the priority system. The priority system still functions normally for different SSIDs.

This improves connectivity in envi [... cut]