esphome/esphome #9734: Missing states since 2025.7.2 release

sebastienfr, 2025-07-20:
### The problem

I have a water meter with the attached configuration. It was working with version 2025.6.x but since the 2025.7.x upgrade, only wifi, uptime and current_water_consumption is updated. 

All other states are not published on home assistant.

### Which version of ESPHome has the issue?

2025.7.2

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP8266

### Component causing the issue

api_apiserver publish state

### YAML Config

```yaml
substitutions:
  name: water
  friendly_name: "water meter"
  pulse_gpio: GPIO12

esphome:
  name: $name
  friendly_name: $friendly_name
  on_boot:
    priority: -10
    then:
      - script.execute: publish_states

esp8266:
  board: esp01_1m

# Enable logging
logger:

# Enable Home Assistant API
api:
  id: api_apiserver
  encryption:
    key: "xxx"
  reboot_timeout: 15min  
  services:
    - service: reset_main_counter
      then:
        - script.execute: reset_main_counter
    - service: reset_secondary_counter
      then:
        - script.execute: reset_secondary_counter

ota:
  - platform: esphome
    password: "xxx"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  # désactiver le mode économie d'énergie pour la stabilité
  power_save_mode: none
  # redémarrage automatique en cas de perte WiFi prolongée
  reboot_timeout: 15min
  # paramètres de reconnexion plus agressifs
[... 266 lines cut ...]
                  api->set_service("esphome.last_consumption_changes");
                  api->play();

                  id(event_quantity) = 0;
              }
            }
          }

          id(last_event_quantity) = id(event_quantity);

  # Track the current consumption
  - interval: 2sec
    then:
      - lambda: |-
          if (id(event_quantity) != id(current_event_quantity)) {
            id(current_water_consumption).publish_state(id(event_quantity));
          }
          id(current_event_quantity) = id(event_quantity);
```

### Anything in the logs that might be useful for us?

```txt

```

sebastienfr, 2025-07-20:
May it be linked to #9521 interval issue ?

Blerk666, 2025-07-20:
You can try to add `use_interrupt: false` to the binary_sensor. See also https://github.com/esphome/esphome/issues/9700

closed 2025-07-20 by jesserockz

sebastienfr, 2025-07-21:
Hello @ssieb ,

Yes I found that in some water meter tutorial : https://github.com/hugokernel/esphome-water-meter 
I didn't challenge it as it was working. What change would you recommend  ?

Regarding the setting, I rolled back to 2025.6.3, that's why its not working, thanks for the update.

Sébastien.