esphome/esphome #9772: Configuring SX127x failed

esnet146, 2025-07-21:
### The problem

[15:40:55][E][sx127x:493]: Configuring SX127x failed
[15:40:55][E][component:144]:   sx127x is marked FAILED: unspecified

### Which version of ESPHome has the issue?

 2025.7.2 2025.8.0-dev20250721

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

### Component causing the issue

sx127x

### YAML Config

```yaml
spi:
  clk_pin:
    number: GPIO05
    ignore_strapping_warning: true    
  mosi_pin: GPIO27
  miso_pin: GPIO19
sx127x:
  dio0_pin: GPIO26
  cs_pin: GPIO18
  rst_pin: GPIO14
  pa_pin: BOOST
  pa_power: 17
  bandwidth: 250_0kHz
  preamble_size: 6
  crc_enable: true
  frequency: 433920000
  modulation: LORA
  rx_start: true
  sync_value: 0x33
  spreading_factor: 12
  coding_rate: CR_4_8
  on_packet:
    then:
      - lambda: |-
          ESP_LOGD("lambda", "packet %s", format_hex(x).c_str());
          ESP_LOGD("lambda", "rssi %.2f", rssi);
          ESP_LOGD("lambda", "snr %.2f", snr);
```

### Anything in the logs that might be useful for us?

```txt
[15:57:39][C][spi:069]: SPI bus:
[15:57:39][C][spi:070]:   CLK Pin: GPIO5
[15:57:39][C][spi:071]:   SDI Pin: GPIO19
[15:57:39][C][spi:072]:   SDO Pin: GPIO27
[15:57:39][C][spi:077]:   Using HW SPI: SPI2_HOST
[15:57:39][C][sx127x:405]: SX127x:
[15:57:39][C][sx127x:406]:   CS Pin: GPIO18
[15:57:39][C][sx127x:407]:   RST Pin: GPIO14
[15:57:39][C][sx127x:408]:   DIO0 Pin: GPIO26
[15:57:39][C][sx127x:413]:   Auto Cal: TRUE
[15:57:39][C][sx127x:413]:   Frequency: 433920000 Hz
[15:57:39][C][sx127x:413]:   Bandwidth: 250000 Hz
[15:57:39][C][sx127x:413]:   PA Pin: BOOST
[15:57:39][C][sx127x:413]:   PA Power: 17 dBm
[15:57:39][C][sx127x:413]:   PA Ramp: 40 us
[15:57:39][C][sx127x:434]:   Modulation: LORA
[15:57:39][C][sx127x:434]:   Preamble Size: 6
[15:57:39][C][sx127x:434]:   Spreading Factor: 12
[15:57:39][C][sx127x:434]:   Coding Rate: 4/8
[15:57:39][C][sx127x:434]:   CRC Enable: TRUE
[15:57:39][C][sx127x:445]:   Sync Value: 0x33
[15:57:39][E][sx127x:493]: Configuring SX127x failed
[15:57:39][E][component:144]:   sx127x is marked FAILED: unspecified
```

### Additional information

If you change the configuration to:
```
external_components:
  - source: github://pr#7490
    components: [ sx127x ]

```
it works fine

@swoboda1337 Can you tell me what the problem is?

M66B, 2025-07-21:
```
ignore_strapping_warning: true
```

This might be the problem. @esnet146 which esp chip/device are you using?

esnet146, 2025-07-21:
> GPIO05 is pulled up and outputs a PWM signal at boot on an esp32 (classic) in any case.

heltec_wifi_lora_32_V2

esnet146, 2025-07-21:
[20:24:36][E][spi:083]: SPIDevice not initialised - did you call spi_setup()?
[20:24:36][E][spi:083]: SPIDevice not initialised - did you call spi_setup()?
[20:24:36][E][spi:083]: SPIDevice not initialised - did you call spi_setup()?
[20:24:36][E][spi:083]: SPIDevice not initialised - did you call spi_setup()?
[20:24:37][E][spi:083]: SPIDevice not initialised - did you call spi_setup()?
[20:24:37][E][sx127x:377]: Set mode failure
[20:24:37][E][component:194]: sx127x was marked as failed
[20:24:37][E][component:286]: sx127x set Error flag: unspecified
[20:24:37][E][spi:083]: SPIDevice not initialised - did you call spi_setup()?
[20:24:37][E][spi:083]: SPIDevice not initialised - did you call spi_setup()?
[20:24:37][E][spi:083]: SPIDevice not initialised - did you call spi_setup()?
[20:24:37][E][spi:083]: SPIDevice not initialised - did you call spi_setup()?
[20:24:37][E][spi:083]: SPIDevice not initialised - did you call spi_setup()?
[20:24:37][E][spi:083]: SPIDevice not initialised - did you call spi_setup()?
[20:24:37][E][spi:083]: SPIDevice not initialised - did you call spi_setup()?
[20:24:37][E][sx127x:377]: Set mode failure
[20:24:37][E][component:194]: sx127x was marked as failed
[20:24:37][E][spi:083]: SPIDevice not initialised - did you call spi_setup()?
[20:24:37][E][spi:083]: SPIDevice not initialised - did you call spi_setup()?
[20:24:41][E][sx127x:288]: Transmit packet failure
[20:24:41][E][spi:083]: SPIDevice not initialised - did you call spi_setup()?
[20:24:41][E][spi:083]: SPIDevice not initialised - did you call spi_setup()?
[20:24:41][E][spi:083]: SPIDevice not initialised - did you call spi_setup()?
[20:24:41][E][spi:083]: SPIDevice not initialised - did you call spi_setup()?
[20:24:41][E][spi:083]: SPIDevice not initialised - did you call spi_setup()?
[20:24:41][E][spi:083]: SPIDevice not initialised - did you call spi_setup()?
[20:24:41][E][sx127x:377]: Set mode failure
[20:24:41][E][component:194]: sx127x was marked as failed
[20:24:41][E][spi:083]: SPIDevice not initialised - did you call spi_setup()?
[20:24:41][E][spi:083]: SPIDevice not initialised - did you call spi_setup()?

swoboda1337, 2025-07-21:
I see. What does your yaml look like?

swoboda1337, 2025-07-21:
The SPI device is initialized in `SX127x::setup` though, `this->spi_setup()`. Not sure how a transmit is being called before setup is done?

swoboda1337, 2025-07-21:
Are you calling send packet in on_boot before sx127x setup is called?

esnet146, 2025-07-21:
It's not critical, thanks for solving the problem

closed 2025-07-21 by esnet146

esnet146, 2025-07-21:
> ```
> sx127x:
>   setup_priority: 800
> ```
I checked, it works