esphome/esphome #9773: AM2320 i2c on esp32-cam board (and NEW ESPHOME 2025.7.X)

corrado-c, 2025-07-21:
### The problem

I'm using some AM2320 I2C sensors on the regular ESP32-CAM board. I've always had some small problems, but it worked up to version 2025.7. Now I can't read the sensor anymore. I wanted to ask if there are any problems or other ways to use I2C sensors on this board. Using it only as a CAM is a waste. It has several pins available that can be used for other things.
And in general the video flow is really slow!!

### Which version of ESPHome has the issue?

2025.7.x

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

### Component causing the issue

i2c, am2320

### YAML Config

```yaml
---------OLD YAML WORKING FILE WITH PREVIOUS ESPHOME 2025.6---------
esphome:
  name: "esp32-cam-th"
  friendly_name: ESP32camTH

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:
  level: DEBUG

# Enable Home Assistant API
api:
  encryption:
    key: "xxxxxxxxxxxxxxxxxxxxxxxxxxxx"
ota:
  - platform: esphome
    password: "xxxxxxxxxxxxxx"

wifi:
  ssid: xxxx
  password: xxxx

captive_portal:

esp32_camera: #pin trovati su internet
  external_clock:
    pin: GPIO0
    frequency: 10MHz
  i2c_pins:
    sda: GPIO26
    scl: GPIO27
  data_pins: 
    - GPIO5
    - GPIO18
    - GPIO19
    - GPIO21
    - GPIO36
    - GPIO39
    - GPIO34
    - GPIO35
  vsync_pin: GPIO25
  href_pin: GPIO23
[... 313 lines cut ...]
[15:33:46][E][component:141]:   am2320.sensor is marked FAILED: unspecified
[15:33:46][C][psram:016]: PSRAM:
[15:33:46][C][psram:019]:   Available: YES
[15:33:46][C][psram:021]:   Size: 8192 KB
[15:33:46][C][captive_portal:099]: Captive Portal:
[15:33:46][C][esphome.ota:073]: Over-The-Air updates:
[15:33:46][C][esphome.ota:073]:   Address: esp32-cam-th.local:3232
[15:33:46][C][esphome.ota:073]:   Version: 2
[15:33:46][C][esphome.ota:080]:   Password configured
[15:33:46][C][safe_mode:018]: Safe Mode:
[15:33:46][C][safe_mode:019]:   Boot considered successful after 60 seconds
[15:33:46][C][safe_mode:019]:   Invoke after 10 boot attempts
[15:33:46][C][safe_mode:019]:   Remain for 300 seconds
[15:33:46][C][web_server.ota:224]: Web Server OTA
[15:33:46][C][api:207]: API Server:
[15:33:46][C][api:207]:   Address: esp32-cam-th.local:6053
[15:33:46][C][api:212]:   Using noise encryption: YES
[15:33:46][C][mdns:122]: mDNS:
[15:33:46][C][mdns:122]:   Hostname: esp32-cam-th
[15:33:50][D][esp32_camera:205]: Got Image: len=61670
```

### Additional information

I did several tests changing the i2c address, but it never found it, it worked before

swoboda1337, 2025-07-21:
You had two different sets of I2C pins in your old config:
```
 i2c_pins:
    sda: GPIO26
    scl: GPIO27

i2c:
  sda: GPIO14
  scl: GPIO15  #trovare altro pin
```
Now you just have:
```
# Unico bus I2C su GPIO26/27
i2c:
  - id: bus_all
    sda: GPIO26
    scl: GPIO27
    scan: true
    frequency: 100000
```

Any reason you didnt do this:
```
i2c:
  - id: bus_a
    sda: GPIO26
    scl: GPIO27
  - id: bus_b
    sda: GPIO14
    scl: GPIO15
```

corrado-c, 2025-07-22:
> External pullups are recommended as the resistance values are typically too high with the internal ones.

Yes, they're recommended, but I'm lazy LOL.
I was hoping it would work like it has so far. Maybe it was just on the edge.

corrado-c, 2025-07-23:
ok, I tried several things, even all the resistors from 10k up to 3k, nothing, then I made a light firmware with esp-idf framework and nothing, lastly I tried arduino framework and now it works... so I think it is now clear it is the framework, I attach the working file.
esphome:
  name: "esp32-cam-th"
  friendly_name: ESP32camTH

esp32:
  board: esp32dev
  framework:
    type: arduino
 
logger:
  level: DEBUG
api:
  encryption:
    key: "3xxxxxxxxxxxxxxxxxxx"

ota:
  platform: esphome
  password: "xxxxxxxxxxxxxxxxxxxxxxx"

wifi:
  ssid: "xxxxxxxxxxxxxxx"
  password: "xxxxxxxxxxxxx"
  ap: {}

captive_portal:

i2c:
  - id: bus_am2320
    sda: GPIO14
    scl: GPIO13
    frequency: 100kHz
    scan: true
    #sda_pullup_enabled: true
    #scl_pullup_enabled: true

sensor:
  - platform: am2320
    i2c_id: bus_am2320
    address: 0x5C
    temperature:
      name: "AM2320 Temperature"
      id: am_temp_sensor
      accuracy_decimals: 1
    humidity:
      name: "AM2320 Humidity"
      id: am_hum_sensor
      accuracy_decimals: 1    
    update_interval: 120s

swoboda1337, 2025-07-23:
Interesting, can you try with 5.4.2:
```
  framework:
    type: esp-idf
    version: 5.4.2
    platform_version: 54.03.21
```

swoboda1337, 2025-07-24:
Must have been some fix in the i2c driver, we should be switching to 5.4.2 next release.

rinyheuvelman, 2025-07-25:
I get the same warning with my camera setup and haven’t added anything else to the i2c but whatever was on the original ESP32-cam board.
“WARNING The `i2c_pins:` config option is deprecated. Use `i2c_id:` with a dedicated `i2c:` definition instead”
As far as I can tell, it doesn’t result in breaking anything so far.

So…
A: why did you change your platform from Arduino to esp-idf? 
B: What does this warning mean?
C: How to eliminate this warning within the Arduino platform? 
D: why does the  esp-idf  framework eliminate the warning? 

Thanks all!

rinyheuvelman, 2025-07-26:
The warning “ _The `i2c_pins:` config option is incompatible with an dedicated `i2c:` block, use `i2c_id` instead._” was logged for the first time after updating from 2025.7.2 to 2025.7.3. Although I use several i2c devices on different setups and boards, all of them are setup with “i2c_pins and none of them has “i2c_id”.
I eliminated that warning by implementing some code from one of the codes mentioned above.
I’ll post it when I have access to my laptop or computer which is much easier to use than my phone…

Thanks for sharing your codes

swoboda1337, 2025-08-20:
Fixed in 5.4.2