esphome/esphome #9786: Yaml coded automations now NEED WiFi - (Workaround included)

3ative, 2025-07-22:
### The problem

Yaml coded automations no longer run unless WiFi has connected first. 
I.E.: ``on_press:``, etc

### Which version of ESPHome has the issue?

2025.7.0

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

### Component causing the issue

https://esphome.io/automations/actions#do-automations-work-without-a-network-connection

### YAML Config

```yaml
binary_sensor:
  - platform: gpio
    id: button1
    pin:
      number: 5
      inverted: True
      mode:
        input: True
        pullup: True
    filters:
      - delayed_on_off: 50ms
    on_press:
      then:
        - lambda: |-
            if (id(fanspeed).state == 1 && id(master).state) {
              id(master).turn_off();
            } else {
              id(fanspeed).publish_state(1);
              id(master).turn_on();
            }
        - switch.turn_off: randomSwitch
```

### Anything in the logs that might be useful for us?

```txt
N/A
```

### Additional information

A Workaround is to delay WiFi:
```yaml
esphome:
  name: "xxxxx"
  #### Wait 10 Seconds, then enable WiFi #####
  on_boot:
    then:
      - delay: 10s
      - wifi.enable:
  ########################

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  #### Disable WiFi on boot to allow local automations to run without waiting for WiFi to connect ####
  enable_on_boot: false
```

ssieb, 2025-07-22:
You didn't include any logs.  There shouldn't be any problem like this.

jesserockz, 2025-07-22:
What actions are you trying to perform in on_press?
Please get the logs via usb/serial

3ative, 2025-07-22:
Nice one fellas....

Do you still need log(s) ???
I built this to test - Still waiting for the Yaml to compile LOL

[image]

3ative, 2025-07-22:
As I suspected....
Nothing in the logs apart from it trying to connect to an AP that is turned off.
Automation doesn't work until either I turn the AP back on or ``enable_on_boot: false`` is added.

Test YAML:
``` yaml
esphome:
  name: "000000"
  friendly_name: 000000

esp32:
  board: esp32-c3-devkitm-1
  framework:
    type: esp-idf

logger:
api:
ota:
  - platform: esphome

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

binary_sensor:
  - platform: gpio
    name: Button
    pin:
      number: 3
      inverted: True
      mode:
        input: True
        pullup: True

    on_press:
      then:
        - switch.toggle: LED1

switch:
  - platform: gpio
    id: LED1
    pin: 4
    name: LED
```

@bdraco 
I see you're having issues with the issues/PRs - Anything I can do here to help/test/etc ??

3ative, 2025-07-22:
> You can try the PR.

Would love too.... Want something Soldered - I'm your man. "_..try the PR._" I have no idea how to do that. :(

3ative, 2025-07-22:
@bdraco & @jesserockz  Sorry guys. V2025.7.3 didn't fix it - Problem remains. 

``on_press:`` and even ``on_state:`` will not work until WiFi is connected.

3ative, 2025-07-22:
Again, automations run if WiFi is delayed:

```yaml
esphome:
  name: "000000"
  friendly_name: 000000
  ### Wait, then Enable Wifi ###
  on_boot:
    then:
      - delay: 3s
      - wifi.enable:

esp32:
  board: esp32-c3-devkitm-1
  framework:
    type: esp-idf

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  ### Turn Off WiFi ###
  enable_on_boot: false
```

bdraco, 2025-07-22:
We are going to need serial logs to be able to determine whats going on here.

bdraco, 2025-07-23:
Also it would be helpful to know if its only gpio binary_sensor thats affected, and if turning off `use_interrupt` solves it https://esphome.io/components/binary_sensor/gpio.html#configuration-variables

3ative, 2025-07-23:
Logs with AP turned off. 
FYI: After I reset the ESP32, I started pushing the physical button... LED Did not Toggle.

[01:03:39][0;32m[I][logger:169]: Log initialized
[01:03:39][C][safe_mode:082]: There have been 1 suspected unsuccessful boot attempts
[01:03:39][D][esp32.preferences:142]: Writing 1 items: 0 cached, 1 written, 0 failed
[01:03:39][I][app:049]: Running through setup()
[01:03:39][C][switch.gpio:011]: Running setup for 'LED'
[01:03:39][D][switch:016]: 'LED' Turning OFF.
[01:03:39][D][switch:055]: 'LED': Sending state OFF
[01:03:39][D][esp-idf:000]: I (301) gpio: GPIO[4]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
[01:03:39][D][es[D][esp-idf:000][wifi]: I (3780) wifi:
[01:03:39][D][esp-idf:000][wifi]: new:<6,1>, old:<1,0>, ap:<255,255>, sta:<6,1>, prof:1, snd_ch_cfg:0x0
[01:03:39][D][esp-idf:000][wifi]: 
[01:03:39][D][esp-idf:000][wifi]: I (3782) wifi:
[01:03:39][D][esp-idf:000][wifi]: state: init -> auth (0xb0)
[01:03:39][D][esp-idf:000][wifi]: 
[01:03:39][D][esp-idf:000][wifi]: I (3786) wifi:
[01:03:39][D][esp-idf:000][wifi]: state: auth -> init (0x2500)
[01:03:39][D][esp-idf:000][wifi]: 
[01:03:39][D][esp-idf:000][wifi]: I (3787) wifi:
[01:03:39][D][esp-idf:000][wifi]: new:<6,0>, old:<6,1>, ap:<255,255>, sta:<6,1>, prof:1, snd_ch_cfg:0x0
[01:03:39][D][esp-idf:000][wifi]: 
[01:03:39][W][wifi_esp32:710]: Disconnected ssid='WizzAir' bssid=[redacted] reason='Authentication Failed'
[01:03:39][W][wifi:666]: Connecting to network failed
[01:03:39][D][wifi:710]: Retrying with hidden networks
[01:03:39][I][wifi:319]: Connecting to 'WizzAir'
[01:03:39][D][esp-idf:000][wifi]: W (3802) wifi:
[01:03:39][D][esp-idf:000][wifi]: Password length matches WPA2 standards, authmode threshold changes from OPEN to WPA2
[01:03:39][D][esp-idf:000][wifi]: 
[01:03:39][D][esp-idf:000][wifi]: I (3836) wifi:
[01:03:39][D][esp-idf:000][wifi]: new:<6,1>, old:<6,0>, ap:<255,255>, sta:<6,1>, prof:1, snd_ch_cfg:0x0
[01:03:39][D][esp-idf:000][wifi]: 
[01:03:39][D][esp-idf:000][wifi]: I (3838) wifi:
[01:03:39][D][esp-idf:000][wifi]: state: init -> auth (0xb0)
[01:03:39][D][esp-idf:000][wifi]: 
[01:03:39][D][esp-idf:000][wifi]: I (3842) wifi:
[01:03:39][D][esp-idf:000][wifi]: state: auth -> init (0x2500)
[01:03:39][D][esp-idf:000][wifi]: 
[01:03:39][D][esp-idf:000][wifi]: I (3844) wifi:
[01:03:39][D][esp-idf:000][wifi]: new:<6,0>, old:<6,1>, ap:<255,255>, sta:<6,1>, prof:1, snd_ch_cfg:0x0
[01:03:39][D][esp-idf:000][wifi]: 
[01:03:39][W][wifi_esp32:710]: Disconnected ssid='WizzAir' bssid=[redacted] reason='Authentication Failed'
[01:03:39][W][wifi:666]: Connecting to network failed
[01:03:39][D][wifi:710]: Retrying with hidden networks
[01:03:43][I][wifi:319]: Connecting to 'WizzAir'
[01:03:43][D][esp-idf:000][wifi]: W (8860) wifi:
[01:03:43][D][esp-idf:000][wifi]: Password length matches WPA2 standards, authmode threshold changes from OPEN to WPA2
[01:03:43][D][esp-idf:000][wifi]: 
[01:03:46][D][esp-idf:000][wifi]: I (11519) wifi:
[01:03:46][D][esp-idf:000][wifi]: new:<6,1>, old:<6,0>, ap:<255,255>, sta:<6,1>, prof:1, snd_ch_cfg:0x0
[01:03:46][D][esp-idf:000][wifi]: 
[01:03:46][D][esp-idf:000][wifi]: I (11521) wifi:
[01:03:46][D][esp-idf:000][wifi]: state: init -> auth (0xb0)
[01:03:46][D][esp-idf:000][wifi]: 
[01:03:46][D][esp-idf:000][wifi]: I (11529) wifi:
[01:03:46][D][esp-idf:000][wifi]: state: auth -> init (0x2500)
[01:03:46][D][esp-idf:000][wifi]: 
[01:03:46][D][esp-idf:000][wifi]: I (11531) wifi:
[01:03:46][D][esp-idf:000][wifi]: new:<6,0>, old:<6,1>, ap:<255,255>, sta:<6,1>, prof:1, snd_ch_cfg:0x0
[01:03:46][D][esp-idf:000][wifi]: 
[01:03:46][W][wifi_esp32:710]: Disconnected ssid='WizzAir' bssid=[redacted] reason='Authentication Failed'
[01:03:46][W][wifi:666]: Connecting to network failed
[01:03:46][D][wifi:710]: Retrying with hidden networks
[01:03:46][I][wifi:319]: Connecting to 'WizzAir'
[01:03:46][D][esp-idf:000][wifi]: W (11545) wifi:
[01:03:46][D][esp-idf:000][wifi]: Password length matches WPA2 standards, authmode threshold changes from OPEN to WPA2
[01:03:46][D][esp-idf:000][wifi]: 
[01:03:46][D][esp-idf:000][wifi]: I (11867) wifi:
[01:03:46][D][esp-idf:000][wifi]: new:<6,1>, old:<6,0>, ap:<255,255>, sta:<6,1>, prof:1, snd_ch_cfg:0x0
[... 120 lines cut ...]
[01:04:10][D][esp-idf:000][wifi]: I (35638) wifi:
[01:04:10][D][esp-idf:000][wifi]: new:<6,0>, old:<6,1>, ap:<255,255>, sta:<6,1>, prof:1, snd_ch_cfg:0x0
[01:04:10][D][esp-idf:000][wifi]: 
[01:04:10][W][wifi_esp32:710]: Disconnected ssid='WizzAir' bssid=[redacted] reason='Authentication Failed'
[01:04:10][W][wifi:666]: Connecting to network failed
[01:04:10][D][wifi:710]: Retrying with hidden networks
[01:04:10][I][wifi:319]: Connecting to 'WizzAir'
[01:04:10][D][esp-idf:000][wifi]: W (35652) wifi:
[01:04:10][D][esp-idf:000][wifi]: Password length matches WPA2 standards, authmode threshold changes from OPEN to WPA2
[01:04:10][D][esp-idf:000][wifi]: 
[01:04:10][D][esp-idf:000][wifi]: I (35934) wifi:
[01:04:10][D][esp-idf:000][wifi]: new:<6,1>, old:<6,0>, ap:<255,255>, sta:<6,1>, prof:1, snd_ch_cfg:0x0
[01:04:10][D][esp-idf:000][wifi]: 
[01:04:10][D][esp-idf:000][wifi]: I (35936) wifi:
[01:04:10][D][esp-idf:000][wifi]: state: init -> auth (0xb0)
[01:04:10][D][esp-idf:000][wifi]: 
[01:04:10][D][esp-idf:000][wifi]: I (35941) wifi:
[01:04:10][D][esp-idf:000][wifi]: state: auth -> init (0x2500)
[01:04:10][D][esp-idf:000][wifi]: 
[01:04:10][D][esp-idf:000][wifi]: I (35942) wifi:
[01:04:10][D][esp-idf:000][wifi]: new:<6,0>, old:<6,1>, ap:<255,255>, sta:<6,1>, prof:1, snd_ch_cfg:0x0
[01:04:10][D][esp-idf:000][wifi]: 
[01:04:10][W][wifi_esp32:710]: Disconnected ssid='WizzAir' bssid=[redacted] reason='Authentication Failed'
[01:04:10][W][wifi:666]: Connecting to network failed
[01:04:10][D][wifi:710]: Retrying with hidden networks

3ative, 2025-07-23:
This time I turned AP back on and used "Reset Device". Again, I started pushing the Physical Button immediately:

[01:07:01]ESP-ROM:esp32c3-api1-20210207
[01:07:01]Build:Feb  7 2021
[01:07:01]rst:0x15 (USB_UART_CHIP_RESET),boot:0xd (SPI_FAST_FLASH_BOOT)
[01:07:01]Saved PC:0x40384786
[01:07:01]SPIWP:0xee
[01:07:01]mode:DIO, clock div:1
[01:07:01]load:0x3fcd5820,len:0x1820
[01:07:01]load:0x403cc710,len:0xb38
[01:07:01]load:0x403ce710,len:0x2fb4
[01:07:01]entry 0x403cc710
[01:07:01]I (24) boot: ESP-IDF 5.3.2 2nd stage bootloader
[01:07:01]I (24) boot: compile time Jul 22 2025 01:58:54
[01:07:01]I (25) boot: chip revision: v0.4
[01:07:01]I (25) boot: efuse block revision: v1.3
[01:07:01]I (26) boot.esp32c3: SPI Speed      : 80MHz
[01:07:01]I (26) boot.esp32c3: SPI Mode       : DIO
[01:07:01]I (27) boot.esp32c3: SPI Flash Size : 4MB
[01:07:01]I (27) boot: Enabling RNG early entropy source...
[01:07:01]I (28) boot: Partition Table:
[01:07:01]I (28) boot: ## Label            Usage          Type ST Offset   Length
[01:07:01]I (29) boot:  0 otadata          OTA data         01 00 00009000 00002000
[01:07:01]I (30) boot:  1 phy_init         RF data          01 01 0000b000 00001000
[01:07:01]I (31) boot:  2 app0             OTA app          00 10 00010000 001c0000
[01:07:01]I (32) boot:  3 app1             OTA app          00 11 001d0000 001c0000
[01:07:01]I (33) boot:  4 nvs              WiFi data        01 02 00390000 0006d000
[01:07:01]I (34) boot: End of partition table
[01:07:01]I (35) esp_image: segment 0: paddr=001d0020 vaddr=3c0a0020 size=1c014h (114708) map
[01:07:01]I (54) esp_image: segment 1: paddr=001ec03c vaddr=3fc92a00 size=02cf0h ( 11504) load
[01:07:01]I (57) esp_image: segment 2: paddr=001eed34 vaddr=40380000 size=012e4h (  4836) load
[01:07:01]I (58) esp_image: segment 3: paddr=001f0020 vaddr=42000020 size=91f7ch (597884) map
[01:07:02]I (154) esp_image: segment 4: paddr=00281fa4 vaddr=403812e4 size=115e4h ( 71140) load
[01:07:02]I (173) boot: Loaded app from partition at offset 0x1d0000
[01:07:02]I (174) boot: Disabling RNG early entropy source...
[01:07:02]I (175) cpu_start: Unicore app
[01:07:02]I (183) cpu_start: Pro cpu start user code
[01:07:02]I (183) cpu_start: cpu freq: 160000000 Hz
[01:07:02]I (183) app_init: Application information:
[01:07:02]I (183) app_init: Project name:     000000
[01:07:02]I (183) app_init: App version:      2025.7.3
[01:07:02]I (184) app_init: Compile time:     Jul 23 2025 00:16:49
[01:07:02]I (184) app_init: ELF file SHA256:  6b051a98b...
[01:07:02]I (184) app_init: ESP-IDF:          5.3.2
[01:07:02]I (185) efuse_init: Min chip rev:     v0.3
[01:07:02]I (185) efuse_init: Max chip rev:     v1.99 
[01:07:02]I (186) efuse_init: Chip rev:         v0.4
[01:07:02]I (186) heap_init: Initializing. RAM available for dynamic allocation:
[01:07:02]I (187) heap_init: At 3FC99E40 len 000261C0 (152 KiB): RAM
[01:07:02]I (188) heap_init: At 3FCC0000 len 0001C710 (113 KiB): Retention RAM
[01:07:02]I (189) heap_init: At 3FCDC710 len 00002950 (10 KiB): Retention RAM
[01:07:02]I (190) heap_init: At 50000200 len 00001DE8 (7 KiB): RTCRAM
[01:07:02]I (191) spi_flash: detected chip: generic
[01:07:02]I (191) spi_flash: flash io: dio
[01:07:02]I (192) sleep: Configure to isolate all GPIO pins in sleep state
[01:07:02]I (193) sleep: Enable automatic switching of GPIO sleep configuration
[01:07:02]I (194) main_task: Started on CPU0
[01:07:02]I (194) main_task: Calling app_main()
[01:07:02]I (258) main_task: Returned from app_main()
[01:07:02][I][logger:169]: Log initialized
[01:07:02][C][safe_mode:082]: There have been 2 suspected unsuccessful boot attempts
[01:07:02][D][esp32.preferences:142]: Writing 1 items: 0 cached, 1 written, 0 failed
[01:07:02][I][app:049]: Running through setup()
[01:07:02][C][switch.gpio:011]: Running setup for 'LED'
[01:07:02][D][switch:016]: 'LED' Turning OFF.
[01:07:02][D][switch:055]: 'LED': Sending state OFF
[01:07:02][D][esp-idf:000]: I (265) gpio: GPIO[4]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
[01:07:02][D][esp32:117]: rtc: 0
[01:07:02][D][switch:016]: 'LED' Turning OFF.
[01:07:02][D][esp-idf:000]: I (266) gpio: GPIO[3]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
[... 183 lines cut ...]
[01:07:07][D][esp-idf:000][wifi]: 
[01:07:07][D][switch:020]: 'LED' Toggling ON.
[01:07:07][D][switch:055]: 'LED': Sending state ON
[01:07:07][D][binary_sensor:026]: 'Button': New state is ON
[01:07:07][D][binary_sensor:026]: 'Button': New state is OFF
[01:07:07][D][switch:020]: 'LED' Toggling OFF.
[01:07:07][D][switch:055]: 'LED': Sending state OFF
[01:07:07][D][binary_sensor:026]: 'Button': New state is ON
[01:07:07][D][binary_sensor:026]: 'Button': New state is OFF
[01:07:07][D][switch:020]: 'LED' Toggling ON.
[01:07:07][D][switch:055]: 'LED': Sending state ON
[01:07:07][D][binary_sensor:026]: 'Button': New state is ON
[01:07:08][D][binary_sensor:026]: 'Button': New state is OFF
[01:07:08][D][switch:020]: 'LED' Toggling OFF.
[01:07:08][D][switch:055]: 'LED': Sending state OFF
[01:07:08][D][binary_sensor:026]: 'Button': New state is ON
[01:07:08][D][binary_sensor:026]: 'Button': New state is OFF
[01:07:08][D][switch:020]: 'LED' Toggling ON.
[01:07:08][D][switch:055]: 'LED': Sending state ON
[01:07:08][D][binary_sensor:026]: 'Button': New state is ON
[01:07:08][D][binary_sensor:026]: 'Button': New state is OFF
[01:07:08][D][switch:020]: 'LED' Toggling OFF.
[01:07:08][D][switch:055]: 'LED': Sending state OFF
[01:07:08][D][binary_sensor:026]: 'Button': New state is ON
[01:07:08][D][binary_sensor:026]: 'Button': New state is OFF

3ative, 2025-07-23:
2nd Result: LED started toggling after the ESP32 connected to Wifi

3ative, 2025-07-23:
> > Hmmm... with No Wifi connected the Log don't show the Button being pressed - Maybe it's NOT the automation not running, rather the GPIO not getting read ????
> 
> Yes that's what the fix in 2025.7.3 was expected to resolve. The loop wasn't getting reenabled until setup finished so it would get the result until the main loop started

Ah... Hmmm... As I said this sort of thing used to work... Albeit 5 Years ago (When I first used yaml automations)

3ative, 2025-07-23:
I'm currently busting on this BEAST of a set up LOL

[image]

... and I know this is a "Core" issue... But if I can help with any more testing/logs/info/etc Lemme know

bdraco, 2025-07-23:
> ... and I know this is a "Core" issue... But if I can help with any more testing/logs/info/etc Lemme know

You might have missed it with all thats going on in this issue but please check https://github.com/esphome/esphome/issues/9786#issuecomment-3105192243

3ative, 2025-07-23:
Are, I did miss that... Do you want me to try again with and ``interrupt_type``` ??? 
- If so, if you give me some yaml to add I'll be happy to test

3ative, 2025-07-23:
I tried this, still didn't work until WiFi connected:
```yaml
binary_sensor:
  - platform: gpio
    name: Button
    pin:
      number: 3
      inverted: True
      mode:
        input: True
        pullup: True
    interrupt_type: ANY
```
Again, started pushing the Button right after "Reset"

Logs:

[01:39:24]ESP-ROM:esp32c3-api1-20210207
[01:39:24]Build:Feb  7 2021
[01:39:24]rst:0x15 (USB_UART_CHIP_RESET),boot:0xd (SPI_FAST_FLASH_BOOT)
[01:39:24]Saved PC:0x40384786
[01:39:24]SPIWP:0xee
[01:39:24]mode:DIO, clock div:1
[01:39:24]load:0x3fcd5820,len:0x1820
[01:39:24]load:0x403cc710,len:0xb38
[01:39:24]load:0x403ce710,len:0x2fb4
[01:39:24]entry 0x403cc710
[01:39:24]I (24) boot: ESP-IDF 5.3.2 2nd stage bootloader
[01:39:24]I (24) boot: compile time Jul 22 2025 01:58:54
[01:39:24]I (25) boot: chip revision: v0.4
[01:39:24]I (25) boot: efuse block revision: v1.3
[01:39:24]I (26) boot.esp32c3: SPI Speed      : 80MHz
[01:39:24]I (26) boot.esp32c3: SPI Mode       : DIO
[01:39:24]I (27) boot.esp32c3: SPI Flash Size : 4MB
[01:39:24]I (27) boot: Enabling RNG early entropy source...
[01:39:24]I (28) boot: Partition Table:
[01:39:24]I (28) boot: ## Label            Usage          Type ST Offset   Length
[01:39:24]I (29) boot:  0 otadata          OTA data         01 00 00009000 00002000
[01:39:24]I (30) boot:  1 phy_init         RF data          01 01 0000b000 00001000
[01:39:24]I (31) boot:  2 app0             OTA app          00 10 00010000 001c0000
[01:39:24]I (32) boot:  3 app1             OTA app          00 11 001d0000 001c0000
[01:39:24]I (33) boot:  4 nvs              WiFi data        01 02 00390000 0006d000
[01:39:24]I (34) boot: End of partition table
[01:39:24]I (34) esp_image: segment 0: paddr=00010020 vaddr=3c0a0020 size=1c014h (114708) map
[01:39:24]I (54) esp_image: segment 1: paddr=0002c03c vaddr=3fc92a00 size=02cf0h ( 11504) load
[01:39:24]I (56) esp_image: segment 2: paddr=0002ed34 vaddr=40380000 size=012e4h (  4836) load
[01:39:24]I (58) esp_image: segment 3: paddr=00030020 vaddr=42000020 size=91f7ch (597884) map
[01:39:24]I (153) esp_image: segment 4: paddr=000c1fa4 vaddr=403812e4 size=115e4h ( 71140) load
[01:39:24]I (173) boot: Loaded app from partition at offset 0x10000
[01:39:24]I (173) boot: Disabling RNG early entropy source...
[01:39:24]I (174) cpu_start: Unicore app
[01:39:24]I (182) cpu_start: Pro cpu start user code
[01:39:24]I (183) cpu_start: cpu freq: 160000000 Hz
[01:39:24]I (183) app_init: Application information:
[01:39:24]I (183) app_init: Project name:     000000
[01:39:24]I (184) app_init: App version:      2025.7.3
[01:39:24]I (184) app_init: Compile time:     Jul 23 2025 00:16:49
[01:39:24]I (185) app_init: ELF file SHA256:  60a5bf64e...
[01:39:24]I (185) app_init: ESP-IDF:          5.3.2
[01:39:24]I (186) efuse_init: Min chip rev:     v0.3
[01:39:24]I (186) efuse_init: Max chip rev:     v1.99 
[01:39:24]I (187) efuse_init: Chip rev:         v0.4
[01:39:24]I (187) heap_init: Initializing. RAM available for dynamic allocation:
[01:39:25]I (188) heap_init: At 3FC99E40 len 000261C0 (152 KiB): RAM
[01:39:25]I (189) heap_init: At 3FCC0000 len 0001C710 (113 KiB): Retention RAM
[01:39:25]I (190) heap_init: At 3FCDC710 len 00002950 (10 KiB): Retention RAM
[01:39:25]I (191) heap_init: At 50000200 len 00001DE8 (7 KiB): RTCRAM
[01:39:25]I (193) spi_flash: detected chip: generic
[01:39:25]I (193) spi_flash: flash io: dio
[01:39:25]I (194) sleep: Configure to isolate all GPIO pins in sleep state
[... 195 lines cut ...]
[01:39:30][D][binary_sensor:026]: 'Button': New state is OFF
[01:39:30][D][switch:020]: 'LED' Toggling OFF.
[01:39:30][D][switch:055]: 'LED': Sending state OFF
[01:39:30][D][binary_sensor:026]: 'Button': New state is ON
[01:39:30][D][binary_sensor:026]: 'Button': New state is OFF
[01:39:30][D][switch:020]: 'LED' Toggling ON.
[01:39:30][D][switch:055]: 'LED': Sending state ON
[01:39:30][D][binary_sensor:026]: 'Button': New state is ON
[01:39:30][D][binary_sensor:026]: 'Button': New state is OFF
[01:39:30][D][switch:020]: 'LED' Toggling OFF.
[01:39:30][D][switch:055]: 'LED': Sending state OFF
[01:39:30][D][binary_sensor:026]: 'Button': New state is ON
[01:39:30][D][binary_sensor:026]: 'Button': New state is OFF
[01:39:30][D][switch:020]: 'LED' Toggling ON.
[01:39:30][D][switch:055]: 'LED': Sending state ON
[01:39:30][D][binary_sensor:026]: 'Button': New state is ON
[01:39:30][D][binary_sensor:026]: 'Button': New state is OFF
[01:39:31][D][switch:020]: 'LED' Toggling OFF.
[01:39:31][D][switch:055]: 'LED': Sending state OFF
[01:39:31][D][binary_sensor:026]: 'Button': New state is ON
[01:39:31][D][binary_sensor:026]: 'Button': New state is OFF
[01:39:31][D][switch:020]: 'LED' Toggling ON.
[01:39:31][D][switch:055]: 'LED': Sending state ON
[01:39:31][D][binary_sensor:026]: 'Button': New state is ON
[01:39:31][D][binary_sensor:026]: 'Button': New state is OFF

bdraco, 2025-07-23:
Specifically I was looking for you to disable the `use_interrupt` option

3ative, 2025-07-23:
It's been a couple of minutes now... Log just show it trying to connect. No Button presses logged and no LED toggling.

3ative, 2025-07-23:
> Specifically I was looking for you to disable the `use_interrupt` option

Oppps... Didn't read properly LOL.... One sec....

3ative, 2025-07-23:
If you say so... _LMAO_ I wish I knew more - I kinda get it thou. 

Progress then! 

Hopefully you can work it out now. Good Luck. 
Thank you for your Service Dude. ❤️

bdraco, 2025-07-23:
looping_components_ is empty during the setup phase 🙈

bdraco, 2025-07-23:
But it works fine on my test branch

bdraco, 2025-07-23:
Let me try 2025.7.3 directly

jesserockz, 2025-07-23:
Maybe a bad cherry-pick?

bdraco, 2025-07-23:
Doesn't work on 2025.7.3

But it works fine on my test branch because I have `calculate_looping_components_` happening sooner on it. But thats not in dev or backported because I was still testing a different change.

So testing error on my part because I had unrelated changes in my test branch. The fix in https://github.com/esphome/esphome/pull/9787 is incomplete. Sorry!

bdraco, 2025-07-23:
I had move the calculate to get rid of the extra loop and reduce the function size but I didn't include that in the fix pr.  I forgot it was in that branch so it just worked

closed 2025-07-23 by bdraco