esphome/esphome #9827: Poor RISC-V code density compared to Xtensa?

Spamfast, 2025-07-23:
### The problem

I've got one Athom Smart Plug C3 that use `board: esp32-c3-devkitm-1` and some of their cheaper 'V2' ones that use `board: esp8285`. I generally flash identical configurations onto both except for the board choice and the platformio flags (gleaned from Athom's YAML on Github).

The code size is always around twice the size for the C3 as for the V2 plugs.

While I am a big fan of Tensilica's Xtensa instruction set architecture, I find it hard to believe it's twice as efficient as RISC-V! Are we missing some optimizer options for the RISC-V compiler or is it really that poor code-density-wise compared to Xtensa?

### Which version of ESPHome has the issue?

2025.7.3

### What type of installation are you using?

pip

### What platform are you using?

ESP8266

### Component causing the issue

na

### YAML Config

```yaml
# C3 plug:-
esphome:
  comment: athom smart plug c3
  platformio_options:
    board_build.mcu: esp32c3
    board_build.variant: esp32c3
    board_build.flash_mode: dio
esp32:
  board: esp32-c3-devkitm-1
  flash_size: 4MB
  variant: ESP32C3
  framework:
    type: arduino
    version: recommended

# 8285 plug:-
esphome:
  comment: athom smart plug v2
  platformio_options:
    board_upload.flash_size: 2MB
    board_upload.maximum_size: 2097152
    board_build.ldscript: eagle.flash.2m256.ld
esp8266:
  board: esp8285
  restore_from_flash: true
```

### Anything in the logs that might be useful for us?

```txt

```

bdraco, 2025-07-23:
The comparison between ESP32-C3 (RISC-V) and ESP8285/ESP8266 (Xtensa) code sizes is not meaningful because these are fundamentally different platforms with vastly different architectures and capabilities.

Spamfast, 2025-07-23:
Well, first of all, thanks very much for ESPHome and kudos over how it has been designed and is being maintained and enhanced. I find it an invaluable adjunct to Home Assistant.

But I can't help being disappointed by the response to my question. (it was a question, not a feature request by the way.)

Of course I know the CPUs are different but it's perfectly valid to compare hetrogenous platforms. I have been doing it for the past thirty-five years when choosing what processor/SoC & other hardware to use for a design and what bare-metal threading library (if any), RTOS framework or high level OS to use for the firmware. If I'm going to have to give one twice as much RAM or flash as another all else being equal I'll go for the lower BOM cost.

In this case, my flash usage is about 1200KiB on C3 vs 600KiB on 8285. Are you saying tha there's about half a megabyte of unused code in the C3 build for features not being used by my configuration?  Note that the C3 code is twice the size of the 8285 for the same YAML whether the C3 uses `type: arduino` or `type: esp-idf`. I imagine all three PlatformIO frameworks (C3 Arduino, C3 ESP-IDF, 8266) use FreeRTOS, lwIP etc. in much the same way.

Last time I checked the GNU linker discards text, data & rodata that isn't referenced. When I use ESP-IDF (or Zephyr or my own FreeRTOS-based platform) directly, I find that if I don't use a feature it doesn't get linked into the final binary. Is ESPHome using ESP-IDF on C3 in such a way that this is not the case? 

I might have assumed RISC-V architecture - with with I am unfamiliar - would be more efficient than the older (3-byte-op-code - I love it!) Xtensa one but perhaps, is it that the RISC-V GCC compiler/optimizer is just poor compared to the Xtensa one, given that RISC-V is a younger architecture? I don't know. That's why I asked.

I'd hoped someone might have a plausible explanation but if not, I'll investigate elsewhere.

No need to reply.

bdraco, 2025-07-23:
Comparing the ESP32S3 to the ESP32C3 it likely a better comparison.  The fixed 32-bit instruction format vs Xtensa's variable-length instructions is likely the cause for the larger binaries.  The tool chain for the C3 is less mature as well.  It might get a better over time but likely not much that can be done.

Spamfast, 2025-07-24:
Thanks for the detailed breakout. I've been doing some more (rather more basic) comparisons too.

I'm going to have to eat some humble pie.

An equivalent config to my original 8285 & C3 ones but with an ESP32 also consumes far more than the 8285 (although not quite as much as the C3) using ESP-IDF. An S3 build is much the same as the ESP32. So - as you tried to tell me - it's down to the platform code and not the compiler/CPU.

I'm still a little surprised about how much more has to remain in compared to 8285 when not using stuff like BLE and the rest. I know the ESP32 has all the MMU stuff for making external flash/PSRAM access transparent to application code and the, surprising simple, changes for dual-core FreeRTOS.

Anyway, thanks for your patience and the further info.

I shall have to bear the overheads in mind when comparing boards for ESPHome in future - upshot, if it's an ESP32{,C3,S3} it needs twice the flash as a 82xx. Luckily they don't seem to come with less than 4MiB anyway.

(I've usually been working with ARM cores for the past N years and I'm close to retirement age so probably won't be doing much with RISC-V or whatever else comes along 'in anger'. Also, while I see the need for frameworks like Zephyr, secure boot & the like (and don't get me started on Linux kernel stuff- what a mess!)),  it takes the fun out of it not to have a blank slate. In the words of Neil Young, "I think I'm gettin' old.")

But I'm rambling. Thanks again. Case closed.