esphome/esphome #9835: Cannot set device_id to main device

danez, 2025-07-23:
### The problem

I have some shared packages (1) which i would love to use the new sub-device feature. But i would like to make it optional.

I experimented, but wasn't able to make it happen. I tried something like this:
```yaml
defaults:
  device_id: ""

sensor:
  - device_id: "${device_id}"
```

But neither and empty string nor null is allowed for `device_id` on sensors.

What I am trying to achieve is to set the sensor either to a subdevice (which works) or to the main device.

With empty string I get:

```
ID must not be empty.
  device_id: ''
```

With null I get:

```
Too many candidates found for 'device_id' type 'Device' Some are 'xxx', 'yyy'.
```

Would it be possible to explicitly set a sensor to the main device? So that in a package I can based on vars decide if set to main device or sub device?

(1) https://github.com/danez/esphome-zendure-bridge/blob/v1/packages/hub2000.yaml

### Which version of ESPHome has the issue?

2025.7.3

### What type of installation are you using?

pip

### What platform are you using?

ESP32

### YAML Config

```yaml

```

### Anything in the logs that might be useful for us?

```txt

```

bdraco, 2025-07-23:
untested, but 0 might work

danez, 2025-07-25:
> untested, but 0 might work

Unfortunately not

```
 First character in ID cannot be a digit.
  device_id: 0
```