esphome/esphome #9870: Compilation error in ping_esp32.h: missing ip_2_ip6 and inet6_ntoa when using IDF framework

luconedj, 2025-07-24:
### The problem

When compiling with `framework: idf` (esp-idf), `ping_esp32.h` references `ip_2_ip6()` and `inet6_ntoa()` which are unavailable in IDF builds, causing compilation to fail.

[image]

### Which version of ESPHome has the issue?

2025.7.2

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

### Component causing the issue

esphome/ping

### YAML Config

```yaml

```

### Anything in the logs that might be useful for us?

```txt
In file included from src/esphome/core/entity_base.h:7,
                 from src/esphome/components/sensor/sensor.h:4,
                 from src/esphome/components/ads1115/sensor/ads1115_sensor.h:6,
                 from src/esphome.h:4,
                 from src/main.cpp:3:
src/esphome/components/ping/ping_esp32.h: In member function 'void esphome::ping::PingSensorESP32::cmd_ping_on_ping_end(esp_ping_handle_t)':
src/esphome/components/ping/ping_esp32.h:106:63: error: 'ip_2_ip6' was not declared in this scope; did you mean 'ip_2_ip4'?
  106 |       ESP_LOGD(TAG, "--- %s ping statistics ---", inet6_ntoa(*ip_2_ip6(&target_addr)));
      |                                                               ^~~~~~~~
src/esphome/core/log.h:99:100: note: in definition of macro 'esph_log_d'
   99 |   ::esphome::esp_log_printf_(ESPHOME_LOG_LEVEL_DEBUG, tag, __LINE__, ESPHOME_LOG_FORMAT(format), ##__VA_ARGS__)
      |                                                                                                    ^~~~~~~~~~~
src/esphome/components/ping/ping_esp32.h:106:7: note: in expansion of macro 'ESP_LOGD'
  106 |       ESP_LOGD(TAG, "--- %s ping statistics ---", inet6_ntoa(*ip_2_ip6(&target_addr)));
      |       ^~~~~~~~
src/esphome/components/ping/ping_esp32.h:106:51: error: 'inet6_ntoa' was not declared in this scope; did you mean 'inet_ntoa'?
  106 |       ESP_LOGD(TAG, "--- %s ping statistics ---", inet6_ntoa(*ip_2_ip6(&target_addr)));
      |                                                   ^~~~~~~~~~
src/esphome/core/log.h:99:100: note: in definition of macro 'esph_log_d'
   99 |   ::esphome::esp_log_printf_(ESPHOME_LOG_LEVEL_DEBUG, tag, __LINE__, ESPHOME_LOG_FORMAT(format), ##__VA_ARGS__)
      |                                                                                                    ^~~~~~~~~~~
src/esphome/components/ping/ping_esp32.h:106:7: note: in expansion of macro 'ESP_LOGD'
  106 |       ESP_LOGD(TAG, "--- %s ping statistics ---", inet6_ntoa(*ip_2_ip6(&target_addr)));
      |       ^~~~~~~~
```

### Additional information

This is only an issue with the IDF framework; Arduino builds still work properly.

swoboda1337, 2025-07-24:
So ping is not an official component: https://github.com/esphome/esphome/tree/dev/esphome/components

You will have to ask the owner of that to fix it.

luconedj, 2025-07-25:
Hi @swoboda1337 , thanks for your feedback, sorry for the mistake.
Luca