esphome/esphome #9873: Nextion display

ZioMauri57, 2025-07-24:
### The problem

With version 2025.7.2 and earlier, it works correctly. 
Updated to version 2025.7.3 no longer works. 
id(nextion1).set_component_value("n1", 1); and 
id(nextion1).set_component_value("n0", id(serratura).state); 
do not update the value.

### Which version of ESPHome has the issue?

2025.7.3

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP8266

### Component causing the issue

nextion display

### YAML Config

```yaml
esphome:
  name: tastiera-ingresso
  friendly_name: Tastiera_ingresso

esp8266:
  board: d1_mini

# Enable logging
logger:
  baud_rate: 0

# Enable Home Assistant API
api:
  encryption:
    key: "OpdF4omIkMJkFiwEldbcrwYTr7xbCqTuWtJrNZ/EJPM="

ota:
  - platform: esphome
    password: "372c8e27271fc77df11a72a43a0f3284"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Tastiera-Ingresso"
    password: "NkWsuNV5zHmw"

# Enable web server
web_server:

captive_portal:
    
# Led stato segnale wifi
status_led:
  pin: GPIO2

uart:
  rx_pin: D1
  tx_pin: D2
  baud_rate: 115200

globals:
  - id: first_page
    type: bool
    restore_value: no  

binary_sensor:
  - platform: gpio
    name: "Pulsante_reset"
    pin:
      number: D5
      inverted: True
      mode: 
        input: True
        pullup: True
    on_press: 
      then:
        - lambda: |-
            id(nextion1).set_component_value("n1", 1);

  - platform: homeassistant #------------------------------------------
    id: serratura
    entity_id: switch.tastiera_ingresso_stato_tedee

  - platform: nextion
    page_id: 0
    component_id: 1
    id: attesa
    on_release:
      then:
        - text_sensor.template.publish: 
            id: comando_tedee
            state: "attesa_pin"
  
  - platform: nextion
    page_id: 0
    component_id: 3
    id: invio_stato
    on_release:
      then:
        - lambda: |-
            id(nextion1).set_component_value("n0", id(serratura).state);

  - platform: nextion
    page_id: 2
    component_id: 1
    id: blocca_tedee
    on_press:
      then:
        - text_sensor.template.publish: 
            id: comando_tedee
            state: "blocca"
        - delay:
            seconds: 2
        - text_sensor.template.publish:
            id: comando_tedee
            state: "attesa_pin"

  - platform: nextion
    page_id: 2
    component_id: 2
    id: apri_tedee
    on_press:
      then:
        - text_sensor.template.publish:
            id: comando_tedee
            state: "apri"
        - delay:
            seconds: 2
        - text_sensor.template.publish:
            id: comando_tedee
            state: "attesa_pin"
  
  - platform: nextion
    page_id: 3
    component_id: 1
    id: sblocca_tedee
    on_press:
      then:
        - text_sensor.template.publish:
            id: comando_tedee
            state: "sblocca"
        - delay:
            seconds: 2
        - text_sensor.template.publish:
            id: comando_tedee
            state: "attesa_pin"

  - platform: nextion
    page_id: 1
    component_id: 1
    id: pin_errato
    on_press:
      then:
        - text_sensor.template.publish:
            id: comando_tedee
            state: "pin_errato"
        - delay:
            seconds: 2
        - text_sensor.template.publish:
            id: comando_tedee
            state: "attesa_pin"

  - platform: nextion
    page_id: 4
    component_id: 1
    id: tastiera_bloccata
    on_press:
      then:
        - text_sensor.template.publish:
            id: comando_tedee
            state: "tastiera_bloccata"

switch:
  - platform: gpio
    pin: 12 
    name: "reset_tastiera"
    inverted: True
    id: reset

  - platform: gpio
    pin: 13
    name: "stato tedee"
    inverted: True
    id: led1

text_sensor:
  - platform: template
    id: comando_tedee
    name: "comando tedee"
display:
  - platform: nextion
    id: nextion1
```

### Anything in the logs that might be useful for us?

```txt

```

edwardtfn, 2025-07-25:
Is Nextion display establishing connection properly? Do you have any error message related to this in your logs?
Could you please share some relevant logs?

ZioMauri57, 2025-07-25:
hi.. 
I searched everywhere, I found no mistakes

edwardtfn, 2025-07-26:
Which nextion model and firmware version you have? You should be able to see this info in your logs.

ZioMauri57, 2025-07-26:
Hi, I solved it.
With all previous versions, after upgrade esphome works perfectly.
With version 2025.7.3 after update no longer works, you need to restart esp32 and everything works again.
Would it be possible to restart remotely? Or insert a button in the web interface? It would be convenient when esphome is difficult to reach for reset.
Thank you

edwardtfn, 2025-07-26:
https://esphome.io/components/button/restart.html

paddenstoeltje, 2025-10-06:
Actually I have the same problem. 

Since Esphome 2025.7.3 and higher I am unable to connect to the nextion display. I always get [nextion:091]: Not connected. See attached log file. Restarting does NOT solve the issue.

Reverting back to a older esphome solves the issue again and the nextion connects without a problem. 

[wandpaneelbven.yaml](https://github.com/user-attachments/files/22721196/wandpaneelbven.yaml)

[log.txt](https://github.com/user-attachments/files/22721197/log.txt)

edwardtfn, 2025-10-07:
Which device are you using?

paddenstoeltje, 2025-10-07:
Sonoff NSPanel

edwardtfn, 2025-10-07:
This should allow you using PSRAM:
```yaml
esp32:
  board: esp32dev
  framework:
    type: esp-idf
    sdkconfig_options:
      CONFIG_D0WD_PSRAM_CLK_IO: "5"
      CONFIG_D0WD_PSRAM_CS_IO: "18"
psram:
  id: ext_ram
```
😉