esphome/esphome #9876: Problems with compiling for m5stick-c since upgrade

NikNakk, 2025-07-24:
### The problem

Since upgrading to ESPHome 2025.7.0, I can't compile .yaml files with a board target of m5stick-c. I get the error:

[redacted]/.platformio/packages/framework-arduinoespressif32/cores/esp32/Arduino.h:43:10: fatal error: pins_arduino.h: No such file or directory

Changing the board to be a generic esp32dev board seems to fix the problem, but not sure if that will create other issues.

I've tried deleting the .esphome folder as well as the .platformio/packages/framework-arduinoespressif32 folder, and this hasn't fixed things.

### Which version of ESPHome has the issue?

2025.7.0

### What type of installation are you using?

pip

### What platform are you using?

ESP32

### Component causing the issue

sensor

### YAML Config

```yaml
substitutions:
  devicename: m5stickc-plus2
  friendly_name: M5StickC Plus2

esphome:
  name: $devicename
  friendly_name: M5StickC Plus2

esp32:
  board: m5stick-c

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: !secret api_password

ota:
  - platform: esphome
    password: !secret ota_password

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails

captive_portal:

sensor:
  - platform: wifi_signal
    name: "WiFi Signal Sensor"
    id: m5stick_wifi_signal
    update_interval: 10s
```

### Anything in the logs that might be useful for us?

```txt
Compiling .pioenvs\m5stickc-plus2\src\esphome\components\api\subscribe_state.cpp.o
In file included from src/esphome/core/macros.h:7,
                 from src/esphome/core/defines.h:2,
                 from src/esphome/components/api/api_connection.h:3,
                 from src/esphome/components/api/api_connection.cpp:1:
[redacted].platformio/packages/framework-arduinoespressif32/cores/esp32/Arduino.h:43:10: fatal error: pins_arduino.h: No such file or directory

**********************************************************************
* Looking for pins_arduino.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:pins_arduino.h"
* Web  > https://registry.platformio.org/search?q=header:pins_arduino.h
*
**********************************************************************

   43 | #include "pins_arduino.h"
      |          ^~~~~~~~~~~~~~~~
compilation terminated.
*** [.pioenvs\m5stickc-plus2\src\esphome\components\api\api_connection.cpp.o] Error 1
In file included from src/esphome/core/macros.h:7,
                 from src/esphome/core/defines.h:2,
                 from src/esphome/components/api/api_pb2.h:5,
                 from src/esphome/components/api/api_pb2.cpp:3:
[redacted].platformio/packages/framework-arduinoespressif32/cores/esp32/Arduino.h:43:10: fatal error: pins_arduino.h: No such file or directory

**********************************************************************
* Looking for pins_arduino.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:pins_arduino.h"
* Web  > https://registry.platformio.org/search?q=header:pins_arduino.h
*
**********************************************************************

   43 | #include "pins_arduino.h"
      |          ^~~~~~~~~~~~~~~~
compilation terminated.
*** [.pioenvs\m5stickc-plus2\src\esphome\components\api\api_pb2.cpp.o] Error 1
In file included from src/esphome/core/macros.h:7,
                 from src/esphome/core/defines.h:2,
                 from src/esphome/components/api/api_pb2_service.h:5,
                 from src/esphome/components/api/api_pb2_service.cpp:3:
[redacted].platformio/packages/framework-arduinoespressif32/cores/esp32/Arduino.h:43:10: fatal error: pins_arduino.h: No such file or directory

**********************************************************************
* Looking for pins_arduino.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:pins_arduino.h"
* Web  > https://registry.platformio.org/search?q=header:pins_arduino.h
*
**********************************************************************

   43 | #include "pins_arduino.h"
      |          ^~~~~~~~~~~~~~~~
compilation terminated.
*** [.pioenvs\m5stickc-plus2\src\esphome\components\api\api_pb2_service.cpp.o] Error 1
In file included from src/esphome/core/macros.h:7,
                 from src/esphome/core/defines.h:2,
                 from src/esphome/components/api/api_server.h:3,
                 from src/esphome/components/api/api_server.cpp:1:
[redacted].platformio/packages/framework-arduinoespressif32/cores/esp32/Arduino.h:43:10: fatal error: pins_arduino.h: No such file or directory

**********************************************************************
* Looking for pins_arduino.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:pins_arduino.h"
* Web  > https://registry.platformio.org/search?q=header:pins_arduino.h
*
**********************************************************************

   43 | #include "pins_arduino.h"
      |          ^~~~~~~~~~~~~~~~
compilation terminated.
*** [.pioenvs\m5stickc-plus2\src\esphome\components\api\api_server.cpp.o] Error 1
In file included from src/esphome/core/macros.h:7,
                 from src/esphome/core/defines.h:2,
                 from src/esphome/components/api/subscribe_state.h:3,
                 from src/esphome/components/api/subscribe_state.cpp:1:
[redacted].platformio/packages/framework-arduinoespressif32/cores/esp32/Arduino.h:43:10: fatal error: pins_arduino.h: No such file or directory

**********************************************************************
* Looking for pins_arduino.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:pins_arduino.h"
* Web  > https://registry.platformio.org/search?q=header:pins_arduino.h
*
**********************************************************************

   43 | #include "pins_arduino.h"
      |          ^~~~~~~~~~~~~~~~
compilation terminated.
*** [.pioenvs\m5stickc-plus2\src\esphome\components\api\subscribe_state.cpp.o] Error 1
In file included from src/esphome/core/macros.h:7,
                 from src/esphome/core/defines.h:2,
                 from src/esphome/components/api/list_entities.h:3,
                 from src/esphome/components/api/list_entities.cpp:1:
[redacted].platformio/packages/framework-arduinoespressif32/cores/esp32/Arduino.h:43:10: fatal error: pins_arduino.h: No such file or directory

**********************************************************************
* Looking for pins_arduino.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:pins_arduino.h"
* Web  > https://registry.platformio.org/search?q=header:pins_arduino.h
*
**********************************************************************

   43 | #include "pins_arduino.h"
      |          ^~~~~~~~~~~~~~~~
compilation terminated.
*** [.pioenvs\m5stickc-plus2\src\esphome\components\api\list_entities.cpp.o] Error 1
In file included from src/esphome/core/macros.h:7,
                 from src/esphome/core/defines.h:2,
                 from src/esphome/components/api/api_frame_helper.h:9,
                 from src/esphome/components/api/api_frame_helper.cpp:1:
[redacted].platformio/packages/framework-arduinoespressif32/cores/esp32/Arduino.h:43:10: fatal error: pins_arduino.h: No such file or directory

**********************************************************************
* Looking for pins_arduino.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:pins_arduino.h"
* Web  > https://registry.platformio.org/search?q=header:pins_arduino.h
*
**********************************************************************

   43 | #include "pins_arduino.h"
      |          ^~~~~~~~~~~~~~~~
compilation terminated.
*** [.pioenvs\m5stickc-plus2\src\esphome\components\api\api_frame_helper.cpp.o] Error 1
========================================================================================= [FAILED] Took 57.28 seconds =========================================================================================
```

greg-nyc, 2025-07-29:
I'm getting the same error using an older m5stick-c

randybb, 2025-07-29:
Boards are going to be obsolete, so even this is a bug just stop using `board:` and switch to `variant:` to make it future proof :)
https://next.esphome.io/components/esp32

Btw, any particular reason why are you using arduino?

kpfleming, 2025-07-31:
> Boards are going to be obsolete, so even this is a bug just stop using `board:` and switch to `variant:` to make it future proof :)
> https://next.esphome.io/components/esp32

`board` is required in the currently released versions of ESPHome, so people can't stop using it.

randybb, 2025-07-31:
Ah yes, this will be true from the next release. In meantime just use the default board for esp32, which is `esp32dev`