esphome/esphome #9878: Update from 2025.6.3 to 2025.7.X breaks wifi

crudolphy, 2025-07-24:
### The problem

I have a Home Assistant installation that runs HAOS now at 2025.7.3.  I have the ESPHome Device Builder addon now at 2025.7.3 (but this problem started at 2025.7.0).  I updated the addon to the 2025.7 series and then as is normal I needed to update two devices.  The problem device is an Wemos ESP32 Mini, that I have a sht3xd temp and humidity sensor attached to.  Up until the update to 2025.7.X I had no problem with this board and sensor connecting and updating over my wifi network for several years.

Upon updating to 2025.7 doing an OTA update it wouldn't update.  So I brought it down and connected it via usb to my computer and tried to update from the computer using a chrome browser.  It connected fine.  It compiled fine.  It installed fine.  Then it wouldn't connect to the wifi.

I googled rolling back to 2025.6.x and it told me that I could add "https://github.com/khendericks/esphome-legacy-addons to the add-on store repositories (which I did) and that I install the version I wanted to roll back to and run it along side the current version.  I did this and installed version 2025.6.3.

I started it up and went to it's web interface the two esp devices were there and I went to the problem device and installed its yaml config.  It did this fine and almost before I could close the message window and open the logs, it showed my device "on-line".

I then unplugged it from my computer and hooked it up to just a power supply.  It came on immediately and I was able to access the logs OTA.

So it would seem to me that something in the wifi setup changed in 2025.7 and needs to be fixed.  I wish I could fix it for you but am not a developer and wouldn't have a clue.  However I am more than willing to test things if you need me to. 

### Which version of ESPHome has the issue?

2025.7.3

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

### Component causing the issue

wifi

### YAML Config

```yaml
# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "e4bSbcheKYtqN9nB7YWAONfexDE+U7NqqDf7UsToeXw="

ota:
  platform: esphome
  password: "29b68e6994f6c76db6f8ebc93fbd23c9"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
    
  domain: ".ccrudolphy.com"
  manual_ip:
    static_ip: 192.168.30.81
    gateway: 192.168.30.1
    subnet: 255.255.255.0
    dns1: 192.168.1.1

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Humiditytempsensor"
    password: "VLzTHCQYfZaZ"

captive_portal:
i2c:
  - sda: 21
    scl: 22
    scan: true
    id: bus_a
sensor:
  - platform: sht3xd
    temperature:
      name: "test_temp_1"
    humidity:
      name: "temp_humidity_1"
    i2c_id: bus_a
    address: 0x44
    update_interval: 60s
```

### Anything in the logs that might be useful for us?

```txt
Current Log of working configuration:
INFO ESPHome 2025.6.3
INFO Reading configuration /config/esphome/humiditytempsensor.yaml...
INFO Starting log output from 192.168.30.81 using esphome API
INFO Successfully resolved humiditytempsensor @ 192.168.30.81 in 0.000s
INFO Successfully connected to humiditytempsensor @ 192.168.30.81 in 0.197s
INFO Successful handshake with humiditytempsensor @ 192.168.30.81 in 0.084s
[17:37:22][I][app:137]: ESPHome version 2025.6.3 compiled on Jul 24 2025, 17:01:12
[17:37:22][C][wifi:613]: WiFi:
[17:37:22][C][wifi:434]:   Local MAC: 30:C6:F7:99:17:A8
[17:37:22][C][wifi:439]:   SSID: [redacted]
[17:37:22][C][wifi:442]:   IP Address: 192.168.30.81
[17:37:22][C][wifi:451]:   BSSID: [redacted]
[17:37:22][C][wifi:451]:   Hostname: 'humiditytempsensor'
[17:37:22][C][wifi:451]:   Signal strength: -48 dB ▂▄▆█
[17:37:22][C][wifi:462]:   Channel: 1
[17:37:22][C][wifi:462]:   Subnet: 255.255.255.0
[17:37:22][C][wifi:462]:   Gateway: 192.168.30.1
[17:37:22][C][wifi:462]:   DNS1: 192.168.1.1
[17:37:22][C][wifi:462]:   DNS2: 0.0.0.0
[17:37:22][C][logger:215]: Logger:
[17:37:22][C][logger:215]:   Max Level: DEBUG
[17:37:22][C][logger:215]:   Initial Level: DEBUG
[17:37:22][C][logger:220]:   Log Baud Rate: 115200
[17:37:22][C][logger:220]:   Hardware UART: UART0
[17:37:22][C][logger:224]:   Task Log Buffer Size: 768
[17:37:22][C][i2c.arduino:072]: I2C Bus:
[17:37:22][C][i2c.arduino:077]:   SDA Pin: GPIO21
[17:37:22][C][i2c.arduino:077]:   SCL Pin: GPIO22
[17:37:22][C][i2c.arduino:077]:   Frequency: 50000 Hz
[17:37:22][C][i2c.arduino:089]:   Recovery: bus successfully recovered
[17:37:22][I][i2c.arduino:099]: Results from bus scan:
[17:37:22][I][i2c.arduino:105]: Found device at address 0x44
[17:37:22][C][sht3xd:049]: SHT3xD:
[17:37:22][D][sht3xd:064]:   Setup successful
[17:37:22][D][sht3xd:065]:   Serial Number: 0x122F7BFD
[17:37:22][D][sht3xd:066]:   Heater Enabled: false
[17:37:22][C][sht3xd:068]:   Address: 0x44
[17:37:22][C][sht3xd:069]:   Update Interval: 60.0s
[17:37:22][C][sht3xd:071]:   Temperature 'test_temp_1'
[17:37:22][C][sht3xd:071]:     State Class: 'measurement'
[17:37:22][C][sht3xd:071]:     Unit of Measurement: '°C'
[17:37:22][C][sht3xd:071]:     Accuracy Decimals: 1
[17:37:22][C][sht3xd:071]:     Device Class: 'temperature'
[17:37:22][C][sht3xd:072]:   Humidity 'temp_humidity_1'
[17:37:22][C][sht3xd:072]:     State Class: 'measurement'
[17:37:22][C][sht3xd:072]:     Unit of Measurement: '%'
[17:37:22][C][sht3xd:072]:     Accuracy Decimals: 1
[17:37:22][C][sht3xd:072]:     Device Class: 'humidity'
[17:37:22][C][captive_portal:089]: Captive Portal:
[17:37:22][C][esphome.ota:077]: Over-The-Air updates:
[17:37:22][C][esphome.ota:077]:   Address: 192.168.30.81:3232
[17:37:22][C][esphome.ota:077]:   Version: 2
[17:37:22][C][esphome.ota:080]:   Password configured
[17:37:22][C][safe_mode:018]: Safe Mode:
[17:37:22][C][safe_mode:025]:   Boot considered successful after 60 seconds
[17:37:22][C][safe_mode:025]:   Invoke after 10 boot attempts
[17:37:22][C][safe_mode:025]:   Remain for 300 seconds
[17:37:22][C][api:185]: API Server:
[17:37:22][C][api:185]:   Address: 192.168.30.81:6053
[17:37:22][C][api:187]:   Using noise encryption: YES
[17:37:22][C][mdns:125]: mDNS:
[17:37:22][C][mdns:125]:   Hostname: humiditytempsensor
[17:37:28][D][sht3xd:097]: Got temperature=31.81°C humidity=35.07%
[17:37:28][D][sensor:099]: 'test_temp_1': Sending state 31.80667 °C with 1 decimals of accuracy
[17:37:28][D][sensor:099]: 'temp_humidity_1': Sending state 35.06981 % with 1 decimals of accuracy
```

### Additional information

I am sorry but I don't have a "failed" log from 2025.7.  I do remember that it scanned the network, found the proper ssid, and the error was "No Response" or something like that.

ssieb, 2025-07-24:
Can you include the rest of the config?
We'll also need to see the serial logs from when it wasn't working.

crudolphy, 2025-07-24:
Hope this is what you need.  I am pretty sure that is all the config.  But
here is a copy and paste again.

esphome:
name: humiditytempsensor

esp32:
board: esp32dev
framework:
type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
encryption:
key: "e4bSbcheKYtqN9nB7YWAONfexDE+U7NqqDf7UsToeXw="

ota:
platform: esphome
password: "29b68e6994f6c76db6f8ebc93fbd23c9"

wifi:
ssid: !secret wifi_ssid
password: !secret wifi_password
domain: ".ccrudolphy.com"
manual_ip:
static_ip: 192.168.30.81
gateway: 192.168.30.1
subnet: 255.255.255.0
dns1: 192.168.1.1

# Enable fallback hotspot (captive portal) in case wifi connection fails
ap:
ssid: "Humiditytempsensor"
password: "VLzTHCQYfZaZ"

captive_portal:
i2c:
- sda: 21
scl: 22
scan: true
id: bus_a
sensor:
- platform: sht3xd
temperature:
name: "test_temp_1"
humidity:
name: "temp_humidity_1"
i2c_id: bus_a
address: 0x44
update_interval: 60s

crudolphy, 2025-07-24:
That is the entire yaml config for this esp32.  Here are the logs

[5c53de3b_esphome_2025-07-24T23-04-42.576Z.log](https://github.com/user-attachments/files/21420345/5c53de3b_esphome_2025-07-24T23-04-42.576Z.log)

closed 2025-07-24 by ssieb