esphome/esphome #9880: Update from 2025.6.3 to 2025.7.x breaks the esp32_touch on ESP32S3

Th30wl, 2025-07-24:
### The problem

I am using built in touch sensor on Seeed Studio XIAO ESP32-S3 and with current config it worked before version 2025.7 

I now that the current config is already a workaround for ESP32-S3 but it worked so far, it likely stopped working because of switching to interrupt based approach. Also since 2025.7 setting `setup_mode: True` does not make it log values for some reason.

Current workaround is to use `esp32_touch` component from the 2025.6.3 version by setting up as external component:

```
external_components:
  - source:
      type: git
      url: https://github.com/esphome/esphome
      ref: 2025.6.3
    components: [esp32_touch]
```

I still get the deprecation warnings on build but everything works so far.

### Which version of ESPHome has the issue?

2025.7.3

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

### Component causing the issue

esp32_touch

### YAML Config

```yaml
#config contains workaround 

external_components:
  - source:
      type: git
      url: https://github.com/esphome/esphome
      ref: 2025.6.3
    components: [esp32_touch]

esphome:
  name: kids-room-main-light
  friendly_name: "Kids' room main light"
  on_boot: 
    then:
      - delay: 1s
      - if:
          condition: 
            - light.is_on: main_light
          then: 
            - light.turn_off: main_light #no idea why is it initially on, so i had to turn it off at boot

esp32:
  board: esp32-s3-devkitc-1
  framework:
    type: esp-idf
  variant: esp32s3 

esp32_touch:
  setup_mode: False
  measurement_duration: 0.25ms
  

logger:
  level: WARN
  baud_rate: 0 

api:
  encryption:
    key:   !secret api_key
  reboot_timeout: 15min
  

ota:
  - platform: esphome
    password:  !secret ota_password

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  fast_connect: true
  reboot_timeout: 5min

output:
  - platform: gpio
    pin: 
      number: GPIO9
      mode:
        output: true
        pullup: false
        pulldown: true
    id: relay_output
    setup_priority: -100
    
  
  - platform: ledc
    pin: GPIO44
    id: led_output
    frequency: 19531Hz
    channel: 0

globals:
  - id: touch_start_time
    type: uint32_t
    restore_value: false
    initial_value: '0'

binary_sensor:
  - platform: esp32_touch
    internal: True
    name: "Light Touch Sensor"
    pin: GPIO8
    threshold: 113000
    filters:
      - delayed_on: 100ms
      - delayed_off: 100ms
    on_press:
      then:
        - globals.set:
            id: touch_start_time
            value: !lambda 'return millis();'
        - lambda: |-
            ESP_LOGI("touch", "touch pressed");
    on_release:
      then:
        - lambda: |-
            ESP_LOGI("touch", "touch released");
            uint32_t press_duration = millis() - id(touch_start_time);
            if (press_duration >= 200 && press_duration < 1000) {
              id(main_light).toggle().perform();
            }
            else if (press_duration >= 5000 && press_duration < 10000) {
              id(restart_button).press();
            }

sensor:
  - platform: uptime
    name: Uptime

  - platform: wifi_signal
    name: "Wifi Signal"
    update_interval: 5min

light:
  - platform: binary
    name: "Main Light"
    output: relay_output
    id: main_light
    initial_state: 
      state: False
    
  
  - platform: monochromatic
    name: "LED Light"
    output: led_output
    id: led_light
    effects:
      - lambda:
          name: "Realistic Fire" #generated by LLM i admit :)
          update_interval: 30ms
          lambda: |-
            static float target_brightness = 0.6;
            static float current_brightness = 0.6;
            static uint32_t last_change = 0;
            
            uint32_t now = millis();
            
            // Change target brightness randomly every 100-300ms (optimized)
            if (now - last_change > (100 + (esp_random() % 200))) {
              // More dramatic brightness variations for visible flicker
              target_brightness = 0.3 + (esp_random() % 50) / 100.0; // 30%-80% range
              last_change = now;
            }
            
            // Smooth transition to target with some random jitter
            float diff = target_brightness - current_brightness;
            current_brightness += diff * 0.2; // Reduced from 0.3 for smoother transition
            
            // Add random jitter for realistic flicker
            float jitter = (esp_random() % 20 - 10) / 100.0; // ±10% jitter
            float final_brightness = current_brightness + jitter;
            
            // Clamp to visible range
            final_brightness = max(0.25f, min(0.8f, final_brightness));
            
            auto call = id(led_light).make_call();
            call.set_brightness(final_brightness);
            call.set_transition_length(0);
            call.perform();

button:
  - platform: restart
    name: "Restart Device"
    id: restart_button
```

### Anything in the logs that might be useful for us?

```txt

```

Th30wl, 2025-07-25:
> You aren't getting any log messages because you set the log level to WARN and the log messages are at DEBUG. 

Ah sorry about that, switched to debug and captured some logs, the touches are being detected in the logs but not by the binary sensor unless I use the 2026.6.3 version of the `esp32_touch` component which makes it even more strange now. 
I didn't change the threshold but since it goes over the threshold wen touched I assume that it is still a good value.

> Also, why did you disable serial logging?

It is in the wall, it would be too difficult to attach a cable so I disabled it.

RaStrNL, 2025-07-27:
I can confirm that problem. I have ESP232S3 custom board, and with the ESPH 2025.7.3 binary sensors are not working propery. Even with reported value over the configured treshold, nothing is changed with the binary sensor. I have to go much higher than what's reasonable in real world, then the binary sensor reacts. With older version of ESPHome it was working, but I haven't updated this device for quiet some time, so I do not know since which version is it broken. I have eve tryed to shorten measurement time to 0.25ms as indicated in the WIKI, I have similar values as shown here with the same results. Can try even shorter interval now.

RaStrNL, 2025-07-27:
I think I have discovered the problem. Looked in the current code and they use some autocalibration value called Benchmark which is automatically added to provided threshold. If you run debug via serial (via WiFi I believe this is omitted from the log), you can see in the initialization something like:
_C][esp32_touch:019]: Config for ESP32 Touch Hub:
                                                   Meas cycle: 0.05ms
                                                                       Sleep cycle: 5.00ms
                                                                                            Low Voltage Reference: 0.5V
                                                                                                                         High Voltage Reference: 2.7V
                                                                                                                                                       Voltage Attenuation: 0V
                                                                                                                                                                                Release Timeout: 100ms
[C][esp32_touch:258]:   Setup Mode ENABLED
[C][esp32_touch:033]:   Touch Pad 'Button 1'
[C][esp32_touch:034]:     Pad: T1
                                     Threshold: 24000
                                                         Benchmark: 21031
[C][esp32_touch:033]:   Touch Pad 'Button 2'
[C][esp32_touch:034]:     Pad: T3
                                     Threshold: 107000
                                                          Benchmark: 21078
Even the formatting of the debug is broken (what a crappy work of the developer). From the value indicated here, I expect this to be some average value detected on boot. This is not described anywhere in the doc. Could be a nice feature if documented properly !
Going to do something meaningful instead of finding quirks in open source :-) !

Th30wl, 2025-08-06:
I think I'm on to something. It seems that in my case the issue is because of [this metod](https://github.com/esphome/esphome/pull/9059/files#diff-c2faaab51670586bf980fbe0a99f3a9e1e167b3d99625c2ef3da0a2a6e5836b6R133) 
In my code i handle the `on_release` and measure the time passed since `on_press` and if the touch has been released between 200ms and 1000ms I toggle the switch, but there is probably a mismatch between this and the calculation in `calculate_release_timeout_()` .

This is still a theory, I will try to test it after I come back from my vacation.

clydebarrow, 2025-08-06:
The ESP32-S3 and -S2 touch implementation changed, and the threshold is now the value delta to trigger on, not the absolute threshold. The docs don't appear to have been updated accordingly.

Th30wl, 2025-08-22:
> The ESP32-S3 and -S2 touch implementation changed, and the threshold is now the value delta to trigger on, not the absolute threshold. The docs don't appear to have been updated accordingly.

I can now confirm this. Changing the threshold to 2000 fixed it for me.