esphome/esphome #9895: Build error after update to 2025.7.3

ok11, 2025-07-25:
### The problem

The build fails.

### Which version of ESPHome has the issue?

2025.7.3

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

### Component causing the issue

esp-idf

### YAML Config

```yaml
substitutions:
  host_name: bt-proxy-02
  friendly_name: Bluetooth Proxy 2
  proxy_name: bt_proxy_02

esphome:
  name: ${host_name}
  friendly_name: ${friendly_name}
  min_version: 2025.5.0
  name_add_mac_suffix: false

esp32:
  board: esp32-c3-devkitm-1
  framework:
    type: esp-idf

web_server:
  port: 80
  include_internal: true

logger:

api:
  encryption:
    key: !secret api_encryption_key

ota:
  - platform: esphome
    password: !secret ota_password

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

esp32_ble_tracker:
  scan_parameters:
    active: true

bluetooth_proxy:
  active: true
```

### Anything in the logs that might be useful for us?

```txt
-- git rev-parse returned 'fatal: not a git repository (or any parent up to mount point /)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).'
-- Building ESP-IDF components for target esp32
-- Configuring incomplete, errors occurred!

fatal: not a git repository (or any parent up to mount point /)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).
CMake Error at /data/cache/platformio/packages/framework-espidf@src-c44434145e05010467d5d5a727b42ef9/tools/cmake/build.cmake:552 (message):
  /root/.platformio/penv/.espidf-5.3.2/bin/python: Error while finding module
  specification for 'idf_component_manager.prepare_components'
  (ModuleNotFoundError: No module named 'idf_component_manager')

Call Stack (most recent call first):
  /data/cache/platformio/packages/framework-espidf@src-c44434145e05010467d5d5a727b42ef9/tools/cmake/project.cmake:710 (idf_build_process)
  CMakeLists.txt:3 (project)
```

### Additional information

The issue is reproducible on two devices (esp-idf based bt proxies).

ok11, 2025-07-25:
I'm just wondering. I have quite a few devices now, will they be automatically re-discovered and added?

closed 2025-07-26 by ok11

fonske, 2025-07-27:
I had the same issue.
A restart of the esphome 2025.7.3 add-on solved it.