esphome/esphome #9901: esp-idf failing to parse json

Kernald, 2025-07-26:
### The problem

I have a board ([Waveshare ESP32-S3-ETH](https://www.waveshare.com/esp32-s3-eth.htm)) configured to fetch OTAs over http_request from a manifest. It doesn't seem to successfully parse the manifest, so I tried doing a similar bit of logic manually (with the domain redacted):

```
button:
  - platform: template
    name: "Fetch latest release"
    on_press:
      - http_request.get:
          url: https://esphome-firmwares.s3.domain.com/${device_name}/manifest.json
          capture_response: true
          on_response:
            then:
              - if:
                  condition:
                    lambda: return response->status_code == 200;
                  then:
                    - logger.log:
                        tag: "ota_check"
                        format: "Fetched JSON body: %s"
                        args: ["body.c_str()"]
                    - lambda: |-
                        json::parse_json(body, [](JsonObject root) -> bool {
                            if (root["version"]) {
                                ESP_LOGI(
                                  "ota_check",
                                  "Latest release: %s",
                                  root["version"].as<String>()
                                );
                                return true;
                            }
                            else {
                              ESP_LOGW("ota_check","Failed to parse json!");
                              return false;
                            }
                        });
                  else:
                    - logger.log:
                        tag: "ota_check"
                        format: "Error: Response status: %d, message %s"
                        args: ["response->status_code", "body.c_str()"]
```

If I compile this with the `arduino` framework, it works just fine (and I suspect the OTA would work as well, but I haven't verified this). However, if I compile with the `esp-idf` framework, I get garbage:

```
[11:17:28][I][ota_check:046]: Latest release: \x90\xcf\xca? 
```

If I compile with the default (which from what I understand is `arduino`), the exact same code fetching the exact same JSON works just fine:

```
[12:02:46][I][ota_check:046]: Latest release: 2025.07.26.129
```

This is the JSON (with domains redacted):

```
{"name":"bt-proxy-tv","version":"2025.07.26.129","builds":[{"chipFamily":"ESP32S3","uploaded_at":"2025-07-26T01:12:27Z","ota":{"path":"/bt-proxy-tv/cb445f3ad290e0ce2c1bca950ebcd8c3173233f0/firmware.ota.bin","size":1264512,"md5":"98d301786cfa9ba901e27de501abe7a5","release_url":"https://git.domain.com/kernald/esphome/commit/cb445f3ad290e0ce2c1bca950ebcd8c3173233f0"},"source":{"commit":"cb445f3ad290e0ce2c1bca950ebcd8c3173233f0","repository":"https://git.domain.com/kernald/esphome","url":"https://git.domain.com/kernald/esphome/commit/cb445f3ad290e0ce2c1bca950ebcd8c3173233f0"}}]}
```

I tried minimizing the JSON like so to try and reduce the memory usage while parsing the string, but this didn't make a difference.

### Which version of ESPHome has the issue?

2025.7.2

### What type of installation are you using?

pip

### What platform are you using?

ESP32

### Component causing the issue

http_request

### YAML Config

```yaml
---
esphome:
  name: ${device_name}
  project:
    # Project name format: [owner].[device_name]
    name: "kernald.${device_name}"

    # Version format: YYYY.MM.DD.BUILD
    # This is automatically overridden in CI/CD pipelines
    version: "2025.07.23.0"
  platformio_options:
    board_build.arduino.memory_type: qio_opi
    board_build.flash_mode: dio
    upload_speed: 921600
    monitor_speed: 115200

esp32:
  board: esp32-s3-devkitc-1
  variant: esp32s3
  flash_size: 16MB
  framework:
    type: esp-idf
    sdkconfig_options:
      CONFIG_ESP32_S3_BOX_BOARD: "y"

psram:
  mode: octal
  speed: 80MHz

ethernet:
  type: W5500
  clk_pin: GPIO13
  mosi_pin: GPIO11
  miso_pin: GPIO12
  cs_pin: GPIO14
  interrupt_pin: GPIO10
  reset_pin: GPIO9

text_sensor:
  - platform: ethernet_info
    ip_address:
      name: ESP IP Address
      address_0:
        name: ESP IP Address 0
      address_1:
        name: ESP IP Address 1
      address_2:
        name: ESP IP Address 2
      address_3:
        name: ESP IP Address 3
      address_4:
        name: ESP IP Address 4
    dns_address:
      name: ESP DNS Address

logger:

web_server:

prometheus:
  include_internal: true

api:
  password: !secret ApiPassword

http_request:
  # verify_ssl isn't supported with arduino, it only works with esp-idf
  verify_ssl: false

ota:
  - platform: http_request
  - platform: "esphome"
    password: !secret OtaPassword

update:
  - platform: http_request
    id: http_firmware_update
    name: "Firmware update"
    source: https://esphome-firmwares.s3.domain.com/${device_name}/manifest.json
    update_interval: 1h
    on_update_available:
      then:
        - update.perform:
...
```

### Anything in the logs that might be useful for us?

```txt
[12:12:52][D][button:010]: 'Fetch latest release' Pressed.
[12:12:52][D][http_request.idf:043]: Received response header, name: content-length, value: 579
[12:12:52][D][http_request.idf:043]: Received response header, name: content-type, value: application/json
[12:12:53][D][ota_check:042]: Fetched JSON body: {"name":"bt-proxy-tv","version":"2025.07.26.129","builds":[{"chipFamily":"ESP32S3","uploaded_at":"2025-07-26T01:12:27Z","ota":{"path":"/bt-proxy-tv/cb445f3ad290e0ce2c1bca950ebcd8c3173233f0/firmware.ota.bin","size":1264512,"md5":"98d301786cfa9ba901e27de501abe7a5","release_url":"https://git.domain.com/kernald/esphome/commit/cb445f3ad290e0ce2c1bca950ebcd8c3173233f0"},"source":{"commit":"cb445f3ad290e0ce2c1bca950ebcd8c3173233f0","repository":"https://git.domain.com/k
[12:12:53][I][ota_check:046]: Latest release: \xc0\xee\xca?
[12:12:53][W][component:407]: api took a long time for an operation (523 ms)
[12:12:53][W][component:408]: Components should block for at most 30 ms
[12:12:53][W][esp32_ble:407]: Dropped 4 BLE events due to buffer overflow
[12:12:53][W][esp32_ble_tracker:177]: Dropped 28 BLE scan results due to buffer overflow
```

ssieb, 2025-07-26:
You can't use `String` in esp-idf because that an Arduino thing.  So what was the code you used in esp-idf?

Kernald, 2025-07-26:
This is exactly what I had - I guess that makes sense that it didn't work though. I didn't think much about it given that it compiled. Using `root["version"].as<std::string>().c_str()` instead does work, and I can properly extract the version number.

I was mostly doing this to try and diagnose why `update.http_request` didn't work, but this didn't help. Pointing at the same JSON manifest, I get this:

```
[16:56:10][E][component:286]: http_request.update set Error flag: Failed to parse JSON from https://esphome-firmwares.s3.enoent.fr/bt-proxy-tv/manifest.json
```

I haven't been able to apply updates on Arduino as well on this device yet, it seems to be causing a reset:

```
[17:04:50]p_request.arduino:140][update_task]: Content-Length: 579
[17:04:50][D][http_request.arduino:140][update_task]: Content-Length: 579
[17:04:50][D][http_request.arduino:140][update_task]: Content-Length: 579
[17:04:50][I][app:164]: ESPHome version 2025.7.2 compiled on Jan  1 1980, 00:00:00
[17:04:50][I][app:166]: Project kernald.bt-proxy-tv version 2025.07.23.0
```

ssieb, 2025-07-26:
Can you get serial logs of the crash?

ssieb, 2025-07-26:
There's a define called `ESPHOME_VARIANT` that contains the value you need.  You can find that in the build directory at `src/esphome/core/defines.h`
It's one of the values from this list: https://github.com/esphome/esphome/blob/dev/esphome/components/esp32/const.py#L36

Kernald, 2025-07-26:
Yeah, it works with that. Sorry about the noise - thanks!

closed 2025-07-26 by Kernald