esphome/esphome #9906: esphome::lerp function signature changed, broken in 2025.7.3

j9brown, 2025-07-26:
### The problem

Change #9246 broke esphome:lerp by replacing the implementation with std::lerp whose parameters are in a different order, breaking all callers of this function.

Before this change:

```
float lerp(float completion, float start, float end) { return start + (end - start) * completion; }
```

After this change:

```
using std::lerp;

// according to https://en.cppreference.com/w/cpp/numeric/lerp.html
constexpr float       lerp( float a, float b, float t ) noexcept;
```

I think this part of the change should be reverted and a deprecation process used to suggest replacement and eventual removal of the function.

Specifically, this change broke one of my custom components.

### Which version of ESPHome has the issue?

2025.7.3

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

### Component causing the issue

core

### YAML Config

```yaml
n/a
```

### Anything in the logs that might be useful for us?

```txt
Not that it will help much, but a sensor in one of my custom components is reporting an erroneous negative value due to the change of the lerp definition.

[22:01:44][D][sensor:104]: 'Fresh Water Level': Sending state 4.80000  with 2 decimals of accuracy
[22:01:44][D][sensor:104]: 'Fresh Water Volume': Sending state -31.89027 L with 1 decimals of accuracy
```

HeMan, 2025-07-31:
Didn't https://github.com/esphome/esphome/pull/9914, which is part of 2025.7.4, make it fail during compilation?

ssieb, 2025-07-31:
The decision was to not have backwards compatibility.  Any external component using this function needs to switch to the `std::` one.  The function was marked as deleted so that we could flag the comment about changing the argument order.

closed 2025-07-31 by ssieb