esphome/esphome #9913: esp32_touch does not work anymore after update to 2025.7.3 only returning 0

DerEmder, 2025-07-26:
### The problem

Raspberry Pi 4
Installationsmethode Home Assistant OS
Core 2025.7.3
Supervisor 2025.07.1
Operating System 16.0
Frontend 20250702.3
ESPHome 2025.7.3

The esp_touch worked flawlessly until the update to 2025.7.3. After that it only returns 0.

The only way to get live values is to set setup_mode to true - but that's not how it should be, shouldn't it. And its flooding the log as a not so welcome side effect.  

I removed/changed the esp_touch parameters as well - has no effect.

In short:

setup_mode: false -> nothing, just 0
setup_mode: true -> works, but floods the log 

Here are the relevant yaml entries:

```
esp32_touch:
  setup_mode: false
  iir_filter: 10ms
  measurement_duration: 6ms
  low_voltage_reference: 0.8V
  high_voltage_reference: 2.5V
  voltage_attenuation: 1.5V
```
```
  - platform: esp32_touch
    name: '$upper_devicename Read T7'
    internal: true
    pin: GPIO27
    threshold: 10
    id: cap_sensor
```
```
  - platform: template
    name: '$upper_devicename CapSensor'
    lambda: |-
      return id(cap_sensor).get_value();
    filters:
      - filter_out: nan
      - min:
          window_size: 15
          send_every: 15
          send_first_at: 3
      - sliding_window_moving_average:
          window_size: 15
          send_every: 15
    device_class: frequency
    unit_of_measurement: 'hz'
    accuracy_decimals: 0
    update_interval: 10s
```

### Which version of ESPHome has the issue?

2025.7.3

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

### Component causing the issue

esp32_touch

### YAML Config

```yaml

```

### Anything in the logs that might be useful for us?

```txt

```

### Additional information

I use the ESP32 board in a water tank to measure the water height or value by connecting two wires of 1,5m each (solar cables) to GND and a touch pin. Works better than I have ever expected.

swoboda1337, 2025-07-26:
Which device is this? ESP32, ESP32 S3 or ESP32 S2?

swoboda1337, 2025-07-26:
If you want values and don't care about the binary sensor then just increase the threshold to the max: 
```
    threshold: 0xFFFF
```

DerEmder, 2025-07-27:
Yes I desperately want values and really don't care about the binary sensor (I swear...). But anyway, the explanation on the ESPHome website is wrong then (or, at least, misleading). As I wrote above - it is a gold old plain ESP32 (absolutely no suffix). And it worked pre-2025.7.3. Why did they change that and break things?

And to answer - yes, I did change the threshold and now it shows values again. Thank you for the info, you saved my sunday and I can close this now :)

ssieb, 2025-07-27:
I don't know why anything would have changed.  Maybe it's something to do with the newer IDF version.  You didn't include the full yaml, so if you're using Arduino, it would be a really big version change.

swoboda1337, 2025-07-27:
@ssieb a lot changed when we moved to interrupt based code, the value doesn't get update if its under the threshold anymore, need to update the docs.

DerEmder, 2025-07-28:
> [@ssieb](https://github.com/ssieb) need to update the docs.

Yes, that would be a very good idea. Maybe it would even be possible to update the docs even before (or at the same time) you release new versions with such breaking changes. But anyway, my problem is solved and the issue closed, thank you all.