esphome/esphome #9915: Unable to run extended config on ESPHome 7.x, worked on ESPHome 6.x

TJB88, 2025-07-26:
### The problem

I have an extended ESPHome YAML config with SSD1306, time, WS2812 and sensor input from HA on an ESP8266. Even ChatGPT could not help me and is suggesting that it has something to do with the RAM memory. I already excluded webserver (large RAM consumer).

What I did so far:

1. I wirelessly updated the device as usual from ESPHome 6.x to ESPHome 7.3 from HA GUI.
2. It came never back online, 
3. I succesfully flashed ESPHome 7.3 with the same YAML file via USB connected to my computer.
4. Display was working correctly, but it was unable to connect to WiFi.

`[20:17:27][W][wifi_esp8266:513]: Disconnected ssid=[redacted] bssid=[redacted] reason='Auth Expired'`

5. I was sure that SSID and Password were OK.

6. To excluded my ESP device, I flashed it with minmal config. And yes, it was correctly connecting the internet and my HA instance (I have some health/status sensors and they all came through). 

Looking to all retries, I found some interesting logging:

`Exception (28): LoadProhibited: A load referenced a page mapped with an attribute that does not permit loads`

`[20:24:52][I][wifi:319]: Connecting to '[redacted]'
[20:24:54][W][wifi_esp8266:513]: Disconnected ssid='[redacted]' bssid=[redacted] reason='Auth Expired'
[20:24:54][W][wifi:666]: Connecting to network failed
[20:24:54][I][wifi:319]: Connecting to '[redacted]'
[20:24:56][I][wifi:630]: Connected
[20:24:56][C][wifi:434]:   Local MAC: CC:50:E3:F3:DD:97
[20:24:56][C][wifi:439]:   SSID: [redacted]
[20:24:56][C][wifi:442]:   IP Address: 192.168.4.1
[20:24:56][C][wifi:442]:   IP Address: 192.168.1.85
[20:24:56][C][wifi:446]:   BSSID: [redacted]
[20:24:56][C][wifi:446]:  Hostname: 'display'
[20:24:56][C][wifi:446]:  Signal strength: -57 dB ▂▄▆█
[20:24:56][C][wifi:455]:   Channel: 11
[20:24:56][C][wifi:455]:  Subnet: 255.255.255.0
[20:24:56][C][wifi:455]:  Gateway: 192.168.1.254
[20:24:56][C][wifi:455]:  DNS1: 192.168.1.254
[20:24:56][C][wifi:455]:  DNS2: 8.8.8.8
[20:24:56]
[20:24:56]--------------- CUT HERE FOR EXCEPTION DECODER ---------------
[20:24:56]
[20:24:56]Exception (28):
[20:24:56]epc1=0x4022bc76 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000014 depc=0x00000000
[20:24:56]
[20:24:56]>>>stack>>>
[20:24:56]
[20:24:56]ctx: cont
[20:24:56]sp: 3ffffdc0 end: 3fffffd0 offset: 0150
[20:24:56]3fffff10:  3fff7600 3fff75f4 00000000 00000000  
[20:24:56]3fffff20:  3ffe913f 3fff434c 3fff424c 4022324c  
[20:24:56]3fffff30:  3fffff38 0000000b 74736557 616b7265  
[20:24:56]3fffff40:  004d6564 00000001 0007023d 4023e0b0  
[20:24:56]3fffff50:  3fff391c 00000000 3fff2c60 3fff2a1c  
[20:24:56]3fffff60:  0007023d 3fff434c 3fff424c 40223424  
[20:24:56]3fffff70:  0007023d 0007022a 00000001 3fff424c  
[20:24:56]3fffff80:  00000000 0007023d 3fff2a1c 40224ce7  
[20:24:56]3fffff90:  0007023d 3fff424c 00000048 3fff3024  
[20:24:56]3fffffa0:  3fffdad0 00000000 3fff2ff8 3fff3024  
[20:24:56]3fffffb0:  3fffdad0 00000000 3fff2ff8 402396f0  
[20:24:56]3fffffc0:  feefeffe feefeffe 3fffdab0 40100495  
[20:24:56]<<<stack<<<
[20:24:56]
[20:24:56]--------------- CUT HERE FOR EXCEPTION DECODER ---------------
[20:24:56]
[20:24:56][C][wifi:455]: ets Jan  8 2013,rst cause:2, boot mode:(3,6)
[20:24:56]
[20:24:57]load 0x4010f000, len 3424, room 16 
[20:24:57]tail 0
[20:24:57]chksum 0x2e
[20:24:57]load 0x3fff20b8, len 40, room 8 
[... 285 lines cut ...]
    update_interval: 300s
    disabled_by_default: false
    

binary_sensor:
  - platform: status  # Extra sensor for the connection state
    name: "Status"
    disabled_by_default: false

text_sensor:
  - platform: wifi_info
    ip_address:
      name: "IP Address"
      disabled_by_default: false
      id: ip_address
    mac_address:
      name: "Mac Address"
      disabled_by_default: false
```

### Anything in the logs that might be useful for us?

```txt

```

ssieb, 2025-07-26:
Most likely you're running out of RAM.  Can you get the stack decoded?  You can use https://esphome.github.io/esp-stacktrace-decoder/

TJB88, 2025-07-26:
Thanks for your quick reply. I loaded both the stack trace and .elf file, but no response.
ChatGPT is also pointing in the direction of a memory issue.

It is a pity, but I think I should do something different or go back to ESPHome 6.x

✅ Common Causes of Exception 28
Dereferencing a null or invalid pointer.
Accessing an uninitialized object (especially in asynchronous callbacks).
Use-after-free errors or heap corruption.
Array index out of bounds.
Using a member of a freed WiFiClient, String, or similar dynamic object.

TJB88, 2025-07-27:
I have a strong suspicion that the display configuration is causing the strange WiFi connection issue and not the memory part.

```
display:
  - platform: ssd1306_i2c
```
To investigate, I added a simple free heap monitor:

```
sensor:
  - platform: template
    name: "Free Heap"
    lambda: "return ESP.getFreeHeap();"
    update_interval: 30s
```
The free heap consistently stays within acceptable thresholds (usually above ~15KB), even with the display enabled.

Is anyone else WiFi connection issues when using ssd1306_i2c working in ESPHome 6.x, but stopped working in ESPHome 7.x?

TJB88, 2025-10-26:
This is the solution

https://community.home-assistant.io/t/esphome-device-does-not-boot-up-when-configuring-i2c/579009/11