esphome/esphome #9940: Warning when compiling i2s component (warning "legacy adc driver is deprecated")

Jn115759, 2025-07-28:
### The problem

When compiling a ESP32 WROOM 32D mcu using the i2c component I noticed a warning regarding a deprecated driver. This might start causing problems in the future. I do not know or I can resolve this locally (always interested to learn). Would appreciate if a skilled person could take a closer look.

### Which version of ESPHome has the issue?

2025.7.3

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

### Component causing the issue

i2c

### YAML Config

```yaml
################################################################################
# Substitution Variables
################################################################################
substitutions:
  name: esp32speaker1
  friendly_name: "Esp32 Speaker 1"
  
  board_id: esp32dev
  framework_used: arduino

  device_sampling_time: 30s

################################################################################
# Globals
################################################################################
globals: ##define global variable to store wifi connection status
  - id: wifi_connection
    type: bool
    restore_value: no
    initial_value: "false"

################################################################################
# Board Configuration
################################################################################
esphome:
  name: ${name}
  friendly_name: ${friendly_name}

esp32:
  board: ${board_id}
  framework:
    type: ${framework_used}

################################################################################
# Enable logging
################################################################################
logger:

################################################################################
# Enable Home Assistant API
################################################################################
api:
  encryption:
    key: !secret api_encryption_key

################################################################################
# OTA
################################################################################
ota:
- platform: esphome
  password: !secret wifi_password

################################################################################
# WiFi
################################################################################
wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

 # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: ${name}-wifi
    password: !secret wifi_password

captive_portal:

################################################################################
# Time
################################################################################
time:
  - platform: homeassistant

################################################################################
# Binary Sensors
################################################################################
binary_sensor:
  - platform: status
    name: "Status"
    id: ${name}_status

################################################################################
# Sensors
################################################################################
sensor:
  - platform: wifi_signal
    name: "WiFi Signal Sensor"
    id: ${name}_wifi_signal_sensor
    update_interval: ${device_sampling_time}

    
################################################################################
# Text Sensors
################################################################################
text_sensor:
  - platform: wifi_info
    ip_address:
      name: IP Address
      id: ${name}_ip_address
    ssid:
      name: Connected SSID
      id: ${name}_connected_ssid

  - platform: version
    name: "ESPHome Version"
    hide_timestamp: true

################################################################################
# Switches
################################################################################
switch:
  - platform: restart
    name: "Restart"
    id: device_restart
  
  - platform: safe_mode
    name: "Use Safe Mode"
    id: device_safe_mode

################################################################################
# Check WiFi connection
################################################################################
interval:
  - interval: ${device_sampling_time}
    then:
      - if:
          condition:
            wifi.connected:
          then:
            - lambda: |-
                id(wifi_connection) = true;
          else:
            - lambda: |-
                id(wifi_connection) = false;

#===============================================================================
# Specific configuration
#===============================================================================
i2s_audio:
  i2s_lrclk_pin: GPIO33
  i2s_bclk_pin: GPIO19

media_player:
  - platform: i2s_audio
    name: ${friendly_name}
    dac_type: external
    i2s_dout_pin: GPIO22
    mode: mono
```

### Anything in the logs that might be useful for us?

```txt
error message during compilation:

In file included from /data/cache/platformio/packages/framework-arduinoespressif32-libs/esp32/include/driver/deprecated/driver/i2s.h:23,
                 from src/esphome/components/i2s_audio/i2s_audio.h:9,
                 from src/esphome/components/i2s_audio/i2s_audio.cpp:1:
/data/cache/platformio/packages/framework-arduinoespressif32-libs/esp32/include/driver/deprecated/driver/adc.h:19:2: warning: #warning "legacy adc driver is deprecated, please migrate to use esp_adc/adc_oneshot.h and esp_adc/adc_continuous.h for oneshot mode and continuous mode drivers respectively" [-Wcpp]
   19 | #warning "legacy adc driver is deprecated, please migrate to use esp_adc/adc_oneshot.h and esp_adc/adc_continuous.h for oneshot mode and continuous mode drivers respectively"
      |  ^~~~~~~
/data/cache/platformio/packages/framework-arduinoespressif32-libs/esp32/include/driver/deprecated/driver/i2s.h:27:2: warning: #warning "This set of I2S APIs has been deprecated, please include 'driver/i2s_std.h', 'driver/i2s_pdm.h' or 'driver/i2s_tdm.h' instead. if you want to keep using the old APIs and ignore this warning, you can enable 'Suppress legacy driver deprecated warning' option under 'I2S Configuration' menu in Kconfig" [-Wcpp]
   27 | #warning "This set of I2S APIs has been deprecated, \
      |  ^~~~~~~
```

### Additional information

[logs_esphome-web-792bd4_run.txt](https://github.com/user-attachments/files/21464609/logs_esphome-web-792bd4_run.txt)

epetwes, 2025-07-29:
ChatGPT answer this:

---

### ⚠️ Compilation Warning Summary:

You're seeing this warning during compilation:

```
#warning "legacy adc calibration driver is deprecated, please migrate to use esp_adc/adc_cali.h and esp_adc/adc_cali_scheme.h"
```

### 🧾 What it means:

The code is using a **deprecated ADC calibration driver** (`esp_adc_cal.h`) that has been replaced by a newer API provided by Espressif:

* Old: `esp_adc_cal.h`
* New: `esp_adc/adc_cali.h` and `esp_adc/adc_cali_scheme.h`

Espressif recommends **migrating to the new calibration scheme** for better maintainability and future support.

---

### ✅ What you can do:

* You can **ignore the warning** for now — it won't break your build.
* Or you can **manually migrate** to the new calibration API (`adc_cali_*`), which requires code changes.
* If you're using **ESPHome**, it's best to wait for an official update unless you're maintaining a custom fork.

---

Jn115759, 2025-07-29:
> ChatGPT answer this:
>
> ### ⚠️ Compilation Warning Summary:
> You're seeing this warning during compilation:
>
> ```
> #warning "legacy adc calibration driver is deprecated, please migrate to use [...]

Dear Epetwes, thank you for your feedback. The reason why I created this issue is to inform ESPHome maintainers about this warning. Hope someone will be able to create a fix for one of the next releases of the ESPHome software.

artlosi, 2025-11-13:
Hi there.
Here using ESPHome in Home Assistant; i2s_audio speaker does not work. After I get warnig messages during compilation about deprecation, at run time I get the error "[i2s_audio.speaker:572] Failed to install I2S legacy driver".
Which is the plan for the official ESPHome update?
Thanks
ESPHome version 2025.10.5