esphome/esphome #9966: -- git rev-parse returned 'fatal: not a git repository (or any parent up to mount point /)

comet424, 2025-07-29:
### The problem

i cant update 5 6 of my 54 esphome devices... the 5 or 6  devices  keep coming up with the same error

```

INFO ESPHome 2025.7.4
INFO Reading configuration /config/esphome/bluetooth-server-1-building.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing bluetooth-server-1-building (board: esp32dev; framework: espidf; platform: https://github.com/pioarduino/platform-espressif32/releases/download/53.03.13/platform-espressif32.zip)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - framework-espidf @ 3.50302.0 (5.3.2) 
 - tool-cmake @ 3.30.2 
 - tool-esptoolpy @ 4.8.6 
 - tool-mklittlefs @ 3.2.0 
 - tool-ninja @ 1.7.1 
 - tool-riscv32-esp-elf-gdb @ 14.2.0+20240403 
 - tool-xtensa-esp-elf-gdb @ 14.2.0+20240403 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-xtensa-esp-elf @ 13.2.0+20240530
Reading CMake configuration...
-- git rev-parse returned 'fatal: not a git repository (or any parent up to mount point /)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).'
-- Building ESP-IDF components for target esp32
-- Configuring incomplete, errors occurred!

fatal: not a git repository (or any parent up to mount point /)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).
CMake Error at /data/cache/platformio/packages/framework-espidf/tools/cmake/build.cmake:552 (message):
  /root/.platformio/penv/.espidf-5.3.2/bin/python: Error while finding module
  specification for 'idf_component_manager.prepare_components'
  (ModuleNotFoundError: No module named 'idf_component_manager')

Call Stack (most recent call first):
  /data/cache/platformio/packages/framework-espidf/tools/cmake/project.cmake:710 (idf_build_process)
  CMakeLists.txt:3 (project)

========================== [FAILED] Took 0.68 seconds ==========================
```

### Which version of ESPHome has the issue?

2025.7.4

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

### Component causing the issue

esphome

### YAML Config

```yaml
here is part of my yaml

esphome:
  name: bluetooth-server-1-building
  friendly_name: "Bluetooth Server #1 (Building)"

#esp32:
#  board: esp-wrover-kit
#  framework:
#    type: arduino
esp32:
  board: esp32dev
  framework:
    type: esp-idf 

# Enable logging
logger:

# Enable Home Assistant API
api:
  #password: !secret api_password
  encryption:
    key: "NDCDMqx3i89RCvONsK9cL8LiTdvx/H2anWI2EAMCI+A="
ota:
  platform: esphome
  password: !secret ota_password
```

### Anything in the logs that might be useful for us?

```txt

```

bojanpotocnik, 2025-07-31:
Clearing the cache using commands below did not resolve the issue:
```shell
# Inside the Advanced SSH & Web Terminal terminal
docker exec -it $(docker ps -qf 'name=esphome') bash
```
```shell
# Inside the ESPHome Docker container
rm -rv /data/cache/*
rm -rv /data/build/*
platformio system prune --cache --core-packages --force
platformio system prune --platform-packages  # Will ask for confirmation
```

Restarting the ESPHome Device Builder Add-On resolved the issue (as seen https://github.com/esphome/esphome/issues/9895#issuecomment-3124248550).

Looks like this is a duplicate of #9626, #9629, #9720, #9895.

comet424, 2025-07-31:
@bojanpotocnik ill try the uninstall esphome and reinstall it as rebooting home assistant doesnt solve that problem and ill see what happens.. frustrating though u gotta uninstall it to fix it if it fixes it

electr1cBugaloo, 2025-08-08:
I just had the same issue using esphome:latest pulled about 2-4 days ago. I just did a new docker pull and the issue seems to have vanished for me.

ssieb, 2025-08-09:
You need to completely rebuild the container to clear the platformio cache.

edfield, 2025-08-24:
Having thew same issue, running Home Assistant as VM in Unraid. Have uninstalled and reinstalled with no change. I have 2 other devices that update with no problems, just the bluetooth proxy is failing. Using the configuration from here:  https://esphome.io/projects/?type=bluetooth

[logs_esphome-web-6be38c_compile.txt](https://github.com/user-attachments/files/21953728/logs_esphome-web-6be38c_compile.txt)

ssieb, 2025-08-28:
Did you check the box for deleting data?

arh-au, 2025-08-28:
> That's not what I mean. When you uninstall an addon, there's a checkbox for removing data.

OK, I can't recall if I did check that box, so I will try the uninstall (specifically noting to check remove data option) and re-install.

arh-au, 2025-08-29:
> > That's not what I mean. When you uninstall an addon, there's a checkbox for removing data.
> 
> OK, I can't recall if I did check that box, so I will try the uninstall (specifically noting to check remove data option) and re-install.

OK, so now re-installed ESPHome again (ensuring remove data option was selected).  Now up to 3rd attempt at compiling a bluetooth proxy based on ESP32-S3 (ESPHome running under HA on Raspberry Pi 5 4Gb).

**Esphome YAML:**
`esphome:
  name: ha-bt-proxy
  friendly_name: ha-bt-proxy

esp32:
  board: esp32-s3-devkitc-1
  framework:
    type: esp-idf

# Enable logging
logger:
  level: DEBUG

# Enable Home Assistant API
api:
  encryption:
    key: "xxx"

ota:
  - platform: esphome
    password: "xxx"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Ha-Bt-Proxy Fallback Hotspot"
    password: "xxx"

captive_portal:

esp32_ble_tracker:
  scan_parameters:
    active: true
    interval: 320ms
    window: 30ms
   
bluetooth_proxy:
  active: true
  connection_slots: 3
`

**First install attempt logs:**
`
INFO ESPHome 2025.8.1
INFO Reading configuration /config/esphome/ha-bt-proxy.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing ha-bt-proxy (board: esp32-s3-devkitc-1; framework: espidf; platform: https://github.com/pioarduino/platform-espressif32/releases/download/54.03.21-2/platform-espressif32.zip)
--------------------------------------------------------------------------------
INFO Package configuration completed successfully
Tool Manager: Installing https://github.com/pioarduino/esp-idf/releases/download/v5.4.2/esp-idf-v5.4.2.zip
INFO Installing https://github.com/pioarduino/esp-idf/releases/download/v5.4.2/esp-idf-v5.4.2.zip
Unpacking  [####################################]  100%          
Tool Manager: framework-espidf@3.50402.0 has been installed!
INFO framework-espidf@3.50402.0 has been installed!
Tool Manager: Installing https://github.com/pioarduino/registry/releases/download/0.0.1/xtensa-esp-elf-14.2.0_20241119.zip
INFO Installing https://github.com/pioarduino/registry/releases/download/0.0.1/xtensa-esp-elf-14.2.0_20241119.zip
Downloading...
Tool Manager: Error: Please read https://bit.ly/package-manager-ioerror
ERROR Error: Please read https://bit.ly/package-manager-ioerror
[... 304 lines cut ...]
    xtensa-esp32s3-elf-g++

  is not a full path and was not found in the PATH.

  Tell CMake where to find the compiler by setting either the environment
  variable "CXX" or the CMake cache entry CMAKE_CXX_COMPILER to the full path
  to the compiler, or to the compiler name if it is in the PATH.
Call Stack (most recent call first):
  CMakeLists.txt:3 (project)

CMake Error at /data/cache/platformio/packages/framework-espidf/tools/cmake/project.cmake:571 (__project):
  The CMAKE_ASM_COMPILER:

    xtensa-esp32s3-elf-gcc

  is not a full path and was not found in the PATH.

  Tell CMake where to find the compiler by setting either the environment
  variable "ASM" or the CMake cache entry CMAKE_ASM_COMPILER to the full path
  to the compiler, or to the compiler name if it is in the PATH.
Call Stack (most recent call first):
  CMakeLists.txt:3 (project)

========================= [FAILED] Took 21.39 seconds =========================
`

Qhilm, 2025-09-01:
ok, thanks. How does the plugin learn about the esphome devices after being reinstalled from scratch? There must be some information I must provide it, for the plugin to find the devices, no?

Qhilm, 2025-09-01:
Thanks. Even if I select (as recommended above) "permanently delete this add-on's data"?

enchido, 2025-09-08:
por aquí sigue apareciendo el mismo error, he desinstalado y vuelto a instalar esphome y nada.

-- git rev-parse returned 'fatal: not a git repository (or any parent up to mount point /)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).'
-- The C compiler identification is unknown
-- The CXX compiler identification is unknown
-- The ASM compiler identification is GNU
-- Found assembler: /data/cache/platformio/packages/toolchain-riscv32-esp/bin/riscv32-esp-elf-gcc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - failed
-- Check for working C compiler: /data/cache/platformio/packages/toolchain-riscv32-esp/bin/riscv32-esp-elf-gcc
-- Check for working C compiler: /data/cache/platformio/packages/toolchain-riscv32-esp/bin/riscv32-esp-elf-gcc - broken
-- Configuring incomplete, errors occurred!

fatal: not a git repository (or any parent up to mount point /)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).
CMake Error at /data/cache/platformio/packages/tool-cmake/share/cmake-3.30/Modules/CMakeTestCCompiler.cmake:67 (message):
  The C compiler

    "/data/cache/platformio/packages/toolchain-riscv32-esp/bin/riscv32-esp-elf-gcc"

  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir: '/data/build/esphome-web-73fddc/.pioenvs/esphome-web-73fddc/CMakeFiles/CMakeScratch/TryCompile-aYnRza'
    
    Run Build Command(s): /data/cache/platformio/packages/tool-ninja/ninja -v cmTC_fb6ed
    [1/2] /data/cache/platformio/packages/toolchain-riscv32-esp/bin/riscv32-esp-elf-gcc   -march=rv32imc_zicsr_zifencei -o CMakeFiles/cmTC_fb6ed.dir/testCCompiler.c.obj -c /data/build/esphome-web-73fddc/.pioenvs/esphome-web-73fddc/CMakeFiles/CMakeScratch/TryCompile-aYnRza/testCCompiler.c
    FAILED: [code=1] CMakeFiles/cmTC_fb6ed.dir/testCCompiler.c.obj 
    /data/cache/platformio/packages/toolchain-riscv32-esp/bin/riscv32-esp-elf-gcc   -march=rv32imc_zicsr_zifencei -o CMakeFiles/cmTC_fb6ed.dir/testCCompiler.c.obj -c /data/build/esphome-web-73fddc/.pioenvs/esphome-web-73fddc/CMakeFiles/CMakeScratch/TryCompile-aYnRza/testCCompiler.c
    riscv32-esp-elf-gcc: fatal error: cannot execute 'as': posix_spawnp: No such file or directory
    compilation terminated.
    ninja: build stopped: subcommand failed.
    
    

  

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  /data/cache/platformio/packages/framework-espidf/tools/cmake/project.cmake:571 (__project)
  CMakeLists.txt:3 (project)

arh-au, 2025-09-09:
> Uninstalling and reinstalling ESPHome did not fix for me. Still an issue. My platform Raspberry Pi 5, HAOS.

Seemed to fix itself after 2025.8.2 upgrade - suspect it was something with one of the downloaded files since corrected and now compiles fine for me.

Bugrabugra, 2025-09-28:
> Try re-installing the add-on with the delete data option checked. Your yaml files are in config and won't be deleted.

It worked after I uninstalled the ESPHOME and install the yaml file for the general esp32c3 config not the specific sensor. During this installation platform.io files were installed as well.

ssieb, 2025-09-28:
And it's that last part that is what fixed it.

na4ma4, 2025-11-08:
If you're on macOS using homebrew, until the formula gets fixed this is what helped me:

```shell
$ HOMEBREW_NO_INSTALL_FROM_API=1 brew edit esphome
.... long time downloading the whole homebrew/core repo ....
```

edit the line saying `python@3.14` to `python@3.13`

```patch
diff --git a/Formula/e/esphome.rb b/Formula/e/esphome.rb
index dbdb502851b..b432381799b 100644
--- a/Formula/e/esphome.rb
+++ b/Formula/e/esphome.rb
@@ -20,7 +20,7 @@ class Esphome < Formula
   depends_on "cryptography" => :no_linkage
   depends_on "libyaml"
   depends_on "pillow" => :no_linkage
-  depends_on "python@3.14"
+  depends_on "python@3.13"
 
   pypi_packages exclude_packages: ["certifi", "cryptography", "pillow"]
 
```

then 

```shell
$ HOMEBREW_NO_INSTALL_FROM_API=1 brew install --build-from-source esphome
```

and finally `brew pin esphome` so it doesn't get automatically updated (until it's fixed, then `brew unpin esphome`

henrykuijpers, 2025-11-10:
Okay, with @na4ma4 's steps this is actually working (which is essentially what the fix that was planned does). @swoboda1337  would be nice to get that new release out, so that this issue is fixed for everyone.

swoboda1337, 2025-11-10:
Its in the list for 2025.10.5 https://github.com/esphome/esphome/milestone/521 and 2025.11 goes out next week

swoboda1337, 2025-11-12:
> Okay, with [@na4ma4](https://github.com/na4ma4) 's steps this is actually working (which is essentially what the fix that was planned does). [@swoboda1337](https://github.com/swoboda1337) would be nice to get that new release out, so that this issue is fixed for everyone.

2025.10.5 is out