esphome/esphome #9982: Duplicate HA Event fired on_press/on_click of binary_sensor

Plawasan, 2025-07-30:
### The problem

I am receiving duplicate HA events as a result of a ESPHome GPIO binary sensor on_press state. The binary sensor is well behaved and debounced (activates for 1s), logger on the same state only generates a single line.

I'm including only the sensor config as the entire YAML is 1500+ lines but I'm confident this is the only instance where a HA event is fired.

### Which version of ESPHome has the issue?

2025.7.1

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

### Component causing the issue

binary_sensor

### YAML Config

```yaml
binary_sensor:
  - platform: gpio
    id: entry_keypad_doorbell
    device_id: entry_keypad_device
    icon: mdi:bell-outline
    name: "Doorbell"
    pin: 
      number: GPIO17
      mode:
        input: true
        pullup: true
      inverted: true
    filters:
      - delayed_on: 100ms
      - delayed_off: 100ms
    on_press:
      then:
        - homeassistant.event:
            event: esphome.doorbell_pressed
        - logger.log: "Sending DOORBELL PRESSED event"  

I have tried using this config however that also fires two events:

    on_click: 
      min_length: 900ms
      max_length: 1200ms  
      then:
        - homeassistant.event:
            event: esphome.doorbell_pressed
        - logger.log: "Sending DOORBELL PRESSED event"
```

### Anything in the logs that might be useful for us?

```txt
Relevant ESPHome Log:
[20:09:56][D][main:838]: Sending DOORBELL PRESSED event
[20:09:56][D][binary_sensor:026]: 'Doorbell': New state is ON
[20:09:57][D][binary_sensor:026]: 'Doorbell': New state is OFF

HA Events received:
event_type: esphome.doorbell_pressed
data:
  device_id: 0ab71255723b3b0dcff16f3c32bcca71
origin: LOCAL
time_fired: "2025-07-30T18:09:56.501855+00:00"
context:
  id: 01K1E8DMRN35HZ2518EVCR01SR
  parent_id: null
  user_id: null

event_type: esphome.doorbell_pressed
data:
  device_id: 0ab71255723b3b0dcff16f3c32bcca71
origin: LOCAL
time_fired: "2025-07-30T18:09:56.511595+00:00"
context:
  id: 01K1E8DMRZE46M8QBP3ZAEJ969
  parent_id: null
  user_id: null
```

Plawasan, 2025-08-08:
I only now read the discussion in the linked PR -  I didn't realize that it was relevant, I do in fact have this device connected to two HA instances (prod + test). I've removed the device from the my HA test instance expecting the issue to be resolved however instead of 2 events, I now get 4 (even after device reboot)

```
Event 3 fired 10:14:
event_type: esphome.doorbell_pressed
data:
  device_id: 0ab71255723b3b0dcff16f3c32bcca71
origin: LOCAL
time_fired: "2025-08-08T08:14:06.480624+00:00"
context:
  id: 01K24BX3MGC7DN6RX9Z8N25T8R
  parent_id: null
  user_id: null

Event 2 fired 10:14:
event_type: esphome.doorbell_pressed
data:
  device_id: 0ab71255723b3b0dcff16f3c32bcca71
origin: LOCAL
time_fired: "2025-08-08T08:14:06.480624+00:00"
context:
  id: 01K24BX3MGC7DN6RX9Z8N25T8R
  parent_id: null
  user_id: null

Event 1 fired 10:14:
event_type: esphome.doorbell_pressed
data:
  device_id: 0ab71255723b3b0dcff16f3c32bcca71
origin: LOCAL
time_fired: "2025-08-08T08:14:06.480417+00:00"
context:
  id: 01K24BX3MG591F82S66SRFF6DY
  parent_id: null
  user_id: null

Event 0 fired 10:14:
event_type: esphome.doorbell_pressed
data:
  device_id: 0ab71255723b3b0dcff16f3c32bcca71
origin: LOCAL
time_fired: "2025-08-08T08:14:06.480417+00:00"
context:
  id: 01K24BX3MG591F82S66SRFF6DY
  parent_id: null
  user_id: null
```

clydebarrow, 2025-08-08:
Well, using your yaml and with the device connected to 2 HA instances, I see no issue - each instance sees the event once.

clydebarrow, 2025-08-08:
Are you sure you don't have anything in HA that would fire that event?

Plawasan, 2025-08-08:
I'll look into that but would that not show a different device_id than the event coming from the ESPHome client?

When I fire the even manually from developer tools, it "looks" different than the event coming from ESPHome:

[image]

```
event_type: esphome.doorbell_pressed
data: {}
origin: REMOTE
time_fired: "2025-08-08T08:49:04.807820+00:00"
context:
  id: 01K24DX4S7NZZ6NEV67QPX06ZD
  parent_id: null
  user_id: 15ff0151b05241c9a6eaca81dcc1c420
```

clydebarrow, 2025-08-08:
Agreed, they look different. The REMOTE vs LOCAL origins look backwards - but that's not likely relevant here.