esphome/esphome #9984: Invalid header: F4.F3.F2.F1

operaldi, 2025-07-30:
### The problem

Since the last esphome version, I have now this warning, in the home-assistant log:

WARNING (MainThread) [homeassistant.components.esphome.manager] device_name: [W][ld2410:469]: Invalid header: F4.F3.F2.F1

I have the same warning in the esphome builder log:

[16:59:54][W][ld2410:469]: Invalid header: F4.F3.F2.F1

### Which version of ESPHome has the issue?

ESPHome 2025.7.4

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

### Component causing the issue

ld2410

### YAML Config

```yaml

```

### Anything in the logs that might be useful for us?

```txt

```

ssieb, 2025-07-30:
What version of Home Assistant?

ssieb, 2025-07-30:
Oh, you're referring to logs from the device, not actually HA logs.  What are the logs around that line?

ssieb, 2025-07-30:
The esphome builder log.  I think you can just ignore it.  It's probably just a communication error.
You didn't include your yaml, so I don't know what you're using now, but you could try switching to esp-idf.

operaldi, 2025-07-30:
I am already with esp-idf.

Olivier