esphome/esphome #9985: sx126x - Transmit packet failure

pazzotranquillo, 2025-07-30:
### The problem

Hello,
I have an Heltec lora32 v3 (https://heltec.org/project/wifi-lora-32-v3/) board and I'd like to transmit packets for a reed sensor state.

Unfortunately I'm getting **[E][sx126x:309]: Transmit packet failure** 

### Which version of ESPHome has the issue?

2025.7.4

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

### Component causing the issue

sx126x

### YAML Config

```yaml
esphome:
  name: ${name2}
  on_boot:
    - priority: 1000
      then:
        - lambda: |-
            pinMode(36, OUTPUT);
            digitalWrite(36, LOW);
            delay(100);

esp32:
  board: esp32-s3-devkitc-1
  framework:
    type: arduino

logger:
#  baud_rate: 0  # disable UART logging to save power
  level: VERBOSE

api:
  encryption:
    key: !secret api_key_esp-lora-002

ota:
  - platform: esphome
    password: !secret ota_pwd_esp-lora-002

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

i2c:
  - id: bus_a
    sda: GPIO17
    scl: GPIO18

spi:
  id: spi0
  clk_pin: GPIO9      # GPIO9 (SCK, Pin 17)
  mosi_pin: GPIO10    # GPIO10 (MOSI, Pin 16)
  miso_pin: GPIO11    # GPIO11 (MISO, Pin 15)

sx126x:
  id: lora_radio
  spi_id: spi0
  cs_pin: GPIO8         # GPIO8 (NSS/CS, Pin 18)
  rst_pin: GPIO12        # GPIO13 (LoRa_RST, Pin 13)
  busy_pin: GPIO13       # GPIO12 (LoRa_Busy, Pin 14)
  dio1_pin: GPIO14       # GPIO14 (DIO1, Pin 12)
  rf_switch: false       # GPIO6 (Pin 19) -- Choose any available GPIO if this is used elsewhere
  modulation: LORA
  frequency: 868000000
  spreading_factor: 7
  bandwidth: 125_0kHz
  coding_rate: CR_4_5
  crc_enable: true
  pa_power: 14
  hw_version: sx1262
  rx_start: true

packet_transport:
  platform: sx126x
  sx126x_id: lora_radio
  #encryption: !secret lora_encryption_key
  #rolling_code_enable: true
  binary_sensors:
    - ${name2_sensor}_contact

binary_sensor:
  - platform: gpio
    pin: GPIO7      # GPIO7 (left side, not used for LoRa)
    name: ${name2_sensor}_contact
    id: ${name2_sensor}_contact
    device_class: door
```

### Anything in the logs that might be useful for us?

```txt

```

pazzotranquillo, 2025-07-31:
Solved by adding the following lines.

```
  tcxo_voltage: 1_8V
  tcxo_delay: 5ms
```

mortification77, 2025-08-07:
Whoa, @pazzotranquillo, I cannot thank you enough for your solution which solved my issue when all else seemed hopeless!

pazzotranquillo, 2025-08-07:
> Whoa, [@pazzotranquillo](https://github.com/pazzotranquillo), I cannot thank you enough for your solution which solved my issue when all else seemed hopeless!

Cheers!

pazzotranquillo, 2025-08-07:
BTW @mortification77 any interesting projects you are working on? I’m on the good path for setting up a remote mailbox door sensor, however I’m struggling with deep-sleep and standby mode. I was wondering if you could help me with that.