esphome/esphome #10017: Exception when connecting to wifi (ESP8266)

csammis, 2025-08-02:
### The problem

Following on from https://github.com/esphome/esphome/issues/9965 - I believe I'm hitting a similar exception in the ESP8266 code, although in this case it's happening on _every_ connection. Rolling back to 2025.6.3 and reinstalling the firmware image fixed the issue.

### Which version of ESPHome has the issue?

2025.7.4

### What type of installation are you using?

Docker

### What platform are you using?

ESP8266

### Component causing the issue

wifi

### YAML Config

```yaml
esphome:
  name: makerspace-air-quality
  friendly_name: Makerspace Air Quality

esp8266:
  board: d1_mini

uart:
  rx_pin: D2
  baud_rate: 9600

sensor:
  - platform: pm1006
    pm_2_5: 
      name: "Particulate Matter 2.5µm Concentration"

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "TurhJiQHwpDRZ5PKbK22AchKRfkf3Y3tn1FwxkDwP5g="

ota:
  - platform: esphome
    password: "22dc2c7a357aaea952d40d1b89c1e273"

wifi:
  ssid: !secret wifi_iot_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Makerspace-Air-Quality"
    password: "uGKAoaEVtPlO"

captive_portal:
```

### Anything in the logs that might be useful for us?

```txt
[19:16:51]--------------- CUT HERE FOR EXCEPTION DECODER ---------------
[19:16:51]
[19:16:51]Exception (28):
[19:16:51]epc1=0x4021fede epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000014 depc=0x00000000
[19:16:51]
[19:16:51]>>>stack>>>
[19:16:51]
[19:16:51]ctx: cont
[19:16:51]sp: 3ffffdc0 end: 3fffffd0 offset: 0150
[19:16:51]3fffff10:  3fff2bd8 3fff2bcc 00000000 00000000  
[19:16:51]3fffff20:  3ffe8e4f 3fff12f4 3fff11f4 4021a26c  
[19:16:51]3fffff30:  3fffff38 0000000a 6d6d6173 692d7369  
[19:16:51]3fffff40:  0000746f 00000001 0000f598 40230b18  
[19:16:51]3fffff50:  3fff3b0c 00000000 3ffefcfc 3ffefb90  
[19:16:51]3fffff60:  0000f598 3fff12f4 3fff11f4 4021a444  
[19:16:51]3fffff70:  0000f598 0000f588 00000001 3fff11f4  
[19:16:51]3fffff80:  00000000 0000f598 3ffefb90 4021b73b  
[19:16:51]3fffff90:  0000f598 3fff11f4 00000018 3ffeff94  
[19:16:51]3fffffa0:  3fffdad0 00000000 3ffeff68 3ffeff94  
[19:16:51]3fffffb0:  3fffdad0 00000000 3ffeff68 4022d308  
[19:16:51]3fffffc0:  feefeffe feefeffe 3fffdab0 40100641  
[19:16:51]<<<stack<<<
[19:16:51]
[19:16:51]--------------- CUT HERE FOR EXCEPTION DECODER ---------------
```

### Additional information

Both of the boards with this issue are d1_mini ESP8266. I flashed one other board with 2025.7.4, but it is esp32-c3-devkitm-1 and is working fine connecting to the same wifi AP.

ssieb, 2025-08-05:
@csammis are you using the linked PR?  Can you get a decoded stack strace?

@mxbchr unless you're getting an exception, it's not the same issue.  Create a new issue with serial logs.

csammis, 2025-08-09:
> Download the elf file from the device menu in the builder, then go to https://esphome.github.io/esp-stacktrace-decoder/

Thank you. Unfortunately this site does not work. Debugging the page's JS source I can see that `wasm.decode()` is returning an empty array given the ELF file and stack trace from the logs, but that's as far as I go. I do not have the skillset necessary to debug a WASM module.

I was able to narrow down the root cause of the ESPHome exception. I replicated the problem in 2025.7.5. The console log and YAML are attached to this post. Note that the target AP, `sammis-iot`, is **hidden**, and `fast_connect` is `False`. When I set `fast_connect` to `True`, there is no exception and the ESP device boots normally.

Downgrading to 2025.6.3 and setting `fast_connect` to `False`, there is no exception and the ESP device boots normally.

The docs specify that `fast_connect` is "required" for hidden APs, so I guess this is not *wrong*, but it *is* a regression and of course an exception is not desirable behavior. According to the logs (and my AP) the 2025.7.5 device *does* connect. It just immediately crashes.

[makerspace-air-quality_logs.txt](https://github.com/user-attachments/files/21700631/makerspace-air-quality_logs.txt)

[makerspace-air-quality.yaml.txt](https://github.com/user-attachments/files/21700639/makerspace-air-quality.yaml.txt)

ssieb, 2025-08-26:
Try https://github.com/esphome/esphome/pull/10445